syssi/esphome-soyosource-gtn-virtual-meter #28: All Values to read from Soyo are unknown

RonBO125, 2022-03-09:
Hi,

I have a problem where I don't know how to find the error.
I've connected everything like described and the command that will send values to soyo is working (Power demand) so I think the communication is working.
But no values could be read, they all are unknown.

Do you have a hint what could be the problem?

Thanks for your help!

ChrisHomewood, 2022-03-09:
RS485 Serial communication (the type used on the inverter) uses the same wire for sending and receiving data, so if you are able to send data successfully you should also be able to receive it. However if you are using a RS485 to RS232 serial converter board (very likely if using a microcontroller) then its possible the RX pin is disconnected from the microcontroller as RS232 uses one wire for transmit and one for receive. The connection will work just fine without that RX line connected, but on transmit (TX pin) only. So go check for a loose or missing connection. On occasions where I have had communication in one direction only it has turned out to be a loose wire.

Other possible situations are a bad quality cable connection to the inverter. RS485 spec recommends using a termination resistor to avoid signal reflections. Usually you can get away with not having it, but again worth adding if you havn't already. Similarly you can often get away with any old bit of wire as a connection, but really you should be using a twisted pair data cable. Under ideal conditions the maximum cable length can be 1200 meters so that is very unlikely to be an issue.

syssi, 2022-03-09:
I've added a debug section to the docs now: https://github.com/syssi/esphome-soyosource-gtn-virtual-meter#debugging

You can enable the debug output of the uart component. So you can see all incoming (`<<<`) and outgoing (`>>>`) traffic (in bytes) at the log. If the inverter responds with garbage you should it.

Bad news: I'm aware of silent Soyosource devices. If you had bad luck your device is one of these devices. I don't know if it is a firmware limitation or the RS485 converter module is broken from the very beginning.

Good news: The limiter is working nevertheless and the inverter feedback (status message) is just sugar. ;-)

ChrisHomewood, 2022-03-09:
Also note that the status message provides wildly inaccurate measurements both in terms of voltage and output power, so don't feel like you are missing out on anything, its nice to have an indication its working but its not going to give accurate statistics.

Interestingly the real inverter power output (measured by a meter) does match the requested output so it can measure itself properly, I think the status message might be showing power drawn from the battery before inverter efficiency losses rather than output from the inverter, but I have not measured that theory yet. In any event voltage is off by a couple volts!

RonBO125, 2022-03-09:
I‘ve activated the debug and found following:

INFO 192.168.69.41: Error while reading incoming messages: Error while reading data: [Errno 104] Connection reset by peer
INFO Disconnected from ESPHome API for 192.168.69.41
WARNING Disconnected from API
INFO 192.168.69.41: Ping Failed: Error while reading data: [Errno 104] Connection reset by peer
INFO Successfully connected to 192.168.69.41

What means Errno 104?

Thanks for your Support and help!

syssi, 2022-03-09:
This is just a connection reset. The TCP connection between your client (`esphome logs` or HA core) and the ESPHome node failed. A few seconds later the client reconnected automatically. This looks like an unstable WiFi connection or unstable power supply.

RonBO125, 2022-03-09:
Ok, so I will have a look at everything. The WiFi had no Problems until now, but I will also Check this.

Thank you very much!

closed 2022-03-09 by RonBO125

syssi, 2022-03-09:
Feel free to open the issue again if you get stuck! :-)