syssi/esphome-soyosource-gtn-virtual-meter #45: Add display support

syssi, 2022-04-16:
Introduce a `soyosource_display` component to replace the ESP8266 dongle of the WiFi version.

jsphuebner, 2022-04-16:
Read settings request >>> 55 03 00 00 00 00 00 00 00 00 00 FC

Write settings has no reply

syssi, 2022-04-16:
Checksum

```
#!/usr/bin/python3

frames = [
  [0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00], # FE
  [0x03, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00], # FC
  [0x0B, 0x2C, 0x2B, 0x5A, 0x63, 0x00, 0x00, 0x00, 0x04, 0x12], # CA
  [0x0B, 0x2D, 0x2B, 0x5A, 0x64, 0x00, 0x00, 0x00, 0x04, 0x12], # C8
  [0x01, 0x72, 0x93, 0x40, 0xD4, 0x30, 0x2C, 0x2B, 0x00, 0xFA, 0x64, 0x5A, 0x03], # A3
  [0x03, 0x84, 0x91, 0x40, 0x01, 0xC5, 0x00, 0xDB, 0x00, 0xF7, 0x63, 0x02, 0xBC], # FE vs. EE
]

for frame in frames:
  sum = 0xff
  for i in frame:
    sum = (sum - i) & 0xff
  print(hex(sum))
```

syssi, 2022-04-16:
Pinout USB-TTL:  5V, TX, RX, GND

syssi, 2022-04-18:
Fixed #46.

closed 2022-04-18 by syssi