syssi/esphome-soyosource-gtn-virtual-meter #50: Add docs to explain the display component

syssi, 2022-04-17:
See #45. Mention the pinout of the USB-TTL connector: 5V, TX, RX, GND

user0x01, 2022-06-24:
Can i flash the wlan interface (V1.7) with the esphome display component,
with a 5V ftdi adapter on den "USB"-Port and GPIO0 on Ground?

syssi, 2022-06-24:
Another thought: I never traced the connection between the USB jack, the microcontroller used as a level shifters (this is a guess) and the ESP. It's possible the RX and TX lines aren't connected to GPIO1 and 3 of the ESP. May be another GPIO using software serial is used here. In other words: If the flashing using the USB jack doesn't work you should connect the FTDI directly to the ESP. As next step you have to identify the GPIOs "connected" to the USB jack to configure the UART component accordingly.

syssi, 2022-06-24:
Feel free to ask for details. I will have a look the WiFi interface and try to provide some support. In best case I can backup the stock firmware and can give it a try too.

user0x01, 2022-06-24:
You're probably right.

I get on grip1+3 only this message AT+INFO=ok
Module Type=0, Software Version=1106
Copyright By www.51byb.com.

The payload pins seem to be different.

syssi, 2022-06-24:
I also made some progress. I've attached a logic analyzer to all soldered pins of the ESP8266. It looks like this:

[image]

If I provide power the board (5V) and listen to the traffic at the TX pin of the USB connector I can see periodic requests:

```
55 02 00 00 00 00 00 00 00 00 00 fd
```

If I check the logic analyzer there is no traffic at the GPIOs of the ESP8266. If I connect to the WiFi accesspoint now and open the Soyosource app the blue LED of the ESP8266 flickers and I can see requests + responses on GPIO1/GPIO3:

[image]

Funny thing: The requests at the USB connector are different. I assume the Nuvoton uC on the back of the PCB isn't a "transparent proxy". It looks like the firmware is slightly "protocol aware".

Long story short: You cannot flash the ESP8266 via the USB connector. You have to use GPIO1 and GPIO3 directly. The 4+1 pin header at the PCB doesn't expose the GPIOs of the ESP. The header is connected to the Nuvoton uC. As soon ESPHome is flashed to the ESP the Nuvoton shouldn't notice the difference and communication using the display protocol should be possible.

user0x01, 2022-06-24:
Cool. 

I try it.
My Jumperwires holds perfect in the holes off the esp. 
Das geht, ohne dass ich was löten muss.

syssi, 2022-06-25:
I've added some more details about the WiFi Dongle V1.7 here: https://github.com/syssi/esphome-soyosource-gtn-virtual-meter/tree/add-wifi-dongle/docs/wifi-dongle

syssi, 2022-06-25:
I've added some instructions how to dump the flash content: https://github.com/syssi/esphome-soyosource-gtn-virtual-meter/tree/add-wifi-dongle/docs/wifi-dongle

I learned: The message `Module Type=0, Software Version=1106\nCopyright By www.51byb.com.` is transmitted from the Nuvoton uC. If you see this message you have to swap RX/TX to be able to talk to the ESP32 as soon the MS51 is stopped by pulling nRESET down.

user0x01, 2022-06-26:
When I use uart in the config, the ESP disappears from router.
I think the Nuvoton is probably the master, the ESP just the wifi interface.

syssi, 2022-06-27:
This sounds like the ESP crashs. Could you attach a USB-to-TTL adapter to GPIO1/GPIO3 to monitor the boot process? You should see the reason for not joining the WiFi here.

syssi, 2022-06-27:
I didn't find any other connections next to the serial line between the ESP and the Nuvoton MS51 yet. This means the MS51 isn't able to reset/reboot the ESP or control some other import GPIOs. Only the serial line is used for communication/instructions between the microcontrollers.

Let's call the MS51 the "primary" because the LEDs and the button are attached to this microcontroller and this uC decides about the operation mode of the ESP (by sending some AT command to switch the mode).

syssi, 2022-06-27:
I don't want to flash my WiFi dongle as long as the following features aren't ready: #51, #52, #53
If restoring the stock firmware doesn't work I'm stuck.

user0x01, 2022-06-27:
I know, the sticks are available as spare parts, but only at crazy prices.

The esphome flasher could not connect to log.
I am checking the esp output with the VSCode console tonight.
And then I try to restore the firmware dump.

syssi, 2022-06-27:
Funny! :-) But good to know: One of my statements above was wrong. The 51byb.com message is from the ESP and not from the MS51:

```
$ strings docs/wifi-dongle/esp8266-stock-firmware-SY22MAR8331.bin  | grep 51by
Copyright By www.51byb.com
51byb.com
51byb

$ strings docs/wifi-dongle/esp8266-stock-firmware-SY22MAR8331.bin  | grep "Module Type"
Module Type=%d, Software Version=%s

$ strings docs/wifi-dongle/esp8266-stock-firmware-SY22MAR8331.bin  | grep "license"
AT+INFO=invalid hardware to run the software,need license pls vechat:lijay910 or qq:948528379 failed
AT+INFO=invalid hardware to run the software,need license pls vechar:lijay910 or qq:948528379 failed
```

syssi, 2022-06-28:
Some new findings:

The frames between the Soyosource and Nuvoton MS51 are different to the frames between the ESP8266 and MS51. Every frame from the Soyosource inverter starts with `0xA6`. Every frame from the MS51 starts with `0x5A`. 

I've flashed this [configuration](https://github.com/syssi/esphome-soyosource-gtn-virtual-meter/blob/add-wifi-dongle-support/esp8266-wifi-dongle-example.yaml) to my WiFi dongle now. It's stable and responsible but I've to improve the display component to be able to decode this new kind of frames: https://github.com/syssi/esphome-soyosource-gtn-virtual-meter/blob/add-wifi-dongle-support/docs/wifi-dongle/nuvoton-esp8266-pdus.txt

I will give it a try but I'm unsure it's worth the effort. The dongle can be easily replaced. There are ESP8266 boards available equipped with a level shifter: https://de.aliexpress.com/item/32848420391.html

user0x01, 2022-06-29:
I get these log messages
`
[19:11:10][D][uart_debug:114]: >>> 55:01:00:00:00:00:00:00:00:00:00:FE
`

My ESPHome does not know esphome::soyosource_display::SoyosourceSelect->at and ->index_of. I commented Select in the config.

syssi, 2022-06-29:
Please use the feature branch by changing the external component URL slightly:

```
substitutions:
  name: soyosource-wifi-dongle
  external_components_source: github://syssi/esphome-soyosource-gtn-virtual-meter@add-wifi-dongle-support
```

user0x01, 2022-06-29:
schon gesehen

user0x01, 2022-06-29:
> Do you use a recent ESPHome version? If so please try `esphome clean config.yaml ; esphome run config.yaml`. This should fix the issue. The TX message looks good.

I was on the March version, the repo moved.
https://esphome.io/changelog/2022.2.0.html

I updated esphome, 
resoldered the usb data wire on the shifter, 
and swap the data lines.

**it can read** :o)

syssi, 2022-06-29:
It's possible to control the settings of the inverter via the WiFi dongle now. I've merged the feature branch. Please change your config back to `github://syssi/esphome-soyosource-gtn-virtual-meter@main`.

user0x01, 2022-06-29:
something is broken,
I test the add-wifi-dongle-support branch tomorrow again from my local copy

syssi, 2022-06-29:
I've deleted the branch at this repository. Please change your configuration back to `@main`.