syssi/esphome-soyosource-gtn-virtual-meter #60: Add how to flash the alternative firmware to the WiFi dongle

syssi, 2022-06-29:
Extend the WiFi dongle page by a "How to flash the alternative firmware" paragraph.

1. Backup the stock firmware
2. Flash the alternative firmware: `esphome run esp8266-wifi-dongle-example.yaml`
3. Remove all wires especially the `nRESET` and `GPIO0` pull-down
4. Plug the WiFi dongle back into the Soyosource inverter
5. Enjoy!

https://github.com/syssi/esphome-soyosource-gtn-virtual-meter/blob/main/docs/wifi-dongle/README.md

mrbluebrett, 2023-01-09:
I’m trying to flash the wifi 1.7 module but i cant figure out how to do it… 
I have a ch340 module ( 
https://a.aliexpress.com/_EIqwewj ) and i connect the wires to the wifi module but i cant read or flash the module. Should I connect the Rx to Rx or the Rx to Tx? 
also which flasher can i use? Web? Espflash?
I have esphome running in docker and usually download the firmware and flash it …

syssi, 2023-01-10:
Did you try this how to to backup the ESP? https://github.com/syssi/esphome-soyosource-gtn-virtual-meter/blob/main/docs/wifi-dongle/README.md#how-to-backup-the-esp8266

syssi, 2023-01-10:
The CH340 based usb-to-serial converter is fine. You can use any flashing tool as soon as you have prepared the wifi dongle properly (see the how to).

mrbluebrett, 2023-01-10:
Ok, I did it thank you.
I was connecting the Tx and Rx to the usb board pins, not to the esp itself.

syssi, 2023-01-10:
The pins of the USB-plug are connected to the MS51FB9AE. The MS51 expects the specific protocol and isn't a transparent bridge / not only acts as a level shifter. So it's required to connect RX/TX directly to the ESP.