syssi/esphome-soyosource-gtn-virtual-meter #68: PIP inverters

ingoooooo, 2022-07-19:
Hi,

first of all thank you very much for your effort. Your ideas and the explanation of how to implement it are really great.
I am wondering if this virtual meter imeplementation does also work with bigger inverters from PIPsolar/Voltronic to reach a zero feed in scenario.

Did anyone try this already or isn't there a chance at all?

Thanks,
Ingo

syssi, 2022-07-19:
Do you own a grid tied PIP inverter? Do you know how to limit the power output? Does the device support zero export if a powermeter is attached?

ingoooooo, 2022-07-20:
No, not yet though I am planning to buy one.
My findings so far even though not 100% validated:

Sorotec is the manufacturer of these inverters (Revo II) which are sold by different companies like EASun, AXpert, Pow??... and so on.
https://www.sorotecpower.com/products-detail-134126

- It has a CT sensor to limit the output.

- It has a serial interface but it seems like the wattage cannot be set there. ( see https://github.com/jblance/mpp-solar/discussions/78 )

- This is the protocol documentation for the serial port https://github.com/jblance/mpp-solar/files/6328557/Revo.20Protocol.20V03--20201112.docx

closed 2023-01-24 by syssi