syssi/esphome-soyosource-gtn-virtual-meter #69: Unable to compile "esp8266-display-example.yaml"

senfkorn, 2022-08-07:
Hi,

I've a problem with compiling the "esp8266-display-example.yaml"

It says there is something with the select component:

src/esphome/components/soyosource_display/select/soyosource_select.cpp: In lambda function:
src/esphome/components/soyosource_display/select/soyosource_select.cpp:19:24: error: 'class esphome::soyosource_display::SoyosourceSelect' has no member named 'at'
   19 |     auto value = this->at(mapping_idx);

src/esphome/components/soyosource_display/select/soyosource_select.cpp: In member function 'virtual void esphome::soyosource_display::SoyosourceSelect::control(const string&)':
src/esphome/components/soyosource_display/select/soyosource_select.cpp:35:20: error: 'class esphome::soyosource_display::SoyosourceSelect' has no member named 'index_of'
   35 |   auto idx = this->index_of(value);

[image]

syssi, 2022-08-07:
Could you provide some details about your build stack (Linux distribution, g++ version, ESPHome version)? In general the CI is able to build the project.

syssi, 2022-08-07:
The `index_of` property was introduced at ESPHome 2022.5. An ESPHome update should fix the issue. I will update the minimum required ESPHome version at the README.

senfkorn, 2022-08-07:
Hi, 

this is my current setup:
-----------------------------------------------------------
 Add-on version: 2022.3.1
 You are running the latest version of this add-on.
 System: Debian GNU/Linux 11 (bullseye)  (amd64 / qemux86-64)
 Home Assistant Core: 2022.8.1
 Home Assistant Supervisor: 2022.07.0
-----------------------------------------

i'll have a look for updating my esphome

syssi, 2022-08-07:
I assume you have to switch to the official esphome add-on.

senfkorn, 2022-08-07:
Thanks for the help, you are really everywhere :)

I wasn't aware, that I've to change the esp addon. 
It's working

Thanks again

syssi, 2022-08-07:
The next touch point will be your solar charger. ;-) Which device did you buy?

closed 2022-08-07 by syssi

senfkorn, 2022-08-07:
Victron MPPT's

But they are working flawlessly.

My next point to tackle is the IR reader for the powermeter. 

here's a pic of my current constellation:

[image]

syssi, 2022-08-07:
Nice setup! Please keep in mind: The `esp8266-display-example.yaml` is the example configuration for generic ESP8266 (plus a logic level shifter) attached to the display port. The `esp8266-wifi-dongle-example.yaml` is the alternative firmware for the WiFi dongle. I'm saying this because you should be aware about the 5V TTL of the display port.

senfkorn, 2022-08-07:
right now I'm trying to use the WiFi dongle to output the rs485 for the limiter signal. It was working until I fried the RS232 to RS485 converter board by building it compact and touch both boards together. Tomorrow the new ones are coming in.

syssi, 2022-08-07:
Shit happens. I hope you didn't fried the internal RS485 converter of the Soyosource too. The replacement board is a bit expensive.

senfkorn, 2022-08-07:
I've checked it already with to original power clamp and it still works.

Shouldn't I ask why you know how expensive these boards are? ;)

syssi, 2022-08-07:
You are on the right track: I fried my "RS485 layer" too (once) and I was pretty happy about this "modular" setup. So I was able replace the fried components/modules only.

senfkorn, 2022-08-07:
does something speak against using the inverter in "battery constant power mode" and the writing the "output power limit" number with my desired power? 

I'm right now testing this configuration and it seems to work. 

Any concerns?

syssi, 2022-08-07:
The only concern: Updating the inverter setting very often will destroy the flash memory some day because of the limited number of write cycles.

senfkorn, 2022-08-07:
OK, thats a good point. 

So, I'll wait for the RS485 modules and do it the right way