syssi/esphome-soyosource-gtn-virtual-meter #71: MQTT documentation

careyer, 2022-08-25:
Hello there!
thank you very much for the awesome project. It looks very sophisticed and well mantained.

Not working with Home Assistant but with NodeRED in conjunction with Mosquitto, I would like to use the WiFi-TTL and RS485 functionlity via MQTT messages. Is there any documentation on how to use your systemw with MQTT messages alone? E.g. I am lacking the information on what MQTT topics to subscribe.
Any hint on this would be very welcome!

Thank you!

syssi, 2022-08-25:
The topics are auto-generated depending on the entity ids / sensor names. I recommend learning by doing here. ;-) Just flash the ESP with the desired example configuration and take a look at the ESPHome log. The node will report all topics on boot as log messages. Home Assistant is able to discover the ESPHome node (via MQTT) automatically because ESPHome publishes some discovery messages to a HA core friendly topic.

Some examples of my setup to illustrate the schema:

```
attic-powerwall/sensor/firstfloor_inverter_operation_mode/state Normal
attic-powerwall/sensor/firstfloor_inverter_limiter_operation_mode/state Auto
attic-powerwall/sensor/inverter0_power_demand/state 190
attic-powerwall/sensor/firstfloor_inverter_power_demand/state 0
attic-powerwall/sensor/firstfloor_inverter_energy_yesterday/state 0.105
attic-powerwall/switch/attic-powerwall_restart/state OFF
attic-powerwall/switch/attic-powerwall_safe_mode/state OFF
attic-powerwall/switch/firstfloor_inverter_manual_mode/state OFF
attic-powerwall/switch/firstfloor_inverter_emergency_power_off/state OFF
attic-powerwall/number/attic-powerwall_manual_power_demand/state 0.000000
attic-powerwall/number/attic-powerwall_max_power_demand/state 900.000000
# node name  / entity_type / entitiy_id / "state"  <value>
```

If you publish the message `ON` to a `command` topic of a `switch` entity the switch entity of the ESPHome node will be turned on. It's pretty the same for other entities (f.e. `number` entitiy).

syssi, 2022-08-25:
Some command example to control the different entity types:

```
mosquitto_pub -t 'basement-btle-gateway/button/button_entity_id_example/command' -m 'PRESS'
mosquitto_pub -t 'basement-btle-gateway/switch/switch_entity_id_example/command' -m 'ON'    # Supported values: ON, OFF, TOGGLE
mosquitto_pub -t 'basement-btle-gateway/number/number_entity_id_example/command' -m 10
mosquitto_pub -t 'basement-btle-gateway/select/select_entity_id_example/command' -m 'option1'
```

syssi, 2022-08-25:
> P.S. I've adapted the esp8266-display-example.yaml to work with my ESP32 DevKit. I want to be able to control both WiFI-TTL and RS485.

Perfect! You've picked the correct configuration example for your scenario (= generic ESP dev board). Please keep in mind the TTL of the WIFI-TTL connector is 5V.

careyer, 2022-08-25:
@syssi thank you very much for comming back to me in no time =)
I think I am getting there... I can see all MQTT topics in the debug log now....

Please allow me another question: How can I query a spific status value?
e.g. what message do I need to publish to request the currest state of e.g. `soyosource-gtn-display/number/soyosource-gtn-display_start_voltage/state`

Thank you very much!

syssi, 2022-08-25:
```
mosquitto_pub -t soyosource-gtn-display/number/soyosource-gtn-display_start_voltage/command 10
```

Sets the start voltage to 10 volts.

The inverter doesn't push updates. We have to retrieve the data (status & settings) periodically. You can control the update/polling interval of the `soyosource_display` component:

```
soyosource_display:
  uart_id: uart0
  update_interval: 5s
```

syssi, 2022-08-25:
> P.S: I am missing the value that the display shows the "total inverted kWh" in the upper left corner. Is there a way to readout this value as well? I could not find it in the exmaple configuration. Thanks man!

The total energy isn't a feature of the Soyosource inverter. It's handled / accumulated by the WiFi dongle. You could use the `total_daily_energy` component (https://esphome.io/components/sensor/total_daily_energy.html) to implement this feature on your own.

syssi, 2022-08-25:
Feel free to ask more questions!

careyer, 2022-08-26:
Thank you @syssi : For now everything seems to work perfectly. Now I need to build a test-rig first and might come back with some more questions. I need to figure out what the "dummy powermeter" stuff is all about... from the console I can see that it injects some funny values in that esp-display-example configuration.
In the long term I need to send the required output power to the RS485 port via MQTT.

syssi, 2022-08-26:
On a production setup you want to remove the dummy power sensor

```
  - platform: template
    name: "dummy powermeter"
    id: powermeter0
    update_interval: 2s
    lambda: |-
      return 80.0;
```

with a MQTT subscription of the instantaneous power consumption of your smartmeter. This will look like this:

```
  # mqtt subscribe example
  - id: powermeter
    internal: true
    platform: mqtt_subscribe
    name: "${name} instantaneous power consumption"
    topic: "smartmeter/sensor/groundfloor/obis/1-0:16.7.0/255/value"
    accuracy_decimals: 2
    unit_of_measurement: W
    device_class: power
    filters:
      - throttle_average: 15s
```

https://github.com/syssi/esphome-soyosource-gtn-virtual-meter/blob/main/esp32-example.yaml#L103-L112

careyer, 2022-08-30:
Awesome! Thanks mate!
Keep up the excellent work!

careyer, 2022-09-11:
Hey @syssi,

sorry for bugging you again... but I have a problem that I don't have a clue on how to solve.
My PowerSensor is sensing and sending the required power every 5s via MQTT.... However the esphome-soyosource-gtn-virtual-meter does only read a new value via MQTT every 60sec or so.... 
Is there a way to speed up the MQTT and process the incomming MQTT Messages faster?

[image]

Thanks for your help!

syssi, 2022-09-11:
Could you create a new issue per question? Since there is no good documentation, I want to make it as easy as possible for users to find details about the implementation.

careyer, 2022-09-11:
Sure! Will do yo! =)