syssi/esphome-soyosource-gtn-virtual-meter #72: MQTT Update interval

careyer, 2022-09-11:
Hey @syssi,

sorry for bugging you again... but I have a problem that I don't have a clue on how to solve.
My PowerSensor is sensing and sending the required power every 5s via MQTT.... However the esphome-soyosource-gtn-virtual-meter does only read a new value via MQTT every 60sec or so....
Is there a way to speed up the MQTT and process the incomming MQTT Messages faster?

[image]

Thanks for your help!

syssi, 2022-09-11:
Could you provide your yaml configuration? It looks like you are throttling the incoming MQTT messages. Something like this:

```
  - id: powermeter
    internal: true
    platform: mqtt_subscribe
    name: "${name} instantaneous power consumption"
    topic: "smartmeter/sensor/groundfloor/obis/1-0:16.7.0/255/value"
    accuracy_decimals: 2
    unit_of_measurement: W
    device_class: power
    filters:
      - throttle_average: 60s
```

The ESPHome node does receive all subscribed topics immediately per default. Please be careful if you play around with these parameters. You will create oscillations if you update the power demand too often.

careyer, 2022-09-11:
My yaml configuration is the same as yours. 

I tried leaving out the "filters" option altogether and also tried with various "filters:
      - throttle_average: [X]s" Settings where [X] was within: 1s, 5s, 10s, 15s 

However the current solution processes new MQTT messages only exactly every 60s. You can also see that on the log Timecode in the above screenshot. 

I would be happy if I could reduce that Intervall to about 5-10s to be more "real-time"

Cheers!

syssi, 2022-09-11:
Could you provide a plaintext log of around 10 minutes? We are talking about the updates of the `instantaneous power consumption` sensor, correct?

syssi, 2022-09-11:
In my case the `instantaneous power consumption` is published (and received by the ESPHome node) every second. Throttling the updates / incoming messages to `15s` leads to a log message every 15 seconds:

```
[20:31:45][D][sensor:125]: 'groundfloor_inverter smartmeter instantaneous power': Sending state 697.32001 W with 2 decimals of accuracy
[...]
[20:32:00][D][sensor:125]: 'groundfloor_inverter smartmeter instantaneous power': Sending state 701.06000 W with 2 decimals of accuracy
```

If you see this log message only once per minute and you don't apply any filters please double check your MQTT traffic (f.e. subscribe to the topic using mosquitto_sub manually). I assume there is just one message per minute published via MQTT.

careyer, 2022-09-11:
Hi Sebastian @syssi,

here is a 10min debug log:

```
[19:42:14][D][soyosource_virtual_meter:073]: 'unnamed': Setting the limiter to 0 watts per inverter (0 in total)
[19:42:34][D][soyosource_virtual_meter:073]: 'unnamed': Setting the limiter to 0 watts per inverter (0 in total)
==>[19:42:54][D][sensor:127]: 'soyosource-gtn-display instantaneous power consumption': Sending state -655.75031 W with 2 decimals of accuracy**
[19:42:54][D][soyosource_virtual_meter:124]: 'unnamed': Using the passthrough method: -655
[19:42:54][D][soyosource_virtual_meter:020]: 'unnamed': New calculated demand: 655 / last demand: 0
[19:42:54][D][sensor:127]: 'soyosource-gtn-display instantaneous power consumption': Sending state -655.75031 W with 2 decimals of accuracy
[19:42:54][D][soyosource_virtual_meter:124]: 'unnamed': Using the passthrough method: -655
[19:42:54][D][soyosource_virtual_meter:020]: 'unnamed': New calculated demand: 655 / last demand: 0
[19:42:54][D][soyosource_virtual_meter:073]: 'unnamed': Setting the limiter to 655 watts per inverter (655 in total)
[19:43:14][D][soyosource_virtual_meter:073]: 'unnamed': Setting the limiter to 655 watts per inverter (655 in total)
[19:43:34][D][soyosource_virtual_meter:073]: 'unnamed': Setting the limiter to 655 watts per inverter (655 in total)
==>[19:43:54][D][sensor:127]: 'soyosource-gtn-display instantaneous power consumption': Sending state -655.20691 W with 2 decimals of accuracy
[19:43:54][D][soyosource_virtual_meter:124]: 'unnamed': Using the passthrough method: -655
[19:43:54][D][soyosource_virtual_meter:020]: 'unnamed': New calculated demand: 655 / last demand: 655
[19:43:54][D][soyosource_virtual_meter:073]: 'unnamed': Setting the limiter to 655 watts per inverter (655 in total)
[19:43:54][D][sensor:127]: 'soyosource-gtn-display instantaneous power consumption': Sending state -655.20691 W with 2 decimals of accuracy
[19:43:54][D][soyosource_virtual_meter:124]: 'unnamed': Using the passthrough method: -655
[19:43:54][D][soyosource_virtual_meter:020]: 'unnamed': New calculated demand: 655 / last demand: 655
[19:44:14][D][soyosource_virtual_meter:073]: 'unnamed': Setting the limiter to 655 watts per inverter (655 in total)
[19:44:34][D][soyosource_virtual_meter:073]: 'unnamed': Setting the limiter to 655 watts per inverter (655 in total)
[19:44:54][D][soyosource_virtual_meter:073]: 'unnamed': Setting the limiter to 655 watts per inverter (655 in total)
==>[19:44:54][D][sensor:127]: 'soyosource-gtn-display instantaneous power consumption': Sending state -654.32379 W with 2 decimals of accuracy
[19:44:54][D][soyosource_virtual_meter:124]: 'unnamed': Using the passthrough method: -654
[19:44:54][D][soyosource_virtual_meter:020]: 'unnamed': New calculated demand: 654 / last demand: 655
[19:44:54][D][sensor:127]: 'soyosource-gtn-display instantaneous power consumption': Sending state -654.32379 W with 2 decimals of accuracy
[19:44:54][D][soyosource_virtual_meter:124]: 'unnamed': Using the passthrough method: -654
[19:44:54][D][soyosource_virtual_meter:020]: 'unnamed': New calculated demand: 654 / last demand: 655
[19:45:14][D][soyosource_virtual_meter:073]: 'unnamed': Setting the limiter to 654 watts per inverter (654 in total)
[19:45:34][D][soyosource_virtual_meter:073]: 'unnamed': Setting the limiter to 654 watts per inverter (654 in total)
[19:45:54][D][soyosource_virtual_meter:073]: 'unnamed': Setting the limiter to 654 watts per inverter (654 in total)
==>[19:45:54][D][sensor:127]: 'soyosource-gtn-display instantaneous power consumption': Sending state -654.81165 W with 2 decimals of accuracy
[19:45:54][D][soyosource_virtual_meter:124]: 'unnamed': Using the passthrough method: -654
[19:45:54][D][soyosource_virtual_meter:020]: 'unnamed': New calculated demand: 654 / last demand: 654
[19:45:54][D][sensor:127]: 'soyosource-gtn-display instantaneous power consumption': Sending state -654.81165 W with 2 decimals of accuracy
[19:45:54][D][soyosource_virtual_meter:124]: 'unnamed': Using the passthrough method: -654
[19:45:54][D][soyosource_virtual_meter:020]: 'unnamed': New calculated demand: 654 / last demand: 654
[19:46:14][D][soyosource_virtual_meter:073]: 'unnamed': Setting the limiter to 654 watts per inverter (654 in total)
[19:46:34][D][soyosource_virtual_meter:073]: 'unnamed': Setting the limiter to 654 watts per inverter (654 in total)
[19:46:54][D][soyosource_virtual_meter:073]: 'unnamed': Setting the limiter to 654 watts per inverter (654 in total)
==>[19:46:54][D][sensor:127]: 'soyosource-gtn-display instantaneous power consumption': Sending state -656.82721 W with 2 decimals of accuracy
[19:46:54][D][soyosource_virtual_meter:124]: 'unnamed': Using the passthrough method: -656
[19:46:54][D][soyosource_virtual_meter:020]: 'unnamed': New calculated demand: 656 / last demand: 654
[19:46:54][D][sensor:127]: 'soyosource-gtn-display instantaneous power consumption': Sending state -656.82721 W with 2 decimals of accuracy
[19:46:54][D][soyosource_virtual_meter:124]: 'unnamed': Using the passthrough method: -656
[19:46:54][D][soyosource_virtual_meter:020]: 'unnamed': New calculated demand: 656 / last demand: 654
[19:46:54][I][ota:105]: Boot seems successful, resetting boot loop counter.
[19:46:54][D][esp32.preferences:113]: Saving 1 preferences to flash...
[19:46:54][D][esp32.preferences:142]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[... cut ...]
My configuration is:
```
  # mqtt subscribe example
  - id: powermeter0
    internal: true
    platform: mqtt_subscribe
    name: "${name} instantaneous power consumption"
    topic: "instantaneous_PowerRequirement_Inverter1"
    accuracy_decimals: 2
    unit_of_measurement: W
    device_class: power
#    filters:
#      - throttle_average: 15s

```
Likewise noted: I tried with the filter setting active already and with various settings "throttle_average:[x]s" (with [x]= 1,2,5,10,15s) but to no avail
Any thoughts?

syssi, 2022-09-11:
Please double check your MQTT traffic. Please use an independent MQTT client (like mosquitto_sub) and subscribe to `instantaneous_PowerRequirement_Inverter1`. Please provide a capture of these messages. If the measurement (message) doesn't change ESPHome doesn't perform a sensor update.

The MQTT subscription component you are using is a core feature of ESPHome. It's an "on_receive" callback which triggers the log message on every received message. Which ESPHome version do you use? I cannot reproduce your issue with 2022.6.3.

syssi, 2022-09-12:
Do you have a spare ESP8266/ESP32? I would like to provide a configuration example to debug the MQTT component / traffic.

careyer, 2022-09-13:
Hey Sebastian,
sorry for not comming back to you in time.... I was busy with my job the last few days. Yes I have a spare ESP32 board at hand. Hoewever I have solved the MQTT issue. In fact my MQTT broker (Mosquitto) was somehow missbehaving. Once restarted everthing worked liked expected. So please excuse my dumb question. However this is the last thing you take into consideration.

However I have analysed the power_demand_calculation methods a bit and I think that there is room for some improvement. I don't know if you are open to talk about that? If yes I suggest we open a new "issue" to talk about it and not to muddy the waters in this thread?

syssi, 2022-09-14:
Perfect! Let's close this issue and feel free to create a new one to discuss improvements. I'm happy about your feedback & suggestions to improve this project.

closed 2022-09-14 by syssi