syssi/esphome-soyosource-gtn-virtual-meter #73: Determining the "Power calculation method"

careyer, 2022-09-11:
In the CLI output it says:  "Using the new method of calculating the power demand" (which overshoots by quite a lot). 
Is there a way to use the "standard method"?

I found out, that this is in soyosource_virtual_meter.cpp (L82-L88ff)
Here I can see there is the determination of what method to use.

However both versions do not fit my needs. 
I want to implement a "PV-first" functionality in my system and therefore I need the unaltered last MQTT power measurement. This power measurment refelcts the current PV input in Watts and I want to command my inverter to invert exaclty this power so that the PV power gets absorbed right awaybefore charging my batteries.

Thanks for any hints mate!

syssi, 2022-09-11:
There ware two methods available:

1. `DUMB_OEM_BEHAVIOR`
2. `NEGATIVE_MEASUREMENTS_REQUIRED`

You can choose between the methods here:

https://github.com/syssi/esphome-soyosource-gtn-virtual-meter/blob/main/esp8266-example.yaml#L60

I cannot recommend to use the dump OEM behavior because the methods is unable to detect overshooting.You should stick to the `NEGATIVE_MEASUREMENTS_REQUIRED` method instead and add some buffer:

https://github.com/syssi/esphome-soyosource-gtn-virtual-meter/blob/main/esp8266-example.yaml#L65-L67

Another option is smoothing the readings of the smart meter locally by applying additional filters:

https://github.com/syssi/esphome-soyosource-gtn-virtual-meter/blob/main/esp8266-example.yaml#L112-L113

careyer, 2022-09-11:
BTW: I felt lucky enough to add a 3rd method called "PASSTHROUGH" (you can observe the changes in my fork: https://github.com/careyer/esphome-soyosource-gtn-virtual-meter -- I am sorry but I had to fork because otherwise the esphome build processs does overwrite my local changes with the status of your online repo while building because it pulls the latest code during buildtime) 🙄

The use-case of this method is the following: I wanted to implement a "Solar/PV-First"-Approach... i.e.: I am measuring what power comes from my solar panels right at the very same moment and I want my SoyoSource Inverter to invert exactly this ammount of power instead of instead charging my batteries.

syssi, 2022-09-11:
Feel free to fork this repository (it's how GitHub works). ;-) I would be happy about a contribution / pull requesting if you want to introduce a third method which fits to your needs.

careyer, 2022-09-13:
Thank you! I see you are embodying the open source idea! Kudos!
Sadly I am not very good at coding... I managed to implement my passthough method meanwhile (just to discover that I had fallen short of a missconception) ¯\_(ツ)_/¯ 

However I had ideas on how to improve on your 'NEGATIVE_MEASUREMENTS_REQUIRED' method:

1. From what I can tell you need the negative input values to determine if you are in an overshoot situation (you are inverting more than the actual demand). As long as you are seeing positive values you keep adding them to the previous power demand and hence approaching the actual power demand (you are incremennting towards the power demand). You do that until you see the very first negative value and then you take a step back. Alright!!!! Good approach.... However you can improve that:
In your situation you will increment as long as you reach an overshoot and try stabilizing arround that point with the chance of hitting more overshoots over time and always stepping back from them.
--Here goes my idea on how to improve on that: Each time when receiving a new power reading you can just observe "Have I been actively inverting the last time I received a value (i.e. output power >0)?"... if so the current received value needs to be added to the power demand. If not (wasn't inverting on the last power reading) --> Inverting had been stopped due to overshoot and we need to start a new approximation cycle. Bennefit: You achieve pretty much the same buth without the need for negative values being needed/allowed (makes it more versatile). The negative prefix is basically replaced by the observation: Was I inverting before or not?

2. Use a PID  controller to approximate the needed power demand without much overshooting (if at all). I believe this might be the most sophisticed method with mimium jitter and less 'approximation cycles'. However coding a PID-controller is way over my head 

What are your thoughts?
Best regards!

syssi, 2022-09-14:
Let's talk about idea 1 in detail: Could you try to explain / outline your idea using example values?

careyer, 2022-09-18:
Hey @syssi,

sorry for answering late: I have been way to busy these days. 
Okay- here comes my proposal for another method that does not overshoot and that can also work without nagative values:

**Preliminary considerations:**
We can determine at any point in time if the inverter is currently active and inverting by taking a look of the currently inverted power (output power >0W). Now lets use that information to approximate our power demand from the lower end without ever overshooting:

Lets take a look at an example:
1.) Given our limiter signals a power demand of 400W
2.) Now start inverting at 90% that power with 400W x 0,9 = 360W
3.) Now upon the next measurement we see that there are 40W of demanded power left -- AND -- we see that our inverter is already inverting (output power 360W>0W). That is: We know we have to add more power to the currently set output power.
4.) Now we add another 90% of the demand that is left, ie:  360W + 40 x 0,9 = 396W
5.) Now upon the next measurement we see that there are 4W of demanded power left -- AND -- we see that our inverter is already inverting (output power 396W>0W). That is: We know we have to add more power to the currently set output power.
6.) Now we add another 90% of the demand that is left, ie:  396W + 4 x 0,9 = 399.6W
7.) Now lets assume our power draw suddenly drops from 400W to only 150W
8.) Out limiter power reading will signal <=0W, i.e. we stop inverting and set the new power output to 0W
9.) On the very next reading the Power limiter value will read 150W and hence we repeat approximating the needed power like before (Back to step 1. but this time with the new value of 150W)

This time we rapidly approach the current power demand without ever overshooting and without the need to negative values.

syssi, 2022-09-18:
I understand your idea. Could you explain why you don't like overshooting and why do you prefer stopping the inverter instead of slowly decrease the limiter signals to find a new "positive sweet spot" (f.e. if the smartweter reports a power consumption of 1W).

careyer, 2022-09-18:
Sure! There are two major reasons for this:
a) Every overshoot means loosing energy to the energy grid (for anyone who does not get paid for it)
b) For blackout prevention I have a offgrid-inverter at hand which I can use to simulate the grid for my Grid-Tie inverters. This works well but one has to make sure that in case of sudden powerdemand drops the overshoot is very quickly resolved - otherwise the offgrid-inverter releases its magic smoke. There is no time to slowly approach a new sweetspot. ;-)

syssi, 2022-09-18:
Last question before implementing the method: Are you able to measure negative values (overshooting/exporting to the (off)grid) in case of b) or are you looking for a "positive measurements method" because of this reason?

careyer, 2022-09-18:
Ohh... yes, I can also measure negative values. Just thought it would be beneficial to have a 3rd method that doesn't rely on this prerequisite. Just to make this whole suite of tools more universal ;-)

syssi, 2022-09-18:
Please give it a try:

```
substitutions:
  name: soyosource-gtn-virtual-meter
  device_description: "Monitor a Soyosource GTN and control the power output on demand both via RS485"
  external_components_source: github://syssi/esphome-soyosource-gtn-virtual-meter@restart-on-crossing-zero-method
  tx_pin: GPIO1
  rx_pin: GPIO3

soyosource_virtual_meter:
  power_id: powermeter
  power_sensor_inactivity_timeout: 20s
  power_demand_calculation: RESTART_ON_CROSSING_ZERO
  min_power_demand: 0
  max_power_demand: 900
  power_demand_divider: 1
  buffer: 10
  operation_mode_id: operation_mode_id
  update_interval: 3s
```

The implemention isn't exactly your approach but it's pretty similar. The configuration above tries to control the limiter to import 10W from the grid always. If your smartmeter reports a value less than 10W the method tries to reach 10W again. If your smartmeter reports a value `<= 0` we stop the inverter and start from scratch.

careyer, 2022-09-22:
Hey Sebastian, 

thanks! Sorry... I was busy again like crazy. I will try it this weekend and will report back to you!
Thanks in advance!

Best regards
Thomas

blacklopo, 2022-09-28:
Hi,
I'm also very concerned of minimising of overshooting.  I have question about calculation of power demand - If my PV has at the moment not enough production, script very quickly rises the power 900W. But generally I only need to produce what is detected by clamp. 
Is in understable? IMHO we have to compare demand with real production of the invertor.
[image]

syssi, 2022-09-28:
@blacklopo Do you have a smartmeter which provides negatives measurements of you overshoot / export to the grid?

blacklopo, 2022-09-28:
> @blacklopo Do you have a smartmeter which provides negatives measurements of you overshoot / export to the grid?

Yes, Shelly 3M. Attaching graph from today.
[image]

syssi, 2022-09-28:
Keep me updated. I will try to provide a similar graph for comparison.

blacklopo, 2022-09-29:
This is something I don't like to see:
[image]

syssi, 2022-09-29:
Could you explain what I can see here? I must admit I didn't get the meaning of FVE and DS yet.

blacklopo, 2022-09-29:
Now I can see problem with settings - RESTART_ON_CROSSING_ZERO. When negative value is detected, system automatically set Demand power to 0, and iniciate "reset", but for next several seconds the inverter still generates higher power than is expected...

blacklopo, 2022-09-29:
- DS: Power from grid. Detected by Shelly 3M
- FVE: "Production" of inverter measured by smart tuya plug.

syssi, 2023-03-08:
Could you provide your YAML configuration? I would like to know the update interval of the powermeter sensor for sure. It is throttled?

syssi, 2023-03-08:
Please try to throttle the `powermeter0` sensor step by step. At the moment your parameter set reacts too fast and the inverter cannot respond as fast. This leads to oscillation and the virtual meter is unable to close the feedback loop.

kev300, 2023-03-09:
Its just that:
`- platform: homeassistant                                                                                                                                                                                                                                                                 id: powermeter0                                                                                                                                                                                                                                                                         name: "smartmeter power consumption"                                                                                                                                                                                                                                                    entity_id: sensor.stromz_power_power_curr`

The sensor itself sends a value as soon as the meter is updating it's measurement, which is about 1s
So not throttled.

I am thinking about changing the calculation to something like
`int16_t power_demand = (importing_now - this->last_power_demand_delta_ / 2) + last_power_demand;`
to "wait" a little bit for the inverter based on the last demand change.

Throttling the sensor would slowdown the initial reaction on sudden load change. Changing the calculation method would only slow down the following adjustments.

I will try to test both methods and see if my assumption can work. ;-)

syssi, 2023-03-09:
If you clone this repository you can make/test all changes locally if you compile the YAML using:

```
esphome -s external_components_source components run esp8266-example.yaml
```

This will build the project using the local component directory instead of cloning the repository.

syssi, 2023-03-09:
I'm happy about possible improvements. My knowledge of control engineering is very limited.

ButschFPV, 2023-03-11:
@kev300 I'm running my inverter with these parameters:
```
power_demand_calculation: NEGATIVE_MEASUREMENTS_REQUIRED
buffer: 0
update_interval: 1s
#- throttle_average: 15s
```
I have no oscillation. My smartmeter updates every second. It's working okay but overshoots for 1-2 seconds are normal.
You mentioned in #102 that you have a quiet soyosource. What Version is it 24V/48V?
What error does it show if you send the power demand too frequently (1 second)?

kev300, 2023-03-12:
It's 48v 1200w
It does not show any errors, but the rs485 logo disappears for 1-2 seconds and it goes to 0w. Then it start working again for 2-3s.

Interesting. I just tested with low power consumption, because my battery is empty and not connected to solar, yet. But mine never stops oscillating. I'm trying a new algorithm at the moment but didn't have enough time yet.

ButschFPV, 2023-03-12:
I have hooked up the original RS485 limiter to a logic analyzer and it's transmitting the power demand every 0.5s.
Maybe it's the status request frame that quiet inverter don't like.

Have you bought your inverter with the original limiter?

syssi, 2023-03-12:
I made the same experiment. The periodic requests are:

```
[12:36:21][D][uart_debug:114]: <<< 24:80:00:48
[12:36:21][D][uart_debug:114]: <<< 24:80:00:48
[12:36:22][D][uart_debug:114]: <<< 24:80:00:48
[12:36:22][D][uart_debug:114]: <<< 24:80:00:48
[12:36:23][D][uart_debug:114]: <<< 24:80:00:48
[12:36:23][D][uart_debug:114]: <<< 24:80:00:48
[12:36:24][D][uart_debug:114]: <<< 24:80:00:48
[12:36:24][D][uart_debug:114]: <<< 24:80:00:48
[12:36:25][D][uart_debug:114]: <<< 24:80:00:48
[12:36:25][D][uart_debug:114]: <<< 24:80:00:48
[12:36:26][D][uart_debug:114]: <<< 24:80:00:48
[12:36:26][D][uart_debug:114]: <<< 24:80:00:48
[12:36:27][D][uart_debug:114]: <<< 24:80:00:48
[12:36:27][D][uart_debug:114]: <<< 24:80:00:48
[12:36:28][D][uart_debug:114]: <<< 24:80:00:48
[12:36:28][D][uart_debug:114]: <<< 24:80:00:48
[12:36:29][D][uart_debug:114]: <<< 24:80:00:48
[12:36:29][D][uart_debug:114]: <<< 24:80:00:48
[12:36:30][D][uart_debug:114]: <<< 24:80:00:48
``` 

This is my workbench setup:

[image]

kev300, 2023-03-12:
Oh interesting. Yes i also have the original limiter. 
So i need to remove the info stuff to be able to use 0.5s updates directly from the esp...

syssi, 2023-03-12:
What do you mean be "info stuff"? You could simply increase the update interval of the virtual limiter.

syssi, 2023-03-12:
I would be happy about another experiment to answer these questions:

1. How fast is the Soyosource able to ramp up from 0 to (f.e.) 500W?
2. How long does it take to ramp down (by sending a 0W demand)?
3. How long does it take to ramp down if no demand is transmitted?

kev300, 2023-03-12:
> What do you mean by "info stuff"? You could simply increase the update interval of the virtual limiter.
> 
> ```
> soyosource_virtual_meter:
>   - id: virtualmeter0
>     soyosource_modbus_id: modbus0
>     [...]
>     update_interval: 500ms
> ```

Sorry I meant 

> Maybe it's the status request frame that quiet inverter don't like.

so the whole information readout, which my inverter does not provide

syssi, 2023-03-12:
Yes. If your inverter is a silent one you should use the

https://github.com/syssi/esphome-soyosource-gtn-virtual-meter/blob/main/esp8266-limiter-example.yaml

example instead of the https://github.com/syssi/esphome-soyosource-gtn-virtual-meter/blob/main/esp8266-example.yaml

The `soyosource_inverter` is removed here.

ButschFPV, 2023-03-12:
@syssi was about to write the same.
The device_description needs to be updated: [https://github.com/syssi/esphome-soyosource-gtn-virtual-meter/blob/9e910338bf31d093daa06e2f71d98dcb35df17ee/esp8266-limiter-example.yaml#L3](https://github.com/syssi/esphome-soyosource-gtn-virtual-meter/blob/9e910338bf31d093daa06e2f71d98dcb35df17ee/esp8266-limiter-example.yaml#L3)

Shouldn't "Monitor an control" only be "Control"?

humpataa, 2023-03-12:
I have a similar setup running and was curious too about how fast the Soyo is adjusting. I don't have decent measurement equipment, just a Shelly. But I am sure there is quite a bit of latency of one or even two seconds until a new limit takes effect. And it looks like a smooth ramp between highly different values, no sudden impacts. I have disabled reading from the limiter "port" because this makes no sense in my eyes and for many people the inverter is silent anyway. After testing I have found that "hammering" the limiter twice a second is not necessary at all. It works very good when updating once every 3 seconds - and taking into account the inverter's latency in adjusting, 2 seconds seems like a very good setting. Above 4 seconds is certainly to big of delay. Just my two cents. Keep up the good work guys!

kev300, 2023-03-12:
Ok, i did a few tests.
1. My inverter works now with 0.5s updates if the information part is removed from the config.

But it's not necessary. ;-)
In my case there is a latency of about 2,5 seconds when the power meter sensor gets to 0W after the inverter demand has been set.
Optimal for me is 1s update for the smart meter and 3s for the inverter.

In addition to that, I can stop the overshoot, if skip the second demand setting after the initial one.
[image]

The fluctuations you see happens, when I turn on or off the lights and it takes a few seconds until it reacts but then its very flat without overshooting.

kev300, 2023-03-15:
This seems to work good for me: https://github.com/kev300/esphome-soyosource-gtn-virtual-meter/tree/overshoot-compensation2

But I'm still testing with fast (0.5s) and slow (3s) update interval today and will create a PR if don't find more issues. ;-)

pfeupfeu, 2023-03-16:
I also like to see the inverter reacting as quick as possible.
The original limiter outputs the demand in 0.5s intervals, but must also take the total latency into account. How does it do that?
Is there an overshoot with the original limiter?

I think triggering the uptdate by a new power sensor reading, is an good idea, but how can I do this?

kev300, 2023-03-16:
> I think triggering the uptdate by a new power sensor reading, is an good idea, but how can I do this?

Just set update_interval of soyosource_virtual_meter to never (and use my branch here: https://github.com/kev300/esphome-soyosource-gtn-virtual-meter/tree/overshoot-compensation2)

Or instead of using my branch also add this to your powermeter sensor:

> on_value:
  then:
    -component.update:virtualmeter0

virtualmeter0 is the id of the soyosource_virtual_meter

kev300, 2023-03-16:
But keep in mind:
If your power sensor reading is inconsistent, the RS485 protocol will fail and reset the inverter to 0W occasionally.
It was the case for me, so I had to go back to update_intervall: 500ms which is reliable.

pfeupfeu, 2023-03-16:
Thanks, is working!

At first I got this Error:
```
  on_value:
    then:

    [-component.update] is an invalid option for [on_value]. Please check the indentation.
    -component.update: virtualmeter0
```
So I put the update command at the end of the Shelly 3EM query:

```
time:
  - platform: sntp
    id: sntp_time
    on_time:
      - seconds: /1
        then:
          - http_request.get:
              url: ${shelly_3em_url}
              headers:
                Content-Type: application/json
              verify_ssl: false
              on_response:
                then:
                  - lambda: |-
                      json::parse_json(id(http_request_data).get_string(), [](JsonObject root) {
                        id(powermeter).publish_state(root["total_power"]);
                      });
          - component.update: virtualmeter0
```
Now it is working, with 1s repetition rate and without overshoot. I'm using syssis unchanged software.

kev300, 2023-03-16:
Cool, seems like your shelly and directly querying it is much faster than my smartmeter setup through home assistant. 😉

pfeupfeu, 2023-03-17:
Yersterday I told you, that I would'nt have overshoots. Ok, that was true, but this morning I found one of my 3 inverters was not working yersterday.
With the third  inverter working I get little overshoots, independet from the update interval.

Operation with one inverter too few yesterday was similar to Kevin's `magic_demand_delta` set to 0,67.
--> Ok, we need the demand delta factor for better control characteristics.

kev300, 2023-03-20:
Yes, Kp fits. 😁 
By the way, I found this right now:
https://esphome.io/components/climate/pid.html

Maybe we can use that and just replace °C by Watts (Adapt our sensor + demand control). Not sure if thats realy possible with that component. But I would only go there if my simplified solution does not work good enough.

pfeupfeu, 2023-05-09:
Wy don't you use the sum of the 3 phases provided by the 3EM http status page? 
My code that I now use has changed since March, to have a finer resolution for the repetition of the 3EM query:
```
interval:
  - interval: 900ms
    then:
      - http_request.get: 
         url: ${shelly_3em_url}
         headers:
            Content-Type: application/json
         verify_ssl: false
         on_response:
            then:
              - lambda: |-
                  std::string response_data = id(http_request_data).get_string();
                  if(!response_data.empty()) {
                  json::parse_json(response_data, [](JsonObject root) {
                  id(powermeter).publish_state(root["total_power"]);
                  });
                  }
              - component.update: virtualmeter0
```
And at 
```
soyosource_virtual_meter: 
  - id: virtualmeter0
[...]
    update_interval: never
```

kev300, 2023-05-10:
@pfeupfeu : do you have graphs with 1s resolution of you power consumption meter? I can't belive that you don't have overshoots or oscillation with the shelly and 900ms (btw. why 900?).
😬 
The shelly must be very quick to react. But the Soyo also takes 0.5-1s to react to demand changes...

Chickenbreast0, 2023-05-10:
> Wy don't you use the sum of the 3 phases provided by the 3EM http status page? My code that I now use has changed since March, to have a finer resolution for the repetition of the 3EM query:
>
> ``` [...]

Thanks for getting back to me.
The reason is simply that I currently lack the time to do further research and I was previously under the impression that there might be a hardware defect. Also, I'm new to this topic - that should answer your question sufficiently ;)

Maybe you can help me again and tell me where my error is so that your code does not work for me?

[image]

[image]

Chickenbreast0, 2023-05-10:
This is my entire code. No matter what I try, it would be appreciated if you can share me yours. So I can find where my mistake is.

[image]

```
substitutions:
  name: epever-soyosource

esphome:
  name: wt32-eth01
  friendly_name: WT32
  platformio_options:
    build_flags:
      - -DCONFIG_ARDUINO_LOOP_STACK_SIZE=8192

esp32:
  board: esp32dev
  framework:
    type: arduino

external_components:
  - source: github://syssi/esphome-soyosource-gtn-virtual-meter@main
    refresh: 0s

ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO0_IN
  phy_addr: 1
  power_pin: GPIO16

# Optional manual IP
  manual_ip:
    static_ip: 192.168.178.9
    gateway: 192.168.178.1
    subnet: 255.255.255.0

# Enable logging
logger: 
  level: DEBUG
  baud_rate: 0

# Enable Home Assistant API
api:
  reboot_timeout: 0s
  encryption:
    key: "blanked out"

ota:
  password: "blanked out"

uart:
  - id: uart_0
    tx_pin: GPIO1 #TXD0 
    rx_pin: GPIO3 #RXD0
    baud_rate: 115200
    stop_bits: 1

  - id: uart_1
    baud_rate: 4800
    tx_pin: GPIO17 #TXD
    rx_pin: GPIO5 #RXD
    debug:
      direction: BOTH

modbus:
  - id: modbus_epever
    uart_id: uart_0
    send_wait_time: 200ms
[... 417 lines cut ...]
    modbus_controller_id: epever
    name: "EPEVER Battery Temperature Status"
    address: 0x3200
    register_type: read
    # bitmask: 56
    raw_encode: HEXBYTES
    lambda: |-
      uint16_t value = modbus_controller::word_from_hex_str(x, 0);
      value = modbus_controller::mask_and_shift_by_rightbit(value, 56);
      switch (value) {
        case 0: return std::string("Normal");
        case 1: return std::string("Overtemperature");
        case 2: return std::string("Undertemperature");
      }
      return str_snprintf("Unknown (0x%04X)", 17, value);
      
  - platform: soyosource_virtual_meter
    soyosource_virtual_meter_id: virtualmeter0
    operation_mode:
      name: "soyo limiter operation mode"

  - platform: soyosource_inverter
    soyosource_inverter_id: inverter0
    operation_status:
      name: "soyo operation status"

pfeupfeu, 2023-05-11:
I'm sorry, I forgot you need also this:

```
http_request:
  id: http_request_data
  useragent: esphome/device
  timeout: 500ms
```

For reference, and in case if I forgot something else, this is my entire code:

```
substitutions:
  name: soyo-inverter
  device_description: "Monitor and control the Soyosource GTN via the display port and control the power output on demand via RS485"
  external_components_source: github://syssi/esphome-soyosource-gtn-virtual-meter@main
  shelly_3em_url: http://192.168.188.71/status/
  tx_pin_ttl_wifi: GPIO4  #d2
  rx_pin_ttl_wifi: GPIO5  #d1
  tx_pin_rs485: GPIO14  #d5
  rx_pin_rs485: GPIO12  #d6
  
     
esphome:
  name: ${name}
  comment: ${device_description}
  project:
    name: "syssi.esphome-soyosource-gtn-virtual-meter"
    version: 2.1.0
    
esp8266:
  board: d1_mini

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi: 
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 192.168.188.44
    gateway: 192.168.188.1
    subnet: 255.255.255.0
    dns1: 1.1.1.1

  ap:
    ssid: "Fallback-AP-soyo"
    password: ""  
    
captive_portal:

web_server:
  local: true
  version: 2

ota:

api:
  reboot_timeout: 3d

logger:
  level: INFO

http_request:
  id: http_request_data
  useragent: esphome/device
  timeout: 500ms
  
uart:
[... 252 lines cut ...]
    errors:
      name: "Soyo errors"
    operation_mode:
      name: "Soyo1 Op mode reported"
    operation_status:
      name: "Op status"

  - interval: 900ms
    then:
      - http_request.get: 
         url: ${shelly_3em_url}
         headers:
            Content-Type: application/json
         verify_ssl: false
         on_response:
            then:
              - lambda: |-
                  std::string response_data = id(http_request_data).get_string();
                  if(!response_data.empty()) {
                  json::parse_json(response_data, [](JsonObject root) {
                  id(powermeter).publish_state(root["total_power"]);
                  });
                  }
              - component.update: virtualmeter0
```

kev300, 2023-07-28:
Are you guys all on vacation? :D Or nobody has overshoots anymore?
I'm just wondering why it's soo quiet here. 😄

syssi, 2023-07-28:
Vacation. ;-)

Chickenbreast0, 2023-07-28:
> Are you guys all on vacation? :D Or nobody has overshoots anymore? I'm just wondering why it's soo quiet here. 😄

Currently am busy with my job and I just have some little time windows, in which I need to take care of more important stuff ;)
But how do we say in german language: "Aufgeschoben ist nicht aufgehoben!"(Postponed is not abandoned!)

mb-software, 2023-08-21:
Hi,
I am also trying to optimize the zero feed controller with this component.
For easier handling of modifications and quick testing of changes I am following a different approach: I set the component to manual demand mode and calculate/set the demand from within a script/action in the EspHome YAML code. This way I can modify, upload and test the controller algorithm simply through the esphome webinterface,

Anyway, currently I have replicated the algorithm of the original component using my method and it works ok (but not perfect) with a 3s interval action.
Now I want to improve it, maybe using a similar approach like @kev300 , though I have to admit that from looking into the code, I don't completely understand how the demand compensation works. Maybe a short prosa description would help here. 
Another approach that I am thinking of could be using the actual output power sensor that I am getting from the soyosource display component. But therefore I would need a short update intervall of the display component, I have no idea how low one can go there. Any experience with that?

kev300, 2023-08-21:
Isn't the original behavior already included in this component by syssi?

Did you try to run the branch from my pull request?
I tested it successfully down to 0.2s update interval with my inverter, but each feedback is welcome. 😄

My solution does react immediately on each power demand change but it subtracts the last demand change from the current calculation, because that change has not reach the powermeter, yet. It is subtracted in each update until the targeted zero consumption is reached or crossed.
In case it is not reached in time, there is a fallback timeout, which resets the calculation. This timeout is configurable in the yaml (default 5s)

The 5 s works good if your powermeter updates each 1-2s.
You need to increase it, if it is slower.

syssi, 2024-08-12:
Let's close this issue. :-) Feel free to create a new one if you want to improve a specific detail!

closed 2024-08-12 by syssi