syssi/esphome-soyosource-gtn-virtual-meter #80: ESP8266

dowermyr, 2022-09-27:
I get this message when trying to flash:

INFO Reading configuration /config/esphome/inverter-l1.yaml...
INFO Updating https://github.com/syssi/esphome-soyosource-gtn-virtual-meter.git@main
Failed config

soyosource_virtual_meter: [source /config/esphome/inverter-l1.yaml:54]
  - 
    Couldn't find ID 'powermeter'. Please check you have defined an ID with that name in your configuration.
    power_id: powermeter
    power_sensor_inactivity_timeout: 20s
    power_demand_calculation: NEGATIVE_MEASUREMENTS_REQUIRED
    min_power_demand: 0
    max_power_demand: 900
    power_demand_divider: 1
    buffer: 10
    operation_mode_id: operation_mode_id
    update_interval: 3s
    address: 0x24

syssi, 2022-09-27:
Could you provide your `inverter-l1.yaml`? It looks like there is no sensor with a `powermeter` identifier anymore.

dowermyr, 2022-09-27:
Ah, thats true! I removed the mqqt and activated api. But must have made something wrong.
Do you have an example with api instead of mqqt ?

syssi, 2022-09-27:
You can subscribe HA entities using the `homeassistant` sensor component:

```
sensor:
 - platform: homeassistant
   id: powermeter
   name: "${name} smartmeter instantaneous power"
   entity_id: sensor.firstfloor_smartmeter_instantaneous_power
   filters:
     - throttle_average: 15s
```

dowermyr, 2022-09-27:
of course, missed those lines. Thanks!

syssi, 2022-09-27:
Feel free to close the issue if it's solved.