syssi/esphome-soyosource-gtn-virtual-meter #81: No communication

dowermyr, 2022-09-28:
Hi!
Everything is now set up but I don't get any coomunication from the inverter.
Only the TX diod is blinking. This is my setup:

`substitutions:
  name: soyosource-l3
  device_description: "Monitor a Soyosource GTN and control the power output on demand both via RS485"
  external_components_source: github://syssi/esphome-soyosource-gtn-virtual-meter@main
  tx_pin: GPIO1
  rx_pin: GPIO3

esphome:
  name: ${name}
  comment: ${device_description}
  project:
    name: "syssi.esphome-soyosource-gtn-virtual-meter"
    version: 1.1.0

esp8266:
  board: nodemcuv2

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:

# If you use Home Assistant please remove this `mqtt` section and uncomment the `api` component!
#mqtt:
#  broker: !secret mqtt_host
#  username: !secret mqtt_username
#  password: !secret mqtt_password
#  id: mqtt_client

logger:
  baud_rate: 0

uart:
  baud_rate: 4800
  tx_pin: ${tx_pin}
  rx_pin: ${rx_pin}

soyosource_modbus:
  # Optional settings
  #
  # The name is used as prefix for some log messages and should
  # help to distinguish between different instances/devices
  name: inverter_l3

soyosource_inverter:

soyosource_virtual_meter:
  # The state of this sensor (instantaneous power in watt) is used as source
  power_id: powermeter

  # Optional settings
  power_sensor_inactivity_timeout: 20s
  power_demand_calculation: NEGATIVE_MEASUREMENTS_REQUIRED
  min_power_demand: 0
  max_power_demand: 900
  # Split/distribute the power demand if you have multiple inverters attached to the same RS485 bus
  power_demand_divider: 1
  # A positive buffer value (10) tries to avoid exporting power to the grid (demand - 10 watts)
  # A negative buffer value (-10) exports power to the grid (demand + 10 watts)
  buffer: 10
  # The operation_mode_id sensor is expected here. Passing the operation_mode won't work
  # The state is used to suspend the limiter if the operation mode of the inverter isn't 0x0 (normal)
  operation_mode_id: operation_mode_id

  # The update interval is important and defaults to 3 seconds. If the demand is sent too frequently
  # or rarely the interverter stops. TODO: Identify and validate the lower and upper update limit
  update_interval: 3s

binary_sensor:
  - platform: soyosource_inverter
    fan_running:
      name: "${name} fan running"

sensor:
  - platform: soyosource_virtual_meter
    power_demand:
      name: "${name} power demand"

  - platform: soyosource_inverter
    operation_mode_id:
      name: "${name} operation mode id"
      id: operation_mode_id
    battery_voltage:
      name: "${name} battery voltage"
    battery_current:
      name: "${name} battery current"
    battery_power:
      name: "${name} battery power"
    ac_voltage:
      name: "${name} ac voltage"
    ac_frequency:
      name: "${name} ac frequency"
    temperature:
      name: "${name} temperature"

  # mqtt subscribe example
 # - id: powermeter
 #   internal: true
 #   platform: mqtt_subscribe
 #   name: "${name} instantaneous power consumption"
 #   topic: "smartmeter/sensor/groundfloor/obis/1-0:16.7.0/255/value"
 #   accuracy_decimals: 2
 #   unit_of_measurement: W
 #   device_class: power
 #   filters:
 #     - throttle_average: 15s

#  # import smartmeter reading from homeassistant
#  # requires the "api" component see above
  - platform: homeassistant
    id: powermeter
    name: "${name} smartmeter"
    entity_id: sensor.smartmeter_l3
    filters:
      - throttle_average: 10s

text_sensor:
  - platform: soyosource_inverter
    operation_mode:
      name: "${name} operation mode"
  - platform: soyosource_virtual_meter
    operation_mode:
      name: "${name} operation mode"

number:
  - platform: soyosource_virtual_meter
    manual_power_demand:
      name: "${name} manual power demand"
    max_power_demand:
      name: "${name} max power demand"
      initial_value: 600
      restore_value: true

switch:
  - platform: soyosource_virtual_meter
    manual_mode:
      name: "${name} manual mode"
      restore_mode: RESTORE_DEFAULT_ON
    emergency_power_off:
      name: "${name} emergency power off"
      restore_mode: RESTORE_DEFAULT_OFF
# Enable Home Assistant API
api:
    `

dowermyr, 2022-09-28:
I also get this in the log:

[12:41:55][W][soyosource_inverter:113]: 'inverter_l3': The inverter didn't respond to the last 15 status requests

syssi, 2022-09-28:
There are some Soyosource inverters which doesn't respond to requests. Do you own the WiFi version with the pink mainboard?

syssi, 2022-09-28:
See https://github.com/syssi/esphome-soyosource-gtn-virtual-meter/issues/48

dowermyr, 2022-09-28:
Yes, I have the model without the LCD-screen and with the USB-wifi dongle

dowermyr, 2022-09-28:
That thread is mostly in German. Is there a way to fix this? I have the ct-clamps and the controller to them. Maybe I can fake the CT?. Or should i go with a different setup?

syssi, 2022-09-28:
The virtual limiter should/will work nevertheless (TX-only). Your inverter model is just unable to respond to the limiter request.

dowermyr, 2022-09-28:
Ok. Thats too bad. I wanted to read out some things

syssi, 2022-09-28:
In your case I recommend this setup:

1. Replace the WiFi dongle with an alternative solution f.e. an ESP8266 + a level shifter because of the 5V TTL: https://de.aliexpress.com/item/32845872410.html
2. Flash this example configuration to the WiFi dongle replacement: https://github.com/syssi/esphome-soyosource-gtn-virtual-meter/blob/main/esp8266-display-example.yaml
3. You are now able to monitor the inverter and change settings using the display port of the inverter.
4. Connect a RS485 module to the WiFi dongle replacement and attach the RS485 module to the CT-Clamp-RS485-Port.

syssi, 2022-09-28:
In the past the Soyosource firmware did provide a status frame (voltage + current + power) via RS485. The WiFi-Dongle-Version of the inverter doesn't respond to any RS485 requests anymore but the RS485 limiter request is processed nevertheless.

dowermyr, 2022-09-28:
I have some esp8266 and some rs485-ttl cards. But where is the display port? Do you meen the USB-port?

syssi, 2022-09-28:
Yes. The USB port isn't an USB bus. It provides power (5V, GND) + a serial line (RX, TX).

syssi, 2022-09-28:
Please keep in mind: The logic level of the serial line (on the USB port) is 5V. There is a risk if you connect the GPIOs of an ESP8266 directly.

dowermyr, 2022-09-28:
Ok. But Then I can use my nodemcu 8266 and the modbus-ttl board with an adapter to usb? Maybe I don't need to connect the 5V pin in the USB? 

And can't I control the power output through that? Do I really need another rs485 with the CT-controller?

syssi, 2022-09-28:
1. Let's assume you don't power your ESP8266 via the inverter (5V of the USB port): The voltage level of the TX signal (from the inverter) is still 5V. This voltage can harm the ESP.
2. You can control/change the "max power output setting" of the inverter using the display port (protocol) but I cannot recommend the change this settings very often. This will wear out a internal flash of the inverter.

syssi, 2022-09-28:
Please feel free to ask more questions. It looks like all details aren't clear yet.

dowermyr, 2022-09-28:
Hmm. This way is very complicated. I will see how I proceed. One other problem is that the app that works with the wifi dongle doesn't work on my Andoid. It keeps crashing. So right now I can't read out any settings

dowermyr, 2022-09-28:
What do you think about this?
https://www.amazon.se/gp/product/B08T17F9QT/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&th=1

Together with a esp8266

syssi, 2022-09-28:
I think you didn't understand one important detail: The USB connector of the Soyosource looks like an USB port but it isn't. The 4 pins of the connector exposes GND, VCC, TX and RX (UART-TTL). There is no D+/D- of an USB bus. IMO you don't need to look for real USB devices. ;-)

syssi, 2022-09-28:
This is the schematics:

[image]

1. Use a ESP8266
2. Connect a level shifter to shift the logic level from the ESP (3.3V) to the logic level of the inverter (5V)
3. Cut off a USB connector and connect the 4 wires to the level shifter (GND, VCC, RX, TX)

Instead of using a level shifter you could try to use a voltage divider (resistors).

dowermyr, 2022-09-28:
Ok, thanks! I will get some level shifters! ( Got 3 inverters)

syssi, 2022-09-28:
The schematics above is the replacement for the WiFi dongle only. If you want to control the limiter you have to / could assign the RS485 module to the same ESP. Monitoring & driving 3 inverters with a single ESP8266 will be tricky because you need 6 UART interfaces in total: 3x Display + 3x RS485.

syssi, 2022-09-28:
I prefer this ESP board because the level shifter is part of the PCB: https://de.aliexpress.com/item/32845872410.html

dowermyr, 2022-09-28:
Ok. I got a bunch of nodemcu esp8266 already though

dowermyr, 2022-10-01:
Hi! How do I control a RS485 on the same esp8266 ? My nodemcu only has one tx/rx pin?

syssi, 2022-10-01:
ESPHome uses a SoftwareSerial implementation if you assign the UART component to any other GPIO (which doesn't support hardware UART).

dowermyr, 2022-10-01:
Hi! Do you have a picture or something of the USB connector, so I know what pin is what on the USB side

syssi, 2022-10-01:
https://github.com/syssi/esphome-soyosource-gtn-virtual-meter/blob/main/docs/wifi-dongle/README.md#usb-connector

Please use a multimeter and measure the voltage at the outer pins. If the voltage is positive you have identified to orientation properly.

dowermyr, 2022-10-03:
> This is the schematics:
>
> [image]
>
> 1. Use a ESP8266
> 2. Connect a level shifter to shift the logic level from the ESP (3.3V) to the logic level of the inverter (5V)
> 3. Cut off a USB connector and [...]

Hi! could i skip the 5V and gnd in the leveshifter and attach to the Vin 5v pin ang gnd directly to the esp? It should work right? Bacause I have done that and I dont get any readings...

syssi, 2022-10-03:
Could you provide a photo of your setup?

dowermyr, 2022-10-03:
[image]
![20221003_125209](https
[image]
://user-images.githubusercontent.com/69361817/193561049-d4434574-6d35-438d-9a9e-67dbddd39e8f.jpg)

I changed to GPIO13 and 15 as I read that nodemcu preferably uses them?

dowermyr, 2022-10-03:
[image]

dowermyr, 2022-10-03:
substitutions:
  name: "soyosource-l3"
  device_description: "Monitor and control the WiFi version of the Soyosource GTN via the TTL-WIFI port and control the power output on demand via RS485"
  external_components_source: github://syssi/esphome-soyosource-gtn-virtual-meter@main
  tx_pin_ttl_wifi: GPIO15
  rx_pin_ttl_wifi: GPIO13
  tx_pin_rs485: GPIO1
  rx_pin_rs485: GPIO3

esphome:
  name: ${name}
  comment: ${device_description}
  project:
    name: "syssi.esphome-soyosource-gtn-virtual-meter"
    version: 1.1.0

esp8266:
  board: nodemcuv2

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:

# If you use Home Assistant please remove this `mqtt` section and uncomment the `api` component!
#mqtt:
#  broker: !secret mqtt_host
#  username: !secret mqtt_username
#  password: !secret mqtt_password
#  id: mqtt_client

api:

logger:
  level: INFO

# Please be careful: The display port has a logic level of 5V.
uart:
  - id: uart0
    baud_rate: 9600
    tx_pin: ${tx_pin_ttl_wifi}
    rx_pin: ${rx_pin_ttl_wifi}

  - id: uart1
    baud_rate: 4800
    tx_pin: ${tx_pin_rs485}
    rx_pin: ${rx_pin_rs485}

soyosource_modbus:
  - id: modbus0
    uart_id: uart1

soyosource_virtual_meter:
  - id: virtualmeter0
    soyosource_modbus_id: modbus0
    update_interval: 3s

    # the state of this sensor (instantaneous power in watt) is used as source
    power_id: powermeter0
    power_sensor_inactivity_timeout: 20s
    power_demand_calculation: NEGATIVE_MEASUREMENTS_REQUIRED
    min_power_demand: 0
    max_power_demand: 900
    # A positive buffer value (10) tries to avoid exporting power to the grid (demand - 10 watts)
    # A negative buffer value (-10) exports power to the grid (demand + 10 watts)
    buffer: 10

soyosource_display:
  uart_id: uart0
  update_interval: 5s

binary_sensor:
  - platform: soyosource_display
    fan_running:
      name: "${name} fan running"
    limiter_connected:
      name: "${name} limiter connected"

button:
  - platform: soyosource_display
    restart:
      name: "${name} restart"

number:
  - platform: soyosource_display
    start_voltage:
      name: "${name} start voltage"
    shutdown_voltage:
      name: "${name} shutdown voltage"
    # Maximum output power in limiter mode / Output power in constant power mode
    output_power_limit:
      name: "${name} output power limit"
    start_delay:
      name: "${name} start delay"

select:
  - platform: soyosource_display
    operation_mode:
      name: "${name} operation mode"
      optionsmap:
        1: "PV"
        2: "Battery Constant Power"
        17: "PV Limit"
        18: "Battery Limit"

sensor:
  - platform: template
    name: "dummy powermeter"
    id: powermeter0
    update_interval: 2s
    lambda: |-
      return 80.0;

  - platform: soyosource_display
    error_bitmask:
      name: "${name} error bitmask"
    operation_mode_id:
      name: "${name} operation mode id"
    operation_status_id:
      name: "${name} operation status id"
    battery_voltage:
      name: "${name} battery voltage"
    battery_current:
      name: "${name} battery current"
    battery_power:
      name: "${name} battery power"
    ac_voltage:
      name: "${name} ac voltage"
    ac_frequency:
      name: "${name} ac frequency"
    temperature:
      name: "${name} temperature"
    output_power:
      name: "${name} output power"

text_sensor:
  - platform: soyosource_display
    errors:
      name: "${name} errors"
    operation_mode:
      name: "${name} operation mode"
    operation_status:
      name: "${name} operation status"

syssi, 2022-10-03:
Could you provide a link to your level shifter? I would like to make sure it's uni- or bidirectional.

dowermyr, 2022-10-03:
https://www.amazon.se/gp/product/B07HB4MLZY/ref=ppx_yo_dt_b_asin_image_o00_s01?ie=UTF8&psc=1

syssi, 2022-10-03:
The level shifter looks good (is birectional). You can find some important details here: https://learn.sparkfun.com/tutorials/bi-directional-logic-level-converter-hookup-guide/all

IMO it's just important to power the level shifter 2 times / from both sides (`LV` = low voltage = 3.3V and `HV` = high voltage = 5V). I would power the complete setup (ESP) using the 5V pin of the inverter. 

```
USB con       NodeMCU         Level Shifter

5V <-------> VIN ESP <---------> HV   LV <----> 3.3V pin of the ESP
TX <---------------------------> HV1  LV1 <---> RX pin
RX <---------------------------> HV2  LV2 <---> TX pin
GND <------> GND ESP <---------> GND  GND <---> GND
```

syssi, 2022-10-03:
If it doesn't work out of the box please swap RX/TX once:

```
  tx_pin_ttl_wifi: GPIO13
  rx_pin_ttl_wifi: GPIO15
# vs.
  tx_pin_ttl_wifi: GPIO13
  rx_pin_ttl_wifi: GPIO15
```

dowermyr, 2022-10-03:
[14:05:01][W][uart.arduino_esp8266:127]:   You're using the same serial port for logging and the UART component. Please disable logging over the serial port by setting logger->baud_rate to 0.

syssi, 2022-10-03:
The logger component uses GPIO1/GPIO3 per default. You have attached the `uart` component to this GPIOs too. Please disable the serial output of the logger by:

```
logger:
  level: INFO
  baud_rate: 0
```

dowermyr, 2022-10-03:
That didn't help either

dowermyr, 2022-10-03:
Is this baudrate correct?
  - id: uart0
    baud_rate: 9600
    tx_pin: ${tx_pin_ttl_wifi}
    rx_pin: ${rx_pin_ttl_wifi}

dowermyr, 2022-10-03:
Noticed I got a powerdrop from the USB 5V on the inverter. So i will hook up a external 5V powersupply and check again

syssi, 2022-10-03:
The display port baudrate is correct (9600 baud). The voltage drop is strange because the WiFi dongle is equipped with an AMS (5v to 3.3V voltage regulator) and an ESP8266 too.

dowermyr, 2022-10-03:
Maybe bad connection when probing. But now I've tried external 5v power. No difference. Can't figure out what's wrong

syssi, 2022-10-03:
Some more ideas:

1. Make sure the OEM WiFi dongle is able to talk to the inverter. If this doesn't work please check the cable between the USB socket and the mainboard. There should be a white plug.
2. Pick another ESP8266 and connect `GPIO14` and `GPIO12` directly (without level shifter) to RX/TX of the display port. The ESP should survive the 5V level just for testing. Does this simple setup work?
3. Enable the debug output of the `uart` component to make all serial traffic visible at the log. The log can be consumed via WiFi even if you set the `baud_rate` of the serial output to `0`.

dowermyr, 2022-10-03:
The run light on the oem dongle is blinking green

syssi, 2022-10-03:
Green blinking means it receives periodic data from the inverter. Thats good!

dowermyr, 2022-10-03:
Here is a new log
[logs_inverter-l3_logs (1).txt](https://github.com/syssi/esphome-soyosource-gtn-virtual-meter/files/9699086/logs_inverter-l3_logs.1.txt)

syssi, 2022-10-03:
I've prepared a simplified version of the configuration yaml to get the display port communication running:

```
substitutions:
  name: soyosource-l3
  device_description: "Monitor and control the WiFi version of the Soyosource GTN via the TTL-WIFI port and control the power output on demand via RS485"
  external_components_source: github://syssi/esphome-soyosource-gtn-virtual-meter@main
  tx_pin_ttl_wifi: GPIO14
  rx_pin_ttl_wifi: GPIO12

esphome:
  name: ${name}
  comment: ${device_description}
  project:
    name: "syssi.esphome-soyosource-gtn-virtual-meter"
    version: 1.1.0

esp8266:
  board: nodemcuv2

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:
api:

logger:
  level: DEBUG
  logs:
    api.service: WARN
    ota: WARN
    sensor: DEBUG

# Please be careful: The display port has a logic level of 5V.
uart:
  - id: uart0
    baud_rate: 9600
    tx_pin: ${tx_pin_ttl_wifi}
    rx_pin: ${rx_pin_ttl_wifi}
    debug:
      direction: BOTH

soyosource_display:
  uart_id: uart0
  update_interval: 5s

binary_sensor:
  - platform: soyosource_display
    fan_running:
      name: "${name} fan running"
    limiter_connected:
      name: "${name} limiter connected"

sensor:
  - platform: soyosource_display
    error_bitmask:
      name: "${name} error bitmask"
    operation_mode_id:
      name: "${name} operation mode id"
    operation_status_id:
      name: "${name} operation status id"
    battery_voltage:
      name: "${name} battery voltage"
    battery_current:
      name: "${name} battery current"
    battery_power:
      name: "${name} battery power"
    ac_voltage:
      name: "${name} ac voltage"
    ac_frequency:
      name: "${name} ac frequency"
    temperature:
      name: "${name} temperature"
    output_power:
      name: "${name} output power"
```

Please attach RX (`GPIO12`), TX (`GPIO14`) and GND directly to your display port. These lines are indicating outgoing traffic:

```
[18:06:45][D][uart_debug:114]: >>> 55:01:00:00:00:00:00:00:00:00:00:FE
```

We are looking for incoming traffic: 

```
[18:06:45][D][uart_debug:114]: <<< ....
```

If you don't see incoming traffic please swap RX and TX another time and try again. You could swap the wires or just change the YAML (+ compile & flash again).

dowermyr, 2022-10-03:
Woohoo! Got it working now. Swaped the RX and TX in the yaml. Don't know why it didn't work before?
Now I need to get it running together with RS485.

Thinking of soldering every wire because I don't trust the jumper cables. And maybe istall the level shifter??
[logs_test_logs.txt](https://github.com/syssi/esphome-soyosource-gtn-virtual-meter/files/9699586/logs_test_logs.txt)

dowermyr, 2022-10-03:
Do you think I should use GPIO1 and 3 for the RS485 or go with some others?

syssi, 2022-10-03:
Please use the hardware UART (GPIO1/3) for the display port connection because of the higher baudrate. SoftwareSerial works better on low baudrates.

dowermyr, 2022-10-03:
Ok. But then I need to disable the logger, right ?

syssi, 2022-10-03:
Correct. You can still consume/view log messages via WiFi.

dowermyr, 2022-10-03:
The problem was the level shifter. It doesn't work with them so running without for now. Maybe order different ones. But one question : The inverter says that the limiter is connected so it can sense the rs485. But nothing happens when I change the value. Is it because I haven't connected anything on the DC side yet? Thought it would change the power output anyways?
[image]

syssi, 2022-10-04:
You have to attach a battery first. The operation mode must change from "Standby" to "Normal".

closed 2023-01-24 by syssi