syssi/esphome-soyosource-gtn-virtual-meter #82: Value of produced Power is not exported

blacklopo, 2022-09-28:
Hi!

I checked your code of soyosource_display.cpp and I have a tip for you.

  // 11    2   0x00 0xCA

Line 209: ESP_LOGD(TAG, "Unknown (raw): %02X %02X", data[11], data[12]);

According Soyo web application this atribute is the generated Power:

      $("#dcdy").html(parseInt(sje.substr(8,4),16)/10 +" V");
      $("#dcdl").html(parseInt(sje.substr(12,4),16)/10 +" A");
        $("#acdy").html(parseInt(sje.substr(16,4),16) +" V");
        $("#pl").html(parseInt(sje.substr(20,2),16)+" HZ");
        **$("#gl").html(parseInt(sje.substr(22,4),16) +" W");**
         $("#jlfdl").html(parseInt(sje.substr(26,4),16)/10 +" KWH");
        $("#wd").html(parseInt(sje.substr(30,2),16)+" ℃");
        $("#fdzt").html(zt1);
        $("#gzms").html(gzms);
        $("#cwts").html(cwtsa);

I'm attaching source code for inspiration.
Paul

PS: This is my first try in Github. I didn't know where to put it.

[soyo-mqtt-redacted.txt](https://github.com/syssi/esphome-soyosource-gtn-virtual-meter/files/9668050/soyo-mqtt-redacted.txt)

syssi, 2022-09-28:
Good catch! I will extend the component to expose the identified value as sensor measurement.

blacklopo, 2022-09-28:
I updated repository (twice), removed old soyosource-wifi-dongle device and added it again, but I still cannot see Output power entity. Any recommendation? Tnx.

syssi, 2022-09-29:
Did you add the new sensor to your configuration YAML and reflashed the device (via OTA)?

blacklopo, 2022-09-29:
Yes, I did. But I will try it once more, just to be sure. Maybe some cache problem.

syssi, 2022-09-29:
I've rebased the `restart-on-crossing-zero-method` feature branch, so there `output_power` sensor is available there now too.

blacklopo, 2022-09-29:
Maybe I'm cursed, but `output_power `sensor is still not available. New clone of repo, updated config, one other sensor renamed, ota update. Still 1 device, 21 entities, 1 important is missing. 
I tried change the branch in config , but same result. Any recommendation how to debug it? I can try another computer :-D. Or I can send you my config, and you can generate firmware for me. :-D

syssi, 2022-09-29:
It's my fault. I did miss an important part. :-P

syssi, 2022-09-29:
I've merged a fix. Please give it another try.

closed 2022-09-29 by syssi

blacklopo, 2022-09-29:
Super! It is OK. I can see output power. We can continue with overshooting. ;-)

PR #83 (2022-09-28): Add output power sensor
Fixes #82
PR #84 (2022-09-29): Fix output power sensor registration
Fixes #82