syssi/esphome-soyosource-gtn-virtual-meter #85: WIFI dongle: Reboot function

blacklopo, 2022-09-29:
IMHO, the function for rebooting inverter via dongle is prepared in code, but not implemented.  According Soyo guys is should be this sequence:

```
bytestream[0]=0x55;
bytestream[1]=0x11;
bytestream[2]=0x00;
bytestream[3]=0x00;
bytestream[4]=0x00;
bytestream[5]=0x00;
bytestream[6]=0x00;
bytestream[7]=0x00;
bytestream[8]=0x00;
bytestream[9]=0x00;
bytestream[10]=0x00;
bytestream[11]=0xEE;

```

Am I correct or I overlooked it somewhere in the code?
PS: I'm not sure if this comment belongs to Issue or Discussion. Plz let me know.

syssi, 2022-09-29:
Feel free to create an issue per feature request. I prefer issues over discussions. I will have a look and will try to expose the reboot command as button entity.

syssi, 2022-09-29:
I've checked the implementation. The `send_command` method is able to send this `0x11` message and the message can be triggered via the `button.restart` entity:

https://github.com/syssi/esphome-soyosource-gtn-virtual-meter/blob/main/components/soyosource_display/button/__init__.py#L21
https://github.com/syssi/esphome-soyosource-gtn-virtual-meter/blob/main/components/soyosource_display/button/soyosource_button.cpp#L12

The payload is a bit different to your example above but I assume the `0x55 0x11` is the important part of the frame. Please make sure the button entity is part of your configuration and give it a try:

```
button:
  - platform: soyosource_display
    restart:
      name: "${name} restart"
```

blacklopo, 2022-09-30:
My fault. Reset is working properly.

closed 2022-09-30 by blacklopo