syssi/esphome-soyosource-gtn-virtual-meter #86: Mulitple inverters on the same RS485 bus

DL6ER, 2022-10-01:
https://github.com/syssi/esphome-soyosource-gtn-virtual-meter/commit/2c7a73fb43bf7c00d7e523c09e80bbb5acd78878 implemented a power divider useful when you have more than one of these inverters connected at the same time. I tried to set up exactly this scenario this weekend (two identical 1200W inverters in parallel on the same RS485 bus), however, the result isn't what I was expecting:

The relevant parts of my ESPHome config are:
``` yaml
soyosource_virtual_meter:
  power_id: powermeter
  min_power_demand: 0
  max_power_demand: 1800
  power_demand_divider: 2
  buffer: -3
  power_sensor_inactivity_timeout: 20s
  power_demand_calculation: NEGATIVE_MEASUREMENTS_REQUIRED
```

Everything works as it should except that **the maximum power demand cannot exceed 900 Watt** (resulting in both of the inverters doing a 450 W each). Obviously, the whole idea of installing more than one inverter is getting more than 900 W, though.

This leads me to the conclusion that `max_power_demand` (set to `1800` as above), is not accepted and the default value of 900
https://github.com/syssi/esphome-soyosource-gtn-virtual-meter/blob/6a5c59fd792cd6a1cce19f5ae31d4ad3875a7478/components/soyosource_virtual_meter/__init__.py#L22
https://github.com/syssi/esphome-soyosource-gtn-virtual-meter/blob/6a5c59fd792cd6a1cce19f5ae31d4ad3875a7478/components/soyosource_virtual_meter/__init__.py#L71-L73
is still used.

This is confirmed by (a) a red border around the input field telling me on hover me that values larger than 900 aren't allowed:
[image]

and (b) by launching a dedicated web server on the very same device:

Indeed, when adding
``` yaml
web_server:
  port: 80
```
showing that the maximum of the slider is still at 900:

[image]

syssi, 2022-10-01:
Good catch! I will have a look and change the limits. 73 de DO2KSM

syssi, 2022-10-01:
I've prepare a fix to increase the hard limits. Please adjust your configuration slightly:

```
soyosource_virtual_meter:
  power_id: powermeter
  min_power_demand: 0
  max_power_demand: 1800
  power_demand_divider: 2
  buffer: -3
  power_sensor_inactivity_timeout: 20s
  power_demand_calculation: NEGATIVE_MEASUREMENTS_REQUIRED

number:
  - platform: soyosource_virtual_meter
    manual_power_demand:
      name: "${name} manual power demand"
      max_value: 1800
    max_power_demand:
      name: "${name} max power demand"
      initial_value: 600
      max_value: 1800
      restore_value: true
```

syssi, 2022-10-01:
I've seen your proposal to solve the issue. It looks like I overlooked something and we have to decide the upper limit is the limit per device (900W) or per setup (2*900W).

syssi, 2022-10-01:
Could you try me approach by passing new limits to the number entities?

DL6ER, 2022-10-01:
I did already test it and left a comment on your PR. Concerning your question: per-setup seems to be less confusing (your PR) while I was thinking that per-device (my PR) is that the code currently tries to implement. I will close my PR in favor of yours as I think the per-setup way of setting it up will be easier to understand.

closed 2022-10-01 by syssi

DL6ER, 2022-10-01:
Thanks for the very quick fix!

73 aus dem Ahrtal

syssi, 2022-10-01:
73 aus Bonn. Ping me if you like to talk about PV on 70cm or 2m! ;-)

PR #88 (2022-10-01): Adjust hard limits of the maximum power demand and extend examples
Fixes #86
commit 1211ea38 (2022-10-01): Adjust hard limits of the maximum power demand and extend examples (#88)

Fixes #86