syssi/esphome-soyosource-gtn-virtual-meter #102: replace broken display with esp8266-display-example.yaml

derFrickler, 2023-02-14:
Well, not a bug, just a question.
Do I get it right, I can use that example to replace my broken Display in the soyo to change the settings via HA? So far I only use the rs485 to set the power.

Thanks a lot!

syssi, 2023-02-14:
Honestly I don't know. The example is dedicated for a Soyosource Inverter without a display / the WiFi dongle version of the device. But it's possible both devices are using the same protocol between the MCU and the display board. You could give it a try!

syssi, 2023-02-14:
If you are afraid we could try to compare the mainboard of your inverter (with a display and without an external display port) with the WiFi version of the Soyosource (no display but a display port).

derFrickler, 2023-02-15:
Thanks , i will just try it, lets see if it works.
What is strange is that the example file and the code used refer to display - even for the wifi version, so it looks to me that is was developed for the display version but works for the wifi aswell. e.g.
https://github.com/syssi/esphome-soyosource-gtn-virtual-meter/blob/822f9f182448dc404026803c996dc68fea319843/components/soyosource_display/soyosource_display.cpp#L120

But i am unable to find any docs or reference to is.

syssi, 2023-02-15:
I made the naming pretty generic because I expect it to work. ;-)

syssi, 2023-02-15:
If you own a cheap logic analyzer you could sniff some frames to be sure the assumption is correct.

derFrickler, 2023-02-19:
Well, protocol for the display seems to be different from the one for the Wifi stick. It did not work like that. 

I used an FTDI with an adum1201 to read the data from and to the display while its showing the status. 
The display continuously sends packages to the Inverter - even it its only connected to Power. 
`5502000000FD
5502000000FD
5502000000FD
5502000000FD
5502000000FD
5502000000FD
5502000000FD
5502000000FD
5502000000FD`

The Inverter answers with the following data: 

`A6025851420101000000DC64019738
A6025851420103000000DC64019736
A6025851420103000000DC64019736
A6025851420103000000DB64019737
A6025851420105000000DB64019834
A602585342D2181E1C00DB64460661
A6025851420103000000DB64019737
A6025851420103000000DB64019935
A6025851420103000000DB64019836
A6025851420103000000DB64019737
A6025851420103000000DB64019836
A6025851420103000000DB64019836
`
As said, my display is broken and nearly unreadable. I can't tell what is shown on the display for this packages till i get a new display. 

For reference. there seems to be somebody that has already worked on this, but seems he is not responding to questions....
https://www.photovoltaikforum.com/thread/192795-soyosource-wr-24v-1000w-display-kaputt-durch-einen-esp8266-ersetzen/?postID=3018115#post3018115

syssi, 2023-02-19:
@derFrickler Could you tell me some details about your Soyosource:

1. What's the operation mode of your device? It's BAT + Limiter, correct?
2. What's the nominal voltage of your battery / which inverter model do you own?
3. You battery wasn't attached and the limiter didn't request some power demand?

This my my first best guess of the frame content:

```
Pos Hex Dec
 0  A6  166
 1  02    2
 2  58   88
 3  51   81
 4  42   66
 5  01    1
 6  01    1
 7  00    0
 8  00    0
 9  00    0
10  DC  220   220V
11  64  100   100/2 = 50Hz
12  01    1
13  97  151
14  38   56
```

derFrickler, 2023-02-19:
Hi syssi, its a bit hard to tell ;-)
As the display is broken and i tried arround to much i guess i have messed up the settings and mode.
Its the 24V Model
Normally it was Bat+Limiter Mode
Battery is attached and Request is 600W, but the Bat Start or Bat Off Voltage are set to high i guess.

derFrickler, 2023-02-20:
Well, a bit of more data and thoughts on the content...

```
Pos Hex Dec
 0  A6  166   Start of frame
 1  02    2         -> 81 at startup, then changes to 00 01 02 and stays on 02
 2  58   88         -> 06 at startup, then changes to 00 86 58 and stays on 58
 3  51   81         -> part of Total?
 4  42   66         -> 51 42 -> 20802 -> Guess 208 total - still 5142 the next days - as the inverter did not produce anything.
 5  01    1         -> part of Bat Volt?
 6  01    1         -> 01 01 -> 257  Bat Volt? Wert schwankt zwischen 25.7 und 26.1
 7  00    0         -> part of Bat Curr?
 8  00    0         -> 00 00 -> 0    Bat Curr 0 da standby?
 9  00    0         -> Always 00
10  DC  220         220V  -> changes between 219 and 220!
11  64  100         100/2 = 50Hz  fix auf 50Hz
12  01    1         -> always 01
13  97  151         -> 97 -> 151 und 153  guess Temp in 0.1 as it was 8.1-8-3 this morning?
14  38   56   Checksum (Pos 1...13)
```
and some more data captured on  PowerOn sequence of Inverter, Captured from and to display at the same time: 
24V 1000W Inverter
220V 50Hz
Bat voltage ~25.6V

```
5502000000FD
A681065300D2181E1C0000004606B5
5501000000FE
A6810651020101000000DF6401518E
5501000000FE
A6810651020101000000DE6401538D
5501000000FE
A6810651020101000000DE6401518F
5501000000FE
A6810651020101000000DE6401528E
5501000000FE
A6810651020103000000DE6401528C
5501000000FE
A6000051420101000000DE640151D6
5501000000FE
A60000514200FF000000DC640152DA
5501000000FE
A6000051420101000000DE640153D4
5501000000FE
A6018651420101000000DE6401534D
5503000000FC
A601865342D2181E1C00DE64460631
5503000000FC
A602585342D2181E1C00DE6446065E
5501000000FE
A6025851420101000000DD6401527C
5501000000FE
A60258514200FF000000DD6401537E
5501000000FE
A6025851420113000000DD6401516B
5501000000FE
A6025851420101000000DD6401527C
5501000000FE
A6025851420101000000DB6401537D
5501000000FE
A6025851420101000000DD6401537B
5503000000FC
A602585342D2181E1C00DD6446065F

-> On startup guess Type request?
5502000000FD
Answer: 
A681065300D2181E1C0000004606B5

-> Status Request:
5501000000FE
Answers:
A681065300D2181E1C0000004606B5
A6810651020101000000DF6401518E
A6810651020101000000DE6401518F
A6810651020101000000DE6401518F
A6810651020103000000DE6401528C
A6000051420101000000DE640151D6
A6018651420101000000DE6401534D
A6025851420101000000DD6401527C

-> Settings Request (when settings screen is opened):
5503000000FC
Answers:
A601865342D2181E1C00DE64460631
A602585342D2181E1C00DE6446065E
A602585342D2181E1C00DD6446065F
```

syssi, 2023-02-20:
Good job! I've added some more thoughts:

```
Pos Hex Dec
 0  A6  166   Start of frame
 1  02    2   Power demand MSB  or   operation mode
 2  58   88   Power demand LSB       0x0258 = 600W
 3  51   81   Total energy MSB?
 4  42   66   Total energy LSB?      0x5142 = 20802. I don't think this is the total energy because I've replaced a display board (LCD+PCB) in the past and the total energy value is stored on the display board.
 5  01    1   Battery voltage MSB
 6  01    1   Battery voltage LSB    256/10 = 25.6V
 7  00    0   Battery current MSB
 8  00    0   Battery current LSB
 9  00    0   Operation mode?
10  DC  220   AC voltage             220V
11  64  100   AC frequency           100/2 = 50Hz
12  01    1   -> always 01
13  97  151   Temperature            151/10 = 15.1°C
14  38   56   Checksum (Pos 1...13)
```

Could you unplug the RS485 limiter cable or change the power demand to another value? Just to get an idea the power demand is part of the frame?

derFrickler, 2023-02-20:
I will try that later, to bad that Udo_17 from the forum is not responding, as he seems to have al the info...

syssi, 2023-02-20:
Yes. It feels like we do the same work a second time. ;-(

derFrickler, 2023-02-20:
Next round, managed to get the inverter back to work on rs485 data by playing with the setting screen more than blindly ;_)

this sequence shows 600W 300W 0W rs485 disconnected
´´´
A60258914000F8011700E464024931
A60258914000F6011500E464024935
A60258914000F8011700E464024931
A60258914000F8011700E464024A30
A60258914000F6011600E464024A33
A60258914000F8011600E464024932
A60258914000F8011400E464024C31
A60258914000F8011700E464024733
A60258914000F8011500E464024A32
A60258914000F8011700E464024A30
A60258914000F8011600E464024A31
A60258914000FA011400E464024B30
A60258914000F8011500E464024933
A6012C914000FD008700E464024BE8
A6012C914000FD008500E464024CE9
A6012C914000FB008500E364024EEA
A6012C914000FF008600E364024CE7
A6012C914000FF008600E364024BE8
A6012C914000FD008600E364024CE9
A6012C914000FF008600E364024CE7
A6012C914000FF008500E364024CE8
A6012C914000FF008500E264024BEA
A6012C914000FF008600E264024EE6
A6012C914000FD008600E264024EE8
A6012C914000FF008600E264024DE7
A6012C914000FF008500E264024DE8
A6012C914000FF008500E264024EE7
A6012C91400101008600E164024DE5
A6000091400103000000E164024D96
A6000091400105000000E164025091
A6000091400105000000E164024D94
A6000091400105000000E164024D94
A6000091400105000000E164024E93
A6000091400107000000E264024B93
A6000091400105000000E264024E92
A6000091400105000000E264024D93
A6000091400105000000E264025090
A6000091400105000000E264024D93
A6000091400105000000E264024D93
A6000091400105000000E164024E93
A6000091400105000000E164024B96
A6000091400105000000E164024D94
A6000091400105000000E164024D94
A6000091400105000000E164024D94
A6000091000107000000E164024CD3
A6000091000105000000E164024CD5
A6000091000105000000E164024CD5
´´´

This confirms, 1-2 are the Power demand
A6   
0258     Power Demand 600W
91         Always 91
40         Status? Seems to change to 00  if RS485 is disconnected.
00F8    Bat Volt. /10
0117    Bat Curr /10
00        ? Always Zero
E4        AC Volts
64        AC Freq.
02        Mode? was 01 before is now 02 with Limiter working
49       Temp/10
31       Chcksum

syssi, 2023-02-20:
Long story short: The frames look like the already supported `soyosource_display` frames. I will try to proof this tomorrow.

derFrickler, 2023-02-20:
Check this project for the protocol definition:
https://github.com/PepeTheFroggie/Soyosource-GridTie-inverter-24V-LCD-replacement

syssi, 2023-02-21:
I made the following experiment: I publish your display traffic frames from above periodically using this configuration on GPIO4/GPIO05:

```
substitutions:
  name: fake-display-traffic
  device_description: "Emulate the display port traffic of the Soyosource GTN (display version)"
  tx_pin: GPIO4
  rx_pin: GPIO5

esphome:
  name: ${name}
  comment: ${device_description}
  project:
    name: "syssi.esphome-soyosource-gtn-virtual-meter"
    version: 2.0.0

esp8266:
  board: d1_mini

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:
api:
logger:

uart:
  baud_rate: 9600
  tx_pin: ${tx_pin}
  rx_pin: ${rx_pin}
  debug:
    direction: BOTH
    dummy_receiver: true

interval:
  - interval: 20s
    then:
      # status responses
      - uart.write: [0xA6, 0x02, 0x58, 0x91, 0x40, 0x00, 0xF8, 0x01, 0x17, 0x00, 0xE4, 0x64, 0x02, 0x4A, 0x30]
      - delay: 1s

      - uart.write: [0xA6, 0x02, 0x58, 0x91, 0x40, 0x00, 0xF8, 0x01, 0x16, 0x00, 0xE4, 0x64, 0x02, 0x4A, 0x31]
      - delay: 1s

      - uart.write: [0xA6, 0x02, 0x58, 0x91, 0x40, 0x00, 0xFA, 0x01, 0x14, 0x00, 0xE4, 0x64, 0x02, 0x4B, 0x30]
      - delay: 1s

      - uart.write: [0xA6, 0x02, 0x58, 0x91, 0x40, 0x00, 0xF8, 0x01, 0x15, 0x00, 0xE4, 0x64, 0x02, 0x49, 0x33]
      - delay: 1s

      - uart.write: [0xA6, 0x01, 0x2C, 0x91, 0x40, 0x00, 0xFD, 0x00, 0x87, 0x00, 0xE4, 0x64, 0x02, 0x4B, 0xE8]
      - delay: 1s

      - uart.write: [0xA6, 0x01, 0x2C, 0x91, 0x40, 0x00, 0xFD, 0x00, 0x85, 0x00, 0xE4, 0x64, 0x02, 0x4C, 0xE9]
      - delay: 1s

      - uart.write: [0xA6, 0x01, 0x2C, 0x91, 0x40, 0x00, 0xFB, 0x00, 0x85, 0x00, 0xE3, 0x64, 0x02, 0x4E, 0xEA]
      - delay: 1s

      - uart.write: [0xA6, 0x01, 0x2C, 0x91, 0x40, 0x00, 0xFF, 0x00, 0x86, 0x00, 0xE3, 0x64, 0x02, 0x4C, 0xE7]
      - delay: 1s

      - uart.write: [0xA6, 0x00, 0x00, 0x91, 0x40, 0x01, 0x03, 0x00, 0x00, 0x00, 0xE1, 0x64, 0x02, 0x4D, 0x96]
      - delay: 1s

      - uart.write: [0xA6, 0x00, 0x00, 0x91, 0x40, 0x01, 0x05, 0x00, 0x00, 0x00, 0xE1, 0x64, 0x02, 0x50, 0x91]
      - delay: 1s

      - uart.write: [0xA6, 0x00, 0x00, 0x91, 0x40, 0x01, 0x05, 0x00, 0x00, 0x00, 0xE1, 0x64, 0x02, 0x4D, 0x94]
[... 190 lines cut ...]
[07:45:12][I][soyosource_display:388]:   Device type: PV 26-56V / BAT 24V (24)
[07:45:12][I][soyosource_display:392]:   Start voltage: 30 V
[07:45:12][D][number:012]: 'soyosource-gtn-display-test start voltage': Sending state 30.000000
[07:45:12][I][soyosource_display:398]:   Shutdown voltage: 28 V
[07:45:12][D][number:012]: 'soyosource-gtn-display-test shutdown voltage': Sending state 28.000000
[07:45:12][I][soyosource_display:412]:   Output power limit: 700 W
[07:45:12][D][number:012]: 'soyosource-gtn-display-test output power limit': Sending state 700.000000
[07:45:12][I][soyosource_display:418]:   Start delay: 6 s
[07:45:12][D][number:012]: 'soyosource-gtn-display-test start delay': Sending state 6.000000
[07:45:13][I][soyosource_display:360]: Settings:
[07:45:13][D][soyosource_display:365]:   Unknown (byte 1): 02
[07:45:13][D][soyosource_display:368]:   Unknown (byte 2): 58
[07:45:13][I][soyosource_display:372]:   Operation mode setting: 02
[07:45:13][D][select:015]: 'soyosource-gtn-display-test operation mode': Sending state Battery Constant Power (index 1)
[07:45:13][I][soyosource_display:385]:   Device model: 1000 W (210)
[07:45:13][I][soyosource_display:388]:   Device type: PV 26-56V / BAT 24V (24)
[07:45:13][I][soyosource_display:392]:   Start voltage: 30 V
[07:45:13][D][number:012]: 'soyosource-gtn-display-test start voltage': Sending state 30.000000
[07:45:13][I][soyosource_display:398]:   Shutdown voltage: 28 V
[07:45:13][D][number:012]: 'soyosource-gtn-display-test shutdown voltage': Sending state 28.000000
[07:45:13][I][soyosource_display:412]:   Output power limit: 700 W
[07:45:13][D][number:012]: 'soyosource-gtn-display-test output power limit': Sending state 700.000000
[07:45:13][I][soyosource_display:418]:   Start delay: 6 s
[07:45:13][D][number:012]: 'soyosource-gtn-display-test start delay': Sending state 6.000000
```

syssi, 2023-02-21:
Which GPIOs did you use for the [Soyosource-GridTie-inverter-24V-LCD-replacement](https://github.com/PepeTheFroggie/Soyosource-GridTie-inverter-24V-LCD-replacement) solution? I would like to provide a drop in replacement.

derFrickler, 2023-02-21:
He is using the hardware serial with swap ( GPIO15 (TX) and GPIO13 (RX) by calling Serial.swap()) for the connection. 
I would keep it like in your code and stay on 
```
  tx_pin_ttl_wifi: GPIO14
  rx_pin_ttl_wifi: GPIO12
  tx_pin_rs485: GPIO4
  rx_pin_rs485: GPIO5
```

syssi, 2023-02-21:
Good catch!! The requests are different:

```
# WiFi dongle display port requests
55:01:00:00:00:00:00:00:00:00:00:FE
55:03:00:00:00:00:00:00:00:00:00:FC

# Display requests
55:02:00:00:00:FD
55:01:00:00:00:FE
55:03:00:00:00:FC
```

syssi, 2023-02-21:
How to write settings: https://github.com/PepeTheFroggie/Soyosource-GridTie-inverter-24V-LCD-replacement/blob/main/SoyoGraph_24v/soyo.ino#L84-L142

In future a message per setting/change is required. The WiFi version of the inverter supports updating all settings in a single frame.

derFrickler, 2023-02-21:
great! 
I will try that the next days, i am busy today. 
Settings is not urgent anymore as its working again now. I will have a look for the settings code.

syssi, 2023-02-21:
Take your time! I would be happy about your feedback to finish the feature some day.

derFrickler, 2023-02-24:
YES, just testet you new branch and it seems to be working: 

```
10:38:51][D][number:012]: 'soyosource-gtn-display start delay': Sending state 6.000000
[10:38:51][D][uart_debug:114]: <<< A6:02:58:63:42:D2:18:17:16:00:DE:63:46:06:5C
[10:38:55][D][uart_debug:114]: >>> 55:01:00:00:00:FE
[10:38:55][I][soyosource_display:225]: Status frame received
[10:38:55][D][sensor:126]: 'soyosource-gtn-display output power': Sending state 600.00000 W with 0 decimals of accuracy
[10:38:55][D][sensor:126]: 'soyosource-gtn-display operation mode id': Sending state 6.00000  with 0 decimals of accuracy
[10:38:55][D][text_sensor:067]: 'soyosource-gtn-display operation mode': Sending state 'PV'
[10:38:55][D][sensor:126]: 'soyosource-gtn-display operation status id': Sending state 2.00000  with 0 decimals of accuracy
[10:38:55][D][text_sensor:067]: 'soyosource-gtn-display operation status': Sending state 'Standby'
[10:38:55][D][sensor:126]: 'soyosource-gtn-display error bitmask': Sending state 2.00000  with 0 decimals of accuracy
[10:38:55][D][text_sensor:067]: 'soyosource-gtn-display errors': Sending state 'DC voltage too low'
[10:38:55][D][sensor:126]: 'soyosource-gtn-display battery voltage': Sending state 24.20000 V with 1 decimals of accuracy
[10:38:55][D][sensor:126]: 'soyosource-gtn-display battery current': Sending state 0.00000 A with 1 decimals of accuracy
[10:38:55][D][sensor:126]: 'soyosource-gtn-display battery power': Sending state 0.00000 W with 0 decimals of accuracy
[10:38:56][D][sensor:126]: 'soyosource-gtn-display ac voltage': Sending state 222.00000 V with 0 decimals of accuracy
[10:38:56][D][sensor:126]: 'soyosource-gtn-display ac frequency': Sending state 50.00000 Hz with 1 decimals of accuracy
[10:38:56][D][sensor:126]: 'soyosource-gtn-display temperature': Sending state 10.70000 °C with 1 decimals of accuracy
[10:38:56][D][uart_debug:114]: <<< A6:02:58:61:42:00:F2:00:00:00:DE:64:01:97:36
[10:38:56][D][uart_debug:114]: >>> 55:03:00:00:00:FC
[10:38:56][I][soyosource_display:360]: Settings:
[10:38:56][D][soyosource_display:365]:   Unknown (byte 1): 02
[10:38:56][D][soyosource_display:368]:   Unknown (byte 2): 58
[10:38:56][I][soyosource_display:372]:   Operation mode setting: 01
[10:38:56][D][select:015]: 'soyosource-gtn-display operation mode': Sending state PV (index 0)
[10:38:56][I][soyosource_display:385]:   Device model: 1000 W (210)
[10:38:56][I][soyosource_display:388]:   Device type: PV 26-56V / BAT 24V (24)
[10:38:56][I][soyosource_display:392]:   Start voltage: 23 V
[10:38:56][D][number:012]: 'soyosource-gtn-display start voltage': Sending state 23.000000
[10:38:56][I][soyosource_display:398]:   Shutdown voltage: 22 V
[10:38:56][D][number:012]: 'soyosource-gtn-display shutdown voltage': Sending state 22.000000
[10:38:56][I][soyosource_display:412]:   Output power limit: 700 W
[10:38:56][D][number:012]: 'soyosource-gtn-display output power limit': Sending state 700.000000
[10:38:56][I][soyosource_display:418]:   Start delay: 6 s
```

Had some problems first as there was nothing on the mqtt because it was not defined in your example anymore. After adding that its back again on mqtt and in HA ;-)

[image]

syssi, 2023-02-24:
Hmm... but it looks like the interpretation of some values isn't fine:

1. Operation mode "PV". It should be "BAT", right?
2. The inverter delivers some power but the `error bitmask` indicates "DC voltage too low" + Operation status "Standby". 

The other values seem correct.

derFrickler, 2023-02-24:
Oh yes, right!
State is Battery and the error message should not be there

syssi, 2023-02-24:
I will have a look!

ButschFPV, 2023-03-09:
Great work, thanks a lot!
I have a similar use case. One display is already broken and the new one is getting also unreadable. So it's time to completely ditch the display.
With the `improve-display-support` branch flashed onto a d1-mini, i'm already getting values from the inverter.
At the moment i don't have a battery connected. The debug output seems fine, except unknown `operation mode` and `operation mode setting`:

```
[22:31:50][D][uart_debug:114]: >>> 55:01:00:00:00:FE
[22:31:50][I][soyosource_display:225]: Status frame received
[22:31:50][D][sensor:124]: 'soyosource-gtn-display output power': Sending state 257.00000 W with 0 decimals of accuracy
[22:31:50][D][sensor:124]: 'soyosource-gtn-display operation mode id': Sending state 12.00000  with 0 decimals of accuracy
[22:31:50][W][soyosource_display:562]:   Operation mode: Unknown (12)
[22:31:50][D][text_sensor:067]: 'soyosource-gtn-display operation mode': Sending state 'Unknown'
[22:31:50][D][sensor:124]: 'soyosource-gtn-display operation status id': Sending state 2.00000  with 0 decimals of accuracy
[22:31:50][D][text_sensor:067]: 'soyosource-gtn-display operation status': Sending state 'Standby'
[22:31:50][D][sensor:124]: 'soyosource-gtn-display error bitmask': Sending state 2.00000  with 0 decimals of accuracy
[22:31:50][D][text_sensor:067]: 'soyosource-gtn-display errors': Sending state 'DC voltage too low'
[22:31:50][D][sensor:124]: 'soyosource-gtn-display battery voltage': Sending state 0.00000 V with 1 decimals of accuracy
[22:31:50][D][sensor:124]: 'soyosource-gtn-display battery current': Sending state 0.00000 A with 1 decimals of accuracy
[22:31:50][D][sensor:124]: 'soyosource-gtn-display battery power': Sending state 0.00000 W with 0 decimals of accuracy
[22:31:50][D][sensor:124]: 'soyosource-gtn-display ac voltage': Sending state 221.00000 V with 0 decimals of accuracy
[22:31:50][D][sensor:124]: 'soyosource-gtn-display ac frequency': Sending state 50.00000 Hz with 1 decimals of accuracy
[22:31:50][D][sensor:124]: 'soyosource-gtn-display temperature': Sending state 14.90000 °C with 1 decimals of accuracy
[22:31:50][D][uart_debug:114]: <<< A6:01:01:C1:02:00:00:00:00:00:DD:64:01:C1:37
[22:31:50][D][uart_debug:114]: >>> 55:03:00:00:00:FC
[22:31:50][I][soyosource_display:360]: Settings:
[22:31:50][D][soyosource_display:365]:   Unknown (byte 1): 01
[22:31:50][D][soyosource_display:368]:   Unknown (byte 2): 01
[22:31:50][I][soyosource_display:372]:   Operation mode setting: 00
[22:31:50][W][soyosource_display.select:015]: Invalid value 0
[22:31:50][I][soyosource_display:385]:   Device model: 1000 W (210)
[22:31:50][I][soyosource_display:388]:   Device type: PV 26-56V / BAT 24V (24)
[22:31:50][I][soyosource_display:392]:   Start voltage: 25 V
[22:31:50][D][number:012]: 'soyosource-gtn-display start voltage': Sending state 25.000000
[22:31:50][I][soyosource_display:398]:   Shutdown voltage: 25 V
[22:31:50][D][number:012]: 'soyosource-gtn-display shutdown voltage': Sending state 25.000000
[22:31:50][I][soyosource_display:412]:   Output power limit: 350 W
[22:31:50][D][number:012]: 'soyosource-gtn-display output power limit': Sending state 350.000000
[22:31:50][I][soyosource_display:418]:   Start delay: 60 s
[22:31:50][D][number:012]: 'soyosource-gtn-display start delay': Sending state 60.000000
[22:31:50][D][uart_debug:114]: <<< A6:01:01:C3:02:D2:18:19:19:00:DD:64:23:3C:7C
```
Without a battery connected, the output power `01:01` = 257W does not make sense. 
Maybe the output power should be ignored in `Operation mode: Unknown (12)`.

syssi, 2023-03-09:
Thanks for your feedback! Do you like to provide some further testing? To move on here some testers would be great.

syssi, 2023-03-09:
> I tried changing that with the write settings code from PepeTheFrogs project, but then the Limiting is not working anymore. Seems there is an issue in the settings stuff aswell.

I don't understand this sentence. :-) Do you talk about the settings stuff of PepeTheFrogs project? My implementation doesn't provide settings/controls yet.

ButschFPV, 2023-03-09:
> Thanks for your feedback! Do you like to provide some further testing? To move on here some testers would be great.

sure, i'm going to do some investigation this weekend

derFrickler, 2023-03-09:
> I don't understand this sentence. :-) Do you talk about the settings stuff of PepeTheFrogs project? My implementation doesn't provide settings/controls yet.

YES, i used the code from Pepe to change the settings. With that i am able to change from Bat to PV and back, but Limiter mode gets disabled as soon as i change settings, ist just on fix limit then and without the display i can't enable it again...

kev300, 2023-03-09:
Sadly, my inverter is a quiet one. :-( Purple board, does not ever send anything back on rs485 (no led blinking)
I could also try to test this.

Can I just connect the 2 tx / rx lines from the display with 10K resistors in between to my esp32?

syssi, 2023-03-09:
> Can I just connect the 2 tx / rx lines from the display with 10K resistors in between to my esp32?

Yes. I recommend adding an optical isolator (ADUM1201) to be super safe.

jhermen-List, 2023-03-09:
Directly connecting (with inline resistors) works fine as long as the esp is powered by the inverter ant there is is no connection to the PC.

ButschFPV, 2023-03-09:
I have connected the esp8266 with a npn transistor as a level shifter. Just make sure to invert the uart rx_pin in YAML
[image]

kev300, 2023-03-09:
> Directly connecting (with inline resistors) works fine as long as the esp is powered by the inverter ant there is is no connection to the PC.

My ESP is powered by a smartphone usb brick.
I suppose you mean it's an issue if the ESP and Soyo do not have a common ground?

[ButschFPV](https://github.com/ButschFPV) solution also works only with common ground, right?

syssi, 2023-03-09:
> I suppose you mean it's an issue if the ESP and Soyo do not have a common ground?

Yes. This is the risk. 

> [ButschFPV](https://github.com/ButschFPV) solution also works only with common ground, right?

The solution is risky only if the ESP isn't powered by the Soyosource.

kev300, 2023-03-09:
Ok, thanks for the hint before I killed my ESP :D
I will buy ADUM1201 then first, because my ESP is also doing other sutff like jk-bms and garage door control, it won't be powered by the soyo.

ButschFPV, 2023-03-09:
> My ESP is powered by a smartphone usb brick. I suppose you mean it's an issue if the ESP and Soyo do not have a common ground?
>
> [ButschFPV](https://github.com/ButschFPV) solution also works only [...]

Clearly optical isolation is the better choice if your esp is connected to other stuff.
In my example the esp is powered by the Soyosource and share common ground.
I flash the esp via OTA so no need to plug in my computer.

derFrickler, 2023-03-09:
So, just attached the working display, set the inverter to Bat Limit and ist back working fine. 
Replaced the displayby your code and now I can confirm that the mode and error are reporting correct. 

[image]

syssi, 2023-03-09:
@derFrickler Perfect! Could you sniff some of the "write settings commands"? I would like to know how a "set operation mode to PV" frame looks like. Same applies to

- Set start voltage
- Set stop voltage
- Set start delay
- Set output power limit

Thanks in advance!

syssi, 2023-03-09:
Awesome! This helps a lot to finish the display support feature soon!

ButschFPV, 2023-03-10:
I had my inverter running in BatCP and BatLimit mode.
Status byte 01 & 02 always shows 01:01.
So my inverter does not report the target output power. The displys seems to show the calculated U*I value.

syssi, 2023-03-10:
Todos:

- [ ] Collect all possible operation mode values (read & write)
- [ ] Fix operation mode (status frame)
- [ ] Fix operation mode setting (settings frame)
- [ ] Ignore byte 1-2 if Soyosource display version (assumption: It's always 0x01 0x01)
- [ ] Build proper control frames to be able update the settings

syssi, 2023-03-10:
@ButschFPV I've read your recent messages another time now and it looks like you you are several steps ahead. :-)

syssi, 2023-03-10:
```
# Start delay
55:83:3B:00:00:41
55:83:3C:00:00:40
      ^^

# Operation mode?
55:03:17:18:02:CB # Constant power mode
55:03:23:64:01:74 # off
55:03:23:64:10:65 # Limiter mode
            ^^

# Power output limit
55:13:23:64:00:65 # 0x23 = 35*10W
55:13:22:64:00:66 # 0x22 = 34*10W
      ^^

# Change stop voltage
55:0B:17:18:00:C5
55:0B:17:17:00:C6
55:0B:17:16:00:C7
         ^^

# Change start voltage
55:0B:17:18:00:C5
55:0B:18:18:00:C4
      ^^
```

ButschFPV, 2023-03-10:
I have the inverter on my desk. So if you have any questions i can collect more data tomorrow.

syssi, 2023-03-10:
@ButschFPV Could you collect some more frames? I'm bit confused because of the second byte (lets call it `function`). The function values of your recorded frames doesn't match with the one of Pepe:

https://github.com/PepeTheFroggie/Soyosource-GridTie-inverter-24V-LCD-replacement/blob/main/SoyoGraph_24v/soyo.ino#L111-L142

Could you set your Soyosource into "PV mode" and set the start and stop voltage again? I assume the function is `0x0A` in this case. Could you collect all "change operation mode" frames and the resulting "operation mode" (status & settings frame)?

derFrickler, 2023-03-10:
As said, the settings code from Pepe is not working that good, setting the limit mode does not seem to work at all, not in PV not in Bat.

syssi, 2023-03-10:
I know but I would like to understand the root cause. I assume the second byte does change depending on the operation mode.

syssi, 2023-03-10:
The feature branch is able to decode the operation modes now probably. This updated YAML could be used for testing: 

https://github.com/syssi/esphome-soyosource-gtn-virtual-meter/blob/improve-display-support/esp8266-display-example.yaml

https://github.com/syssi/esphome-soyosource-gtn-virtual-meter/blob/improve-display-support/components/soyosource_display/soyosource_display.cpp#L587-L618

There are two values (0x05, 0x06) which aren't mapped yet because the meaning is unknown.

ButschFPV, 2023-03-11:
I had a "quick" look and recorded some more data.

```
Access menu:
>>> 55:03:00:00:00:FC
<<< A6:01:01:C3:02:D2:18:17:16:00:DA:64:03:3C:A4

Set PV Mode
>>> 55:03:00:00:01:FB
<<< A6:01:01:63:02:D2:18:17:16:00:D9:64:03:3C:05

Set BatCP Mode
>>> 55:03:00:00:02:FA
<<< A6:01:01:63:02:D2:18:17:16:00:D9:64:03:3C:05

Set BatLimit Mode
>>> 55:03:00:00:10:EC
<<< A6:01:01:C3:02:D2:18:17:16:00:D8:64:03:3C:A6

The Following changes are done in PV Mode
Set Bat start voltage to 26V
>>> 55:0B:1A:15:00:C5
<<< A6:01:01:63:02:D2:18:1A:15:00:DA:64:3C:3C:C9

Set Bat start voltage to 27V
>>> 55:0B:1B:15:00:C4
<<< A6:01:01:63:02:D2:18:1B:15:00:DA:64:3C:3C:C8

Set Bat stop voltage to 22V
>>> 55:0B:18:16:00:C6
<<< A6:01:01:63:02:D2:18:18:16:00:DA:64:3C:3C:CA

Set Bat stop voltage to 21V
>>> 55:0B:18:15:00:C7
<<< A6:01:01:63:02:D2:18:18:15:00:DA:64:3C:3C:CB

Set Bat CP Mode Power to 40W
>>> 55:13:04:64:00:84
<<< A6:01:01:63:02:D2:18:18:17:00:DA:64:04:3C:01

Set Bat CP Mode Power to 600W
>>> 55:13:3C:64:00:4C
<<< A6:01:01:63:02:D2:18:18:17:00:DA:64:3C:3C:C9

Set Delay Start to 59s
>>> 55:83:3B:00:00:41
<<< A6:01:01:63:02:D2:18:1A:15:00:DA:64:3C:3B:CA

Set Delay Start to 60s
>>> 55:83:3C:00:00:40
<<< A6:01:01:63:02:D2:18:1A:15:00:DA:64:3C:3C:C9

The Following changes are done in BatLimit Mode
Set Bat start voltage to 25V
>>> 55:0B:19:15:00:C6
<<< A6:01:01:C3:02:D2:18:19:15:00:D9:64:3C:3C:6B

Set Bat start voltage to 26V
>>> 55:0B:1A:15:00:C5
<<< A6:01:01:C3:02:D2:18:1A:15:00:D9:64:3C:3C:6A
```

I assume the other settings are identical throughout the modes.

I was a little bit confused about my previously recorded frames.
Set PV Mode looked like this: `55:03:17:18:01:CC`
And now like this: `55:03:00:00:01:FB`
Well turns out Byte 02 & 03 shows the last changed value(s).
In my previously recorded frame the last change was Battery start/stop voltage. Hence the `17:18`.
This is then also visible in the normal status request: `55:01:17:18:00:XX`.
If i had changed the BatCP Mode Power to 30W, then the status request looked like: `55:01:03:64:00:XX` or sometimes `55:01:03:63:00:XX`. I'm not completely sure if Byte 03 is the last received Mains-Frequency value.

The question is, does it even matter? Is it just gap filler?

syssi, 2023-03-11:
I assume in some cases it matters in some not. For example:

`0x0B` is the command to update the start/stop voltage. The first byte of the data is the `start voltage`, the second byte `stop voltage` and it looks like the third byte should be always `0x00`. There is no instruction which one of the voltages should be updated to we have to pass both voltages always.

`0x03` is the "change operation mode" command. The third byte indicates the new operation mode. If we pass `0x00` as third byte the settings frame is requested / nothing gets changed. Could you repeat the "change operation mode" tests? These responses must not be equal:

```
Set PV Mode
>>> 55:03:00:00:01:FB
<<< A6:01:01:63:02:D2:18:17:16:00:D9:64:03:3C:05 <--

Set BatCP Mode
>>> 55:03:00:00:02:FA
<<< A6:01:01:63:02:D2:18:17:16:00:D9:64:03:3C:05 <--
```

`0x83` is the command to change the start delay.

`0x13` the command to set the "output power limit" (please keep in mind this affects the CP mode and the Limiter mode. In both cases its the upper bound) Let's pass the grid frequency here too. It's pretty similar to the display protocol of the WiFi version.

syssi, 2023-03-11:
Could you do me another favour? If you have changed to operation mode to "PV" please go back to the status page to retrieve a status frame. The operation mode should change here too. I would like to know these operation mode values are equal to the settings frame or shifted.

ButschFPV, 2023-03-11:
Hm this is strange. I was focused on the request, so I didn't noticed.
I did run the same test again:

```
Set BatLimit Mode
>>> 55:03:00:00:10:EC
<<< A6:01:01:63:02:D2:18:17:16:00:DB:64:02:3C:04

Set PV Mode
>>> 55:03:00:00:01:FB
<<< A6:01:01:63:02:D2:18:17:16:00:DC:64:02:3C:03

Set BatCP Mode
>>> 55:03:00:00:02:FA
<<< A6:01:01:C3:02:D2:18:17:16:00:D9:64:02:3C:A6
```
Again different values.
If I had to guess, maybe another random value.
I discovered some frames every time I exited the configuration menu. So maybe this is where you get the confirmation of the actual mode:

```
Exit menu in BatCP Mode
>>> 55:03:00:00:00:FC
<<< A6:01:01:53:02:D2:18:17:16:00:D9:64:02:3C:16

Exit Menu in PV Mode
>>> 55:03:00:00:00:FC
<<< A6:01:01:63:02:D2:18:17:16:00:DC:64:02:3C:03

Exit Menu in BatLimit Mode
>>> 55:03:00:00:00:FC
<<< A6:01:01:C3:02:D2:18:17:16:00:DB:64:02:3C:A4
```

Here are the status page responses:

```
Status page response in BatCP Mode
<<< A6:01:01:51:02:00:00:00:00:00:DA:64:01:BC:AF

Status page response in PV Mode
<<< A6:01:01:61:02:00:00:00:00:00:DC:64:01:BE:9B

Status page response in BatLimit Mode
<<< A6:01:01:C1:02:00:00:00:00:00:DA:64:01:C0:3B

```

syssi, 2023-03-13:
This is my new test setup to study the behaviour of the display.

[image]

syssi, 2023-03-15:
The display protocol support for the "Soyosource display version" is ready and merged into `main`.

closed 2023-03-15 by syssi

derFrickler, 2023-03-16:
Thanks a lot for the great work! Its doing just fine now!
Just one remark, the example file is still using the now non existing dev-branch, it should now use main:
https://github.com/syssi/esphome-soyosource-gtn-virtual-meter/blob/446e41735fe8d2baa7a8a96704f45114f29f67f9/esp8266-display-display-version-example.yaml#L4

and i did a working config for the display version including the limiting via rs485:

esp8266-display-display-version-limiter-example.yaml
```
substitutions:
  name: soyosource-gtn-display
  device_description: "Monitor and control the Soyosource GTN (WiFi version) via the display port and control the power output on demand via RS485"
  external_components_source: github://syssi/esphome-soyosource-gtn-virtual-meter@main
  tx_pin_ttl_wifi: GPIO4
  rx_pin_ttl_wifi: GPIO5
  tx_pin_rs485: GPIO14
  rx_pin_rs485: GPIO12

esphome:
  name: ${name}
  comment: ${device_description}
  project:
    name: "syssi.esphome-soyosource-gtn-virtual-meter"
    version: 2.1.0

esp8266:
  board: d1_mini

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:

# If you use Home Assistant please remove this `mqtt` section and uncomment the `api` component!
mqtt:
  broker: !secret mqtt_host
  username: !secret mqtt_username
  password: !secret mqtt_password
  id: mqtt_client

# api:

logger:
  level: INFO

# Please be careful: The display port has a logic level of 5V.
uart:
  - id: uart0
    baud_rate: 9600
    tx_pin: ${tx_pin_ttl_wifi}
    rx_pin: ${rx_pin_ttl_wifi}

  - id: uart1
    baud_rate: 4800
    tx_pin: ${tx_pin_rs485}
    rx_pin: ${rx_pin_rs485}

soyosource_modbus:
  - id: modbus0
    uart_id: uart1

soyosource_virtual_meter:
  - id: virtualmeter0
    soyosource_modbus_id: modbus0

    # The state of this sensor (instantaneous power in watt) is used as source
[... 128 lines cut ...]
switch:
  - platform: soyosource_virtual_meter
    soyosource_virtual_meter_id: virtualmeter0
    manual_mode:
      name: "${name} manual mode"
      restore_mode: RESTORE_DEFAULT_ON
    emergency_power_off:
      name: "${name} emergency power off"
      restore_mode: RESTORE_DEFAULT_OFF

text_sensor:
  - platform: soyosource_virtual_meter
    soyosource_virtual_meter_id: virtualmeter0
    operation_mode:
      name: "${name} limiter operation mode"

  - platform: soyosource_display
    errors:
      name: "${name} errors"
    operation_mode:
      name: "${name} operation mode"
    operation_status:
      name: "${name} operation status"

```

syssi, 2023-03-18:
Pinout and isolated data cable example for the internal display port:

https://github.com/syssi/esphome-soyosource-gtn-virtual-meter/blob/main/docs/display-version/README.md

ButschFPV, 2023-03-18:
Nice work.
For me the connector looks like XH 2.54.

[7 more replies]