syssi/esphome-soyosource-gtn-virtual-meter #105: Wifi Dongle: SoyosourceDisplay CRC Check failed!

blacklopo, 2023-02-21:
Every few random minutes an following error message appeared in the log:  SoyosourceDisplay CRC Check failed! At the same time I can see error in HA Logbook (see screenshot).

Any advice what does it mean?
[logs_soyo-wifi-dongle_upload.txt](https://github.com/syssi/esphome-soyosource-gtn-virtual-meter/files/10795784/logs_soyo-wifi-dongle_upload.txt)
[image]

syssi, 2023-02-21:
Could you increase the log level to `DEBUG` and enable the debug output of the `uart` component?

```
uart:
  - id: uart0
    baud_rate: 9600
    tx_pin: ${tx_pin}
    rx_pin: ${rx_pin}
    debug:
```

I would like to see the raw frames at the logs.

blacklopo, 2023-02-21:
[18:13:37][I][soyosource_display:350]:   Maximum PV voltage: 45 V
[18:13:41][W][soyosource_display:108]: SoyosourceDisplay CRC Check failed! B6 != BE
[18:13:41][W][soyosource_display:108]: SoyosourceDisplay CRC Check failed! 43 != AE
[18:13:41][W][soyosource_display:108]: SoyosourceDisplay CRC Check failed! 10 != 37
[18:13:41][W][soyosource_display:108]: SoyosourceDisplay CRC Check failed! A7 != 37
[18:13:41][W][soyosource_display:108]: SoyosourceDisplay CRC Check failed! DF != 37
[18:13:41][W][soyosource_display:108]: SoyosourceDisplay CRC Check failed! 9B != BE
[18:13:41][W][soyosource_display:108]: SoyosourceDisplay CRC Check failed! 2B != 36
[18:13:41][W][soyosource_display:108]: SoyosourceDisplay CRC Check failed! 23 != 26
[18:13:41][W][soyosource_display:108]: SoyosourceDisplay CRC Check failed! C8 != BB
[18:13:41][W][soyosource_display:108]: SoyosourceDisplay CRC Check failed! 48 != BB
[18:13:41][W][soyosource_display:155]: Unhandled response received: A6.FE.37.B7.26.26.37.37.36.B7.36.26.26.B3 (14)
[18:13:42][W][soyosource_display:108]: SoyosourceDisplay CRC Check failed! 7F != 37
[18:13:42][W][soyosource_display:108]: SoyosourceDisplay CRC Check failed! F8 != B3
[18:13:42][W][soyosource_display:108]: SoyosourceDisplay CRC Check failed! 90 != 37
[18:13:42][W][soyosource_display:108]: SoyosourceDisplay CRC Check failed! BB != FF
[18:13:42][W][soyosource_display:108]: SoyosourceDisplay CRC Check failed! EF != 37
[18:13:42][W][soyosource_display:108]: SoyosourceDisplay CRC Check failed! 88 != FE
[18:13:42][W][soyosource_display:108]: SoyosourceDisplay CRC Check failed! 7B != 26
[18:13:43][E][soyosource_display:137]: No data from the Soyosource inverter received yet. This should never happen if the dongle is plugged into the inverter
[18:13:47][I][soyosource_display:163]: Status frame received
[18:13:47][I][soyosource_display:282]: Settings:

syssi, 2023-02-21:
I don't see any debug messages. :-) Did you update your ESP?

blacklopo, 2023-02-21:
My mistake. Wrong setup of Logger component. Better now?

18:34:16
[logs_soyo-wifi-dongle_upload-3.txt](https://github.com/syssi/esphome-soyosource-gtn-virtual-meter/files/10796573/logs_soyo-wifi-dongle_upload-3.txt)

syssi, 2023-02-21:
```
[18:34:16][D][uart_debug:114]: <<< 5A:01:93:40:D2:18:17:16:00:DF:64:46:37:06:1A:2D:07
[18:34:17][D][uart_debug:114]: >>> 55:01:00:00:00:00:00:00:00:00:00:FE
[18:34:18][W][soyosource_display:108]: SoyosourceDisplay CRC Check failed! 7B != 26
[18:34:19][W][soyosource_display:108]: SoyosourceDisplay CRC Check failed! 10 != 37
[18:34:19][W][soyosource_display:108]: SoyosourceDisplay CRC Check failed! 03 != 36
[18:34:19][D][uart_debug:114]: <<< 60:BF:B3:B7:36:26:26:F3:37:36:26:26:77:77:7E:37:B6:26:26:B3:37:36:66:33:A6:BB:26:37:37:36:B7:B6:26:26:B3:37:36:26:26:37:37:36:37:36:26:26:37:37:36:B7:B6:26:27:B3:37:36:26:26:37:37:36:37:33:36:37:36:26:26:B3:37:36:26:26:37:37:36:37:36:26:26:B3:37:36:E7:7B:A7:FB:26:7F:37:36:37:36:26:26:B3:37:36:A6:26:37:B6:36:37:B7:A6:26:37:37:36:37:36:26:26:B3:37:36:26:26:37:37:36:A6:33:36:37:36:26:26:B3:37:36:26:26:37:37:36:37:B6:26:26:37:37:36:B7:B6:26:26:B3:37
[18:34:19][W][soyosource_display:108]: SoyosourceDisplay CRC Check failed! C8 != BB
[18:34:19][W][soyosource_display:108]: SoyosourceDisplay CRC Check failed! 3F != BF
[18:34:20][D][uart_debug:114]: <<< 6E:66:37:36:66:B3:26:BB:A6:B7:BF:B6:37:36:26:AE:BB:37:36:26:66:26:BB:26:37:37:36:37:36:26:26:B3:37:36:26:26:37:37:36:37:FE:26:26:37:37:B6:37:36:26:AE:B3:B7:36:26:26:37:37:36:37:B6:66:26:37:37:36:B7:36:26:26:B3:37:36:26:A6:B7:B7:36:37:36:A6:6E:66:37:36:66:B3:26:BB:26:B7:B7:36:37:36:26:A6:B3:B7:36:26:76:26:BB:A6:B7:BF:B6:B7:B6:26:A6:B3:B7:36:A6:26:B7:37:36:37:EE:AE:A6:B7:BF:7E:66:37:36:66:33:26:BB:26:37:37:B6:37:36:A6:26:B3:37:36:26:66:26:BB:26:37
[18:34:20][W][soyosource_display:108]: SoyosourceDisplay CRC Check failed! 24 != 7F
[18:34:20][D][uart_debug:114]: <<< FF:66:66:7F:77:7E:B7:B6:26:27:B3:37:36:26:26:37:37:36:37:B6:26:26:37:37:B6:37:36:A6:26:B3:37:36:26:26:37:37:36:37:B6:26:66:37:77:7E:B7:B6:26:27:B3:37:B6:26:26:37:37:36:B7:FF:26:26:37:37:36:B7:B6:26:26:F3:37:36:26:26:37:37:36:37:36:66:26:37:37:36:37:B6:26:26:B3:37:36:26:26:37:37:36:B7:36:26:F6:66:37:36:66:33:A6:BB:26:37:37:36:37:36:26:26:B3:37:B6:26:76:A7:BB:26:7F:37:36:37:B6:26:26:F3:37:36:26:26:37:37:36:37:FF:26:26:37:37:36:FF:FF:66:27:B3:37:36
[18:34:20][W][soyosource_display:108]: SoyosourceDisplay CRC Check failed! 2B != AE
[18:34:21][W][soyosource_display:108]: SoyosourceDisplay CRC Check failed! FB != AE
[18:34:21][D][uart_debug:114]: <<< 26:37:37:36:37:EE:26:26:37:37:36:37:36:26:26:B3:37:36:26:26:37:37:36:B7:36:66:AE:B7:37:36:37:36:AE:AE:BB:37:36:26:26:37:37:36:37:36:AE:66:66:37:B6:36:B3:26:BB:BE:B7:B7:B6:37:36:A6:A6:B3:B7:36:AE:66:26:BB:AE:37:37:B6:37:36:AE:BE:BB:B7:36:26:AE:B7:37:B6:37:EE:AE:26:B7:BF:B6:37:36:BE:27:B3:B7:36:AE:A6:B7:BF:B6:37:36:66:AE:B7:BF:B6:37:36:BE:AE:B3:37:36:26:26:37:BF:B6:37:36:26:6E:66:BF:B6:66:33:26:BB:A6:B7:37:36:37:36:26:26:B3:37:36:AE:66:26:BB:A6:B7
[18:34:21][W][soyosource_display:108]: SoyosourceDisplay CRC Check failed! 67 != 37
[18:34:21][W][soyosource_display:108]: SoyosourceDisplay CRC Check failed! 40 != 26
[18:34:21][W][soyosource_display:108]: SoyosourceDisplay CRC Check failed! 9B != A6
[18:34:22][D][uart_debug:114]: <<< BF:B6:37:36:AE:A6:B3:B7:36:AE:26:37:37:36:37:EE:26:AE:B7:BF:B6:37:36:26:A6:B3:37:36:BE:BE:B7:B7:B6:37:36:66:26:37:BF:B6:37:36:AE:BE:B3:B7:36:A6:26:37:37:B6:37:36:26:76:66:B7:36:66:B3:26:BB:AE:B7:BF:36:37:36:A6:26:B3:37:36:26:66:26:BB:26:37:37:36:37:37:26:26:B3:B7:36:26:A6:37:37:36:37:FE:A6:26:37:37:36:37:36:AE:A6:B3:B7:36:AE:AE:B7:B7:B6:37:36:DE
[18:34:22][D][uart_debug:114]: >>> 55:01:00:00:00:00:00:00:00:00:00:FE
```

Could you provide another log tomorrow? This time there was some CRC errors because of a lot of random noise but the sensor wasn't updated with garbage. I would like to see one case where the `start voltage` changes to `0 V`.

syssi, 2023-02-21:
In other words I would like to see the RAW traffic which is causing this:

```
[16:21:10][I][soyosource_display:163]: Status frame received
[16:21:10][I][soyosource_display:282]: Settings:
[16:21:10][I][soyosource_display:291]:   Operation mode setting: 12
[16:21:10][I][soyosource_display:305]:   Device model: 0 W, 110 V (0)
[16:21:10][W][soyosource_display:583]:   Device type: Unknown (0)
[16:21:10][I][soyosource_display:308]:   Device type: Unknown
[16:21:10][I][soyosource_display:309]:   Battery voltage: 0 V
[16:21:10][I][soyosource_display:313]:   Start voltage: 0 V
[16:21:10][I][soyosource_display:319]:   Shutdown voltage: 0 V
[16:21:10][I][soyosource_display:333]:   Output power limit: 0 W
[16:21:10][I][soyosource_display:342]:   Start delay: 0 s
[16:21:10][I][soyosource_display:347]:   Minimum PV voltage: 0 V
[16:21:10][I][soyosource_display:350]:   Maximum PV voltage: 0 V
[16:21:15][I][soyosource_display:163]: Status frame received
[16:21:15][I][soyosource_display:282]: Settings:
[16:21:15][I][soyosource_display:291]:   Operation mode setting: 12
[16:21:15][I][soyosource_display:305]:   Device model: 1000 W, 220 V (210)
[16:21:15][I][soyosource_display:308]:   Device type: PV 26-56V / BAT 24V
[16:21:15][I][soyosource_display:309]:   Battery voltage: 24 V
[16:21:15][I][soyosource_display:313]:   Start voltage: 23 V
[16:21:15][I][soyosource_display:319]:   Shutdown voltage: 22 V
[16:21:15][I][soyosource_display:333]:   Output power limit: 700 W
[16:21:15][I][soyosource_display:342]:   Start delay: 6 s
[16:21:15][I][soyosource_display:347]:   Minimum PV voltage: 26 V
[16:21:15][I][soyosource_display:350]:   Maximum PV voltage: 45 V
```

blacklopo, 2023-02-21:
Here is one more log file. Full log in attachment. Tomorrow I can do another log. 
- Is there any way how to optimise scope of logging? 
- Is another way how to gather the log or I have to have esphome opened in browser and disable sleep mode?

PS> thanks for your proactive cooperation!

[19:31:53][D][sensor:126]: 'soyosource-wifi-dongle temperature': Sending state 39.00000 °C with 1 decimals of accuracy
[19:31:53][D][uart_debug:114]: <<< 5A:01:91:40:00:F6:00:1A:00:E2:32:00:37:04:E7:27:C0
[19:31:54][D][uart_debug:114]: >>> 55:03:00:00:00:00:00:00:00:00:00:FC
[19:31:55][D][uart_debug:114]: <<< 61:7F:B3:77:36:BE:BE:BB:B7:36:BE:BE:B7:BF:F6:37:36:BE:BE:BB:B7:B6:36:B3:36:BB:AE:B7:BF:B6:37:36:AE:AE:BB:B7:36:BE:FE:B7:BF:BE:B7:36:BE:BE:B7:BF:B6:37:36:BE:BE:BB:F7:36:AE:BE:B7:B7:B6:37:BB:B6:37:36:AE:BE:BB:B7:36:AE:AE:B7:BF:B6:37:36:BE:AE:BB:BF:B6:26:B3:26:BB:AE:B7:BF:B6:37:36:BE:BE:FB:B7:36:BE:AE:B7:36:BE:B7:36:26:AE:B7:BF:B6:37:36:BE:BE:BB:B7:36:BE:BE:B7:BF:F6:F6:BB:B6:37:36:AE:BE:BB:B7:F6:FE:FE:F7:FF:F6:77:F6:FE:FE:F7:BF:B6:37:36:A6:AE:B3:B7
[19:31:56][W][soyosource_display:108]: SoyosourceDisplay CRC Check failed! 37 != BF
[19:31:56][W][soyosource_display:108]: SoyosourceDisplay CRC Check failed! 40 != BB
[19:31:56][D][uart_debug:114]: <<< 66:66:BF:B6:66:B3:26:BB:BE:B7:BF:66:36:BF:B6:66:B3:26:BB:A6:37:37:B6:37:36:AE:A6:B3:B7:36:AE:66:26:BB:6E:36:BF:B6:66:33:26:BB:AE:B7:B7:B6:37:36:BB:37:36:AE:66:26:BB:AE:B7:37:36:37:36:A6:26:B3:37:36:26:26:37:37:B6:37:EE:AE:26:37:3F:36:37:36:26:26:B3:37:36:26:26:37:B7:B6:37:36:76:2E:37:B7:36:B7:B6:26:26:B3:37:36:2E:26:37:37:B6:37:36:26:66:66:37:36:66:33:26:BB:26:B7:37:36:37:36:26:26:B3:37:36:26:66:26:BB:A6:B7:BF:B6:37:36:26:26:B3:37:36:AE:AE:37:37
[19:31:56][W][soyosource_display:108]: SoyosourceDisplay CRC Check failed! F3 != 26
[19:31:57][D][uart_debug:114]: <<< 26:AE:BB:37:36:26:AE:B7:37:B6:37:36:26:36:66:37:36:66:33:26:BB:26:B7:37:B6:37:36:26:AE:BB:37:BF:26:33:A6:BB:26:37:BF:B6:37:36:26:26:B3:37:36:A6:26:26:26:B3:B7:36:26:26:37:37:36:37:36:26:F6:66:37:36:E6:33:A7:BB:26:37:37:36:37:36:26:26:B3:37:B7:26:33:26:BB:26:37:37:36:37:36:26:26:B3:37:36:A6:26:37:BF:36:37:36:26:26:7F:37:36:37:36:26:26:B3:37:36:AE:A6:37:BF:36:37:36:B3:37:36:26:26:B3:37:36:26:26:37:37:36:37:37:26:26:B3:37:37:E6:7B:A6:BB:26:37:37:36
[19:31:57][W][soyosource_display:108]: SoyosourceDisplay CRC Check failed! E7 != 37
[19:31:57][W][soyosource_display:108]: SoyosourceDisplay CRC Check failed! 47 != 37
[19:31:57][D][uart_debug:114]: <<< 26:66:26:BB:AE:B7:37:36:37:36:26:26:B3:37:36:26:26:37:37:36:37:FE:26:26:37:37:36:37:36:26:2E:B3:37:36:BE:AE:F7:37:36:37:36:66:BE:37:37:B6:37:36:AE:A6:B3:B7:36:BE:26:B7:B7:36:37:36:26:F7:66:37:36:66:33:26:BB:26:37:37:36:37:B6:26:26:B3:37:36:26:FE:26:BB:AE:37:37:B6:37:36:26:26:BB:B7:36:26:26:37:B7:36:37:FE:A6:26:37:37:36:37:36:26:26:B3:37:BF:66:26:37:37:36:37:36:66:26:B7:B7:B6:37:36:AE:AE:B3:37:36:26:26:37:37:36:B7:B6:26:7E:66:37:36:66:33:26:BB:26
[19:31:58][W][soyosource_display:108]: SoyosourceDisplay CRC Check failed! B3 != 36
[19:31:58][D][uart_debug:114]: <<< 77:37:36:37:36:A6:AE:BB:B7:36:26:66:26:BB:AE:37:37:36:37:36:2E:26:B3:B7:36:26:26:37:37:36:37:EE:26:26:37:37:36:37:36:26:27:B3:37:36:26:26:37:3F:36:37:36:76:26:37:37:36:37:36:26:26:B3:37:B6:26:26:37:37:36:37:36:26:66:E6:37:36:66:33:26:BB:26:77:37:36:B7:FF:26:26:F3:37:B7:26:F6:EF:BB:26:37:37:36:37:36:26:26:B3:3F:BF:66:66:37:37:7E:FF:FE:26:26:37:37:76:FF:36:26:26:B3:37:BE:26:26:37:37:36:37:B6:D6
[19:31:58][D][uart_debug:114]: >>> 55:01:00:00:00:00:00:00:00:00:00:FE
[19:31:58][E][soyosource_display:137]: No data from the Soyosource inverter received yet. This should never happen if the dongle is plugged into the inverter
[19:31:58][D][uart_debug:114]: <<< 5A:01:00:20:00:00:00:00:00:00:10:00:00:04:E7:00:E3
[19:32:02][D][uart_debug:114]: >>> 55:01:00:00:00:00:00:00:00:00:00:FE
[19:32:02][I][soyosource_display:163]: Status frame received

[19:27:23][D][uart_debug:114]: <<< 5A:01:91:40:00:F6:00:1B:00:E1:32:00:39:04:E6:27:BF
[19:27:24][D][uart_debug:114]: >>> 55:03:00:00:00:00:00:00:00:00:00:FC
[19:27:25][D][uart_debug:114]: <<< B5:7F:B3:37:36:FE:FE:FB:F7:76:BE:FE:F7:BF:B6:37:76:BE:BE:BB:B7:B6:26:B3:26:BB:AE:B7:B7:B6:37:36:BE:AE:BB:B7:36:FE:FE:F7:FF:FE:B7:36:93:37:36:AE:AE:FB:F7:36:AE:AE:F7:FF:F6:37:36:BE:BE:36:FF:F6:B6:F3:F6:FB:FE:B7:7F:E6:F7:F6:BE:AE:BB:B7:36:BE:FE:36:FB:FE:B7:BF:B6:37:36:BE:FE:FB:B7:36:BE:FE:B7:BF:B6:37:76:FE:BE:B7:BF:F6:37:36:BE:FE:FB:F7:F6:FE:FE:F7:FF:E6:F7:F6:FE:FE:F7:FF:B6:F7:76:BE:BE:FB:B7:F6:FE:BE:F7:7F:F6:F7:76:FE:FE:B7:BF:F6:F7:F6:BE:FE:FB:F7
[19:27:26][D][uart_debug:114]: <<< F7:F6:FE:FE:FB:F7:76:FE:E6:76:BB:BE:B7:BF:B6:37:36:BE:BE:BB:B7:36:FE:FE:F7:FF:F6:F7:66:FE:FE:B7:BF:B6:37:36:BE:BE:BB:B7:36:BE:FE:B7:BF:B6:37:36:66:A6:B7:BF:B6:37:36:A6:BE:BB:B7:F6:BE:AE:B7:BF:B6:37:F6:BE:66:36:BF:B6:26:B3:26:BB:BE:F7:BF:B6:77:36:BE:BE:BB:B7:36:BE:66:36:BB:BE:B7:BF:B6:37:36:26:AE:B3:B7:36:26:AE:B7:BF:B6:37:6E:BE:BE:B7:BF:B6:37:36:BE:BE:BB:B7:36:BE:AE:B7:BF:B6:37:F6:66:AE:B7:BF:B6:37:36:FE:BE:BB:B7:36:BE:BE:B7:BF:B6:37:36:BE:6E:36
[19:27:26][W][soyosource_display:108]: SoyosourceDisplay CRC Check failed! 20 != 36
[19:27:26][W][soyosource_display:108]: SoyosourceDisplay CRC Check failed! 0B != AE
[19:27:27][D][uart_debug:114]: <<< BF:B6:37:36:BE:A6:BB:B7:36:AE:AE:B7:B7:B6:37:36:6E:AE:B7:BF:B6:37:36:AE:BE:BB:B7:36:A6:AE:B7:BF:B6:37:36:A6:6E:26:BF:B6:36:B3:36:BF:B6:37:36:AE:A6:B3:B7:36:BE:A6:B7:BF:36:37:36:6E:AE:B7:B7:36:37:36:AE:AE:BF:B6:37:76:AE:AE:BB:B7:36:AE:26:37:37:B6:37:36:AE:36:66:37:B6:36:B3:26:BB:AE: [... cut ...]
[19:27:28][E][soyosource_display:137]: No data from the Soyosource inverter received yet. This should never happen if the dongle is plugged into the inverter
[19:27:28][D][uart_debug:114]: <<< 5A:01:00:20:00:00:00:00:00:00:10:00:00:04:E6:00:E4
[19:27:32][D][uart_debug:114]: >>> 55:01:00:00:00:00:00:00:00:00:00:FE
[19:27:32][I][soyosource_display:163]: Status frame received
[19:27:32][D][soyosource_display:171]: Unknown (raw): 01 (always 0x01?)

[logs_soyo-wifi-dongle_upload-4.txt.zip](https://github.com/syssi/esphome-soyosource-gtn-virtual-meter/files/10796989/logs_soyo-wifi-dongle_upload-4.txt.zip)

blacklopo, 2023-02-21:
> In other words I would like to see the RAW traffic which is causing this:
>
> ```
> [16:21:10][I][soyosource_display:163]: Status frame received
> [16:21:10][I][soyosource_display:282]: Settings: [...]

Sorry I missed this comment. Just for sure I can remove and re-add the device to HA. Just to be sure. I fully reinstalled it few weeks a go.

syssi, 2023-02-26:
Could you provide another log capture including the mentioned issue? Thanks in advance!

blacklopo, 2023-02-26:
Sorry to be late.
[soyo.yaml.zip](https://github.com/syssi/esphome-soyosource-gtn-virtual-meter/files/10834143/soyo.yaml.zip)
[logs_soyo-wifi-dongle_logs-3.txt.zip](https://github.com/syssi/esphome-soyosource-gtn-virtual-meter/files/10834145/logs_soyo-wifi-dongle_logs-3.txt.zip)

syssi, 2023-02-27:
I've pushed a small change into `main`. Please build & flash your Dongle (via OTA) again.

This is some background about the phenomenon: The WiFi Dongle consists of two main components the ESP8266 and a second microcontroller called MS51. The MS51 tracks the `total energy` and acts like an gateway between the Soyo and the ESP8266. Every now and then the MS51 crashs. This incoming traffic / garbage

```
[19:27:25][D][uart_debug:114]: <<< B5:7F:B3:37:36:FE:FE:FB:F7:76:BE:FE:F7:BF:B6:37:76:BE:BE:BB:B7:B6:26:B3:26:BB:AE:B7:B7:B6:37:36:BE:AE:BB:B7:36:FE:FE:F7:FF:FE:B7:36:93:37:36:AE:AE:FB:F7:36:AE:AE:F7:FF:F6:37:36:BE:BE:36:FF:F6:B6:F3:F6:FB:FE:B7:7F:E6:F7:F6:BE:AE:BB:B7:36:BE:FE:36:FB:FE:B7:BF:B6:37:36:BE:FE:FB:B7:36:BE:FE:B7:BF:B6:37:76:FE:BE:B7:BF:F6:37:36:BE:FE:FB:F7:F6:FE:FE:F7:FF:E6:F7:F6:FE:FE:F7:FF:B6:F7:76:BE:BE:FB:B7:F6:FE:BE:F7:7F:F6:F7:76:FE:FE:B7:BF:F6:F7:F6:BE:FE:FB:F7
[19:27:26][D][uart_debug:114]: <<< F7:F6:FE:FE:FB:F7:76:FE:E6:76:BB:BE:B7:BF:B6:37:36:BE:BE:BB:B7:36:FE:FE:F7:FF:F6:F7:66:FE:FE:B7:BF:B6:37:36:BE:BE:BB:B7:36:BE:FE:B7:BF:B6:37:36:66:A6:B7:BF:B6:37:36:A6:BE:BB:B7:F6:BE:AE:B7:BF:B6:37:F6:BE:66:36:BF:B6:26:B3:26:BB:BE:F7:BF:B6:77:36:BE:BE:BB:B7:36:BE:66:36:BB:BE:B7:BF:B6:37:36:26:AE:B3:B7:36:26:AE:B7:BF:B6:37:6E:BE:BE:B7:BF:B6:37:36:BE:BE:BB:B7:36:BE:AE:B7:BF:B6:37:F6:66:AE:B7:BF:B6:37:36:FE:BE:BB:B7:36:BE:BE:B7:BF:B6:37:36:BE:6E:36
```

is the output of the MS51 bootloader at a different baud rate while booting up. After the reboot the MS51 has to warm-up.  This means the MS51 must receive a `status` and a `settings` frame from the Soyo. If the ESP8266 asks the MS51 about the current status/settings between the "reboot" and the "warm-up" the MS51 responds with a frame full of zeros. The proof this assumption I've added some more logging and a safe-guard to reject "empty frames". This should stabilize the sensor measurements pushed to Home Assistant. The CRC errors will pop up constantly every time the MS51 reboots (for unknown and unfixable reasons).