syssi/esphome-soyosource-gtn-virtual-meter #107: Firmware 2022 801 can´t read nor write - controller in reboot-loop

klusou, 2023-03-06:
Hi,
I have the 1200w 48v Modell with display.
it seems like newer firmware doesn´t send any status codes. The "Esp-Soyosource-Controller" also doesn´t get values but the "rs485" shows in the display and I can send a target Power. So I think the Hardware is okay but because there are no values - this firmware crashes.
Is there anyone out there with a solution ?

Whats about this "displayport" Didn´t find any description to it.

Kind Regards 

David

syssi, 2023-03-07:
This project does work for silent inverters too in general. What do you mean by "this firmware crashs"?

syssi, 2023-03-07:
Do you use MQTT or the native `api`? Could you provide some logs of the crash?

klusou, 2023-03-07:
Okay Im getting the logs later because Im at work.
Testet both - mqtt und Homeassistant mode
in mqtt mode its a login - logout reboot -login loop
in Homeassistant mode it send 3 times something to serial that starts with 24 ...., then sets some defaults - waits till some timeout - would say 5 to 10 sec. then reboot - loop
I used the example yaml with and without display, because I don´t know what that display port thing stands for - The Soyosource with display or some device with an Serial Port for a Display ...

syssi, 2023-03-07:
I've pushed reduced YAML configuration examples:

https://github.com/syssi/esphome-soyosource-gtn-virtual-meter/blob/main/esp32-limiter-example.yaml
https://github.com/syssi/esphome-soyosource-gtn-virtual-meter/blob/main/esp8266-limiter-example.yaml

Please flash one of these examples and provide the logs / crashs later the day. I will try to get an idea what's going on.

klusou, 2023-03-07:
Hi,
I flashed my 2 boards with esp32 and esp8266 (D1 mini) and that settings;
Both stopped to reboot.
The ESP32 didn´t work with either the RS485 boards (SOYOSOURCE shows no RS485 in Display) and power-demand value has no effect.
The logs from ESP32 are:
```log
INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from soyosource-gtn-virtual-meter/debug
INFO Connected to MQTT broker!
[21:01:39][C][soyosource_inverter:152]:   State Class: 'measurement'
[21:01:49][I][mqtt:216]: MQTT Connected!
[21:01:49][I][app:062]: setup() finished successfully!
[21:01:49][I][app:102]: ESPHome version 2023.2.4 compiled on Mar  7 2023, 21:01:15
[21:01:49][I][app:104]: Project syssi.esphome-soyosource-gtn-virtual-meter version 2.0.0
[21:06:39][I][ota:113]: Boot seems successful, resetting boot loop counter.
```
The "old" D1 mini (8266) worked - RS485 in display of SOYOSOURCE and also with both RS485 Boards and power-demand value is working.
Maybe the Pinout is not all the same... 
Kind Regards

David

klusou, 2023-03-07:
https://cdn.shopify.com/s/files/1/1509/1638/files/ESP32_D1-Mini_480x480.png?v=1611046537
I used the same place as with the older d1 mini. Seems to me like GPO01 an GPO03.

syssi, 2023-03-08:
That should have worked. Please double check your RS485 converter and please make sure you didn't swap the TX/RX lines accidently. There are many devices that already come dead from the factory.

Does it work now for the ESP8266 and your setup is stable?

klusou, 2023-03-08:
Hi,
my ESP8266 is stable and solving my Problem. 
Because this 2 Boards have same Pinout with tx/rx 5v , GND and 3,3v I made 2 Boards with socket for the D1 mini with each of the RS485 Boards. So I can switch the ESP as I like.
That's why I'm pretty sure the RS485 are both running because they work with the older one.
Maybe Im going to try some other GPIO on the ESP32 - they seem to have a second RX/TX on GPIO16/17

syssi, 2023-03-08:
The ESP32 has a GPIO matrix / IO mux so the hardware UARTs can be routed to any pin. Feel free to attach the RS485 converter to any pin.

closed 2023-03-08 by syssi