syssi/esphome-soyosource-gtn-virtual-meter #109: Help setting things up needed

humpataa, 2023-03-07:
Hello, appreciate your work.
But I am having trouble in setting things up. I have put things together, connected to USB of my Mac. ESPHome installed successfully. Download GIT successfully. I am using NodeMCU board. So I have changed esp8266-example.yaml to use it instead of D1. I have also setup the secret file with correct wifi login.
I don't have MQTT, I don't have homeassistant, I don't want them. So I have commented out everything that looks like mqtt in esp8266-example.yaml. I will not be using the meter that came with the inverter either. I want to set the limiter by hand first, via JSON later but I understand that's way to go - so I just tried saving the example to the esp and monitor for now, "manual mode" if this is the right name for it.
But "esphome run" breaks with "Couldn't find ID 'powermeter0'. Please check ..." of course, because I don't have a power meter - I don't want it to use a power meter!?
Am I using the wrong yaml file? What is the best way for just a very simple basic to start with? No meter. No mqtt. No assistant. Just connect to the inverter and watch it working.
Any hint is greatly appreciated.

syssi, 2023-03-07:
Your usecase (no powermeter) isn't supported yet. I will push a quick fix asap to make the powermeter optional and provide a simple YAML. Do you use a ESP32 or ESP8266? You would like to use a small web page to control the device?

syssi, 2023-03-07:
Instead of making the powermeter optional I've prepared a workaround. This YAML looks like this:

https://github.com/syssi/esphome-soyosource-gtn-virtual-meter/blob/make-powermeter-optional/esp8266-example.yaml

I've added the `webserver` component. This adds a small web interface to the ESPHome node:

[image]

Please try to understand my changes here:

https://github.com/syssi/esphome-soyosource-gtn-virtual-meter/pull/110/files

And please be aware: I've changed the `uart` pins to GPIO4 and GPIO5 so you are able to receiving some log messages via GPIO1/GPIO3. In other words: Please attach the RS485 converter to GPIO4/GPIO5!

humpataa, 2023-03-07:
I will try this tomorrow, thank you for the quick reply & help!

humpataa, 2023-03-08:
Hmm. Sorry, first, I dont know how to try this new version. I understand the need to update my locally stored copy of your files, but when I use "git merge" it just says "Already up to date"?!

Second, I don't know what you mean with "GPIOx" ... I have connected the RS485 to Rx/Tx and power (see image: purple is Rx, blue is Tx, bound to RO / DI, black is GND, white is 3,3v).

[image]

syssi, 2023-03-08:
A clone of the repository isn't required. Just download and use this YAML configuration:

https://raw.githubusercontent.com/syssi/esphome-soyosource-gtn-virtual-meter/make-powermeter-optional/esp8266-example.yaml

Remove the wires from `RX/TX` and use `D1` (new RX) / `D2` (new TX) instead because of these lines:

https://github.com/syssi/esphome-soyosource-gtn-virtual-meter/blob/make-powermeter-optional/esp8266-example.yaml#L6-L7

That's all!

humpataa, 2023-03-08:
Thank you, looks like it works! :-)
It's spitting out a lot of log data and access via LAN works too, including adjusting power using the slider. However, the inverter does not respond in any way. Guess this is because I have not used the app to set it up? Problem is that I don't have a proper Android device to do this, will have to get one just for this, annoying. I can see the inverter's AP and its IP but cannot connect to it via browser. Chinese stupidity, the price for paying less.

Thoughts afaiu: your code sends log data via Serial back as well as to the webserver on ESP. Wouldn't it make sense to plug the ESP into the so called TTL-WIFI connector of the inverter via its USB-connector, read the data (similar to what the display shows) and even make settings to the machine? If this works it would get in conflict with your program using Serial to log things, wouldn't it?

syssi, 2023-03-08:
You are on the right track. This project provides a component/implementation of the TTL-WIFI protocol too. 

You could flash this YAML configuration to your ESP8266: https://raw.githubusercontent.com/syssi/esphome-soyosource-gtn-virtual-meter/main/esp8266-display-example.yaml

And attach GPIO4/GPIO5 (D1/D2) to the RX/TX lines of the TTL-WIFI USB port. This is the pinout of the USB connector:

https://github.com/syssi/esphome-soyosource-gtn-virtual-meter/tree/main/docs/wifi-dongle#usb-connector

This component can be used to control the Soyosource and change the settings. As last step will will build a combined solution. So you will attach the same ESP8266 to the TTL-WIFI port and the RS485:

https://github.com/syssi/esphome-soyosource-gtn-virtual-meter/blob/main/esp8266-display-limiter-example.yaml

Here we have to replace the powermeter again.

humpataa, 2023-03-08:
Very nice indeed, but what I actually thought was to use the micro-USB of ESP with an adapter cable to plug it into the USB-A of the inverter. I can connect the Rx/Tx of the inverter to D1/D2 as well, just need some cut off USB plug to get access to the TTL-WIFI connector. What I don't understand: why connect them to D1/D2, these are connected to RS485 already?! Or do you mean just for setup of the inverter, then connect it back to RS485 like it is now?

humpataa, 2023-03-08:
Update: after connecting a battery I see that the manual limiter works. Great stuff! But I seem to have a silent inverter - it shows zero responds. So I am not sure if it is actually in "battery limiter" mode or something else. The output is always 20W less than defined (a Shelly plug is measuring). The inverter itself seems to draw just 2W. Hmm?!

syssi, 2023-03-08:
>  Or do you mean just for setup of the inverter, then connect it back to RS485 like it is now?

Yes. I wanted to proceed step by step and combine all feature / solutions at the end.

- [ ] Be able to push power demands to the inverter using RS485
- [ ] Be able to read the status and control the settings using the TTL-WIFI port
- [ ] Combine both setups using different GPIOs per `uart`

Regarding the USB port of the inverter: Please keep in mind this isn't a real USB port. Never ever plug the ESP8266 using a Micro-USB cable into the inverter. It's just a connector with 4 pins (VCC, GND, RX, TX). We can use the port to power the ESP8266 and we can use the port to control the inverter like the WiFi dongle.

humpataa, 2023-03-08:
Understand. I have written esp8266-display-example.yaml to the ESP, it runs, log looks like this:

```
INFO Successfully uploaded program.
INFO Starting log output from /dev/cu.usbserial-0001 with baud rate 115200
[16:32:50]rll\x9c\x9e\x8f\x84$\xe0|

                                   \x8c\x80l\xec
                                                c|\x8f\x82\xe4\x92r\x93c\x8cp
                                                                             \x8c\xf2oN\x9f$NN\xdc\xe2\xec
                                                                                                          cp\xec\x8ecl rlp\xf2n\xe0

                                                                                                                                   \x83
                                                                                                                                       l\x84\x9c"
                                                                                                                                                 n\xe2|\xec
                                                                                                                                                           \x8e\x9c

                                                                                                                                                                   c\x8c\xf2NN\xef$\x84\x8fl`\x90NN\x8frlo{\x8f۲orr\x92 p\xf3n\xe0

                                                                                                                                                                                                                                  r\x84\x9c\x9c\x9c\x80\x84
                                                                                                                                                                                                                                                           b
                                                                                                                                                                                                                                                            o\xe2|
                                                                                                                                                                                                                                                                  \x8f"\x84\xf3nn\xee
         \x8el`\x90NN\x8erlo{\x8f\xb2\x9e{r$ s\x93\x9bo

                                                       rd \xdc
                                                              l\x8ex8c\x80l`\xfc\x82N\x9cl[I][logger:258]: Log initialized
[16:32:50][I][app:029]: Running through setup()...
[16:32:56][I][wifi:444]: - 'MYWLAN' (B0:F2:08:0F:E8:EC) ▂▄▆█
[16:32:56][I][wifi:257]: WiFi Connecting to 'MYWLAN'...
[16:32:57][I][wifi:518]: WiFi Connected!
[16:32:57][W][ota:102]: Last Boot was an unhandled reset, will proceed to safe mode in 8 restarts
[16:32:57][I][app:062]: setup() finished successfully!
[16:32:57][I][app:102]: ESPHome version 2023.2.4 compiled on Mar  8 2023, 16:32:35
[16:32:57][I][app:104]: Project syssi.esphome-soyosource-gtn-virtual-meter version 2.0.0
[16:32:57][W][ota:102]: Last Boot was an unhandled reset, will proceed to safe mode in 8 restarts
```

Hmm. Doesn't look that bad, the WLAN scanner finds it, but I cannot connect to it with the browser?

syssi, 2023-03-08:
The web interface isn't added. I will prepare a version with a web interface included. Wait a sec.

humpataa, 2023-03-08:
Thank you so much, here's some more log output:

```
[16:40:20][W][soyosource_display:108]: CRC Check failed! 4B != FF
[16:40:21][W][soyosource_display:108]: CRC Check failed! 00 != FC
[16:40:57][W][soyosource_display:108]: CRC Check failed! C8 != F7
[16:41:10][W][soyosource_display:108]: CRC Check failed! 90 != EA
[16:41:15][W][soyosource_display:108]: CRC Check failed! 6F != FF
[16:41:23][W][soyosource_display:108]: CRC Check failed! A0 != D2
[16:41:23][W][soyosource_display:108]: CRC Check failed! BC != FF
[16:41:30][W][soyosource_display:108]: CRC Check failed! 11 != BA
```

syssi, 2023-03-08:
Please use this YAML to talk to your TTL-WIFI port: https://raw.githubusercontent.com/syssi/esphome-soyosource-gtn-virtual-meter/make-powermeter-optional/esp8266-display-example.yaml

humpataa, 2023-03-08:
Thank you. Running. Something's going on, but doesn't look right:

```
[17:39:23][D][uart_debug:114]: >>> 55:01:00:00:00:00:00:00:00:00:00:FE
[17:39:24][W][soyosource_display:108]: CRC Check failed! A3 != FD
[17:39:25][D][uart_debug:114]: <<< FF:FF:FF:FB:FF:DF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FB:FF:FF:FB:FB:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:DF:FB:FF:F7:FF:FF:FF:FD:FF:FF:FF:FF:FF:FF:FF:FF:FE:FF:EE:FE:FE:FF:EF:FF:FF:FF:FF:FF:FF:FF:FF:F9:FF:FF:EF:FF:FF:FF:FF:FF:FF:FF:F7:FF:DF:FF:FC:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:E7:FF:F7:FF:FF:EF:FF:FD:FF:FF:FF:FF:FF:FF:FF:FF:FF:A6:8D:7E:91:00:01:09:00:00:00:D4:FC:EF:F7:FD:FB:FF:EF:FF:DD:FF:FF:FF:FF:FB:FF:FF:FF:FF:FF:F7:FF:F7:FF:FF:FF:FB:FE:FF:FF:FB:FF:FF:FF
[17:39:26][D][uart_debug:114]: <<< FF:EF:FF:FF:FF:FF:FF:FF:FF:FF:7F:FE:BF:FF:FF:FF:FF:FE:FF:F7:FF:EB:FF:EB:FF:EF:FF:FF:FF:FF:EF:FF:FF:FF:EE:FF:FF:FF:FB:7B:FF:FF:FF:FF:FF:FF:FF:FF:FE:FF:FF:EF:FE:FF:FF:FF:FF:FB:FE:EE:FF:FF:FF:FF:FF:FF:FF:EF:FF:FF:FF:FF:FF:FF:FF:F6:FF:FF:FD:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FE:FF:FF:FF:FF:FE:FF:FF:FF:FF:FF:F7:FF:EF:FF:FB:FF:FF:FF:FF:FF:FF:DF:FF:EE:FF:FF:FF:FE:FF:FF:FF:FF:FF:EF:FF:FF:FF:FF:FF:FF:FF:FF:FF:7F:FF:F7:FF:FF:FF:FF:FF:FB:FE:EF:FF:FF:ED:FF:FF:FF
[17:39:28][D][uart_debug:114]: <<< FF:FF:FF:FF:FD:FF:FF:FB:FF:F7:FF:DF:FF:FF:FF:FB:FF:FF:FF:FF:FF:DB:FF:FA:EF:EF:FF:FF:FF:FF:FF:FE:FF:FF:FF:FF:FF:FF:FF:FF:FF:FB:FF:FF:FF:FF:FF:F7:FF:FF:FF:FD:FF:FF:FE:EF:FF:EF:FF:FF:FF:FB:FF:F3:FF:FF:FF:E3:FF:FF:FF:FF:FF:FF:FF:FF:FF:FE:FD:FF:FD:FE:FF:FF:FF:FF:FF:FA:FF:FF:FF:FA:FB:FF:FE:FF:FF:FF:FF:FF:FF:FB:FF:FF:FF:7F:FF:DF:FF:FE:F7:FF:FB:EF:FF:FD:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FB:FF:FF:FF:EB:FF:FE:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FB:FF:F7
[17:39:28][D][uart_debug:114]: <<< FF:FE:FF:FC:FF:F7:FF:FB:FF:FB:FF:FF:FF:FF:FF:EF:FF:FF:FF:FD:FF:FF:FF:FF:F7:FF:FF:FF:EF:FF:FF:F7:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:7E:FF:FF:FF:FF:FF:FF:FF:EE:FF:FF:FF:FB:FF:FB:FF:EF:FF:FF:FF:FF:FF:FF:FE:FF:F7:FF:FF:FF:F7:FF:FF:FF:FF:FF:FB:FF
[17:39:28][D][uart_debug:114]: >>> 55:01:00:00:00:00:00:00:00:00:00:FE
[17:39:30][D][uart_debug:114]: <<< E9:FF:FE:FF:EF:FF:FB:FF:75:FF:FF:FF:FD:FF:FF:EF:FF:FA:FF:FF:FF:FE:FF:FF:FF:FE:FB:FF:FE:FF:FF:FF:FF:FF:FB:FF:FE:FF:FF:FF:FC:FF:FF:FF:EF:FF:FF:F9:FF:FF:FD:FF:FF:FF:FF:FF:FD:FE:FF:EF:FF:FF:FF:FF:FF:FA:FF:FD:FF:FF:FF:FE:FF:FE:FF:FF:FF:FE:FF:FF:EF:FD:F7:FF:FF:FF:FF:FF:FF:FF:FF:FF:FE:FF:FF:FF:FF:EF:FF:FF:FF:FF:FF:FF:FF:EF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:EF:FF:FF:FF:F7:FF:FF:FF:FF:FF:FE:FF:FE:FF:FE:FF:FF:FF:FF:FF:FF:FE:FF:FF:FE:FF:FB:FF:F7:FF:FF
[17:39:31][D][uart_debug:114]: <<< FF:FF:FF:FF:FF:EF:FF:F3:FF:FF:FF:FF:FF:FB:FF:FF:FF:FF:FF:FF:FB:FF:FF:FF:FE:FF:FF:FF:FF:FF:FF:FF:DF:FE:FE:FF:FF:FF:FF:FF:EB:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:F7:FF:FF:FD:FE:FF:FF:FF:F7:FF:FF:FD:FF:FF:FF:BF:FB:FF:FF:FF:FF:FF:FB:FF:FF:FF:FF:FF:FF:FF:FD:FF:FF:FF:FF:FF:FF:FF:FB:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:EF:FF:FF:FF:FF:FF:EF:FF:FF:FF:FB:FF:FF:FF:F3:FF:FF:FF:FF:FF:FE:FF:FF:FF:FF:FF:EF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FD:FF:F7:FF
[17:39:33][D][uart_debug:114]: <<< FF:FF:FF:F7:FF:FF:FF:FF:FD:FF:FF:FF:FF:FF:FF:FF:FF:FF:EF:FF:FF:FF:EF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FD:FF:FD:FF:7E:FF:F6:FF:FF:FF:FF:FB:FF:F7:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:E6:FF:FF:FF:FF:EF:FF:FF:F7:FF:FF:FF:FF:FF:FF:FF:F7:FF:FF:EE:FF:FF:F7:FF:FF:FF:FF:FF:FF:F6:FF:FF:FF:FF:DF:FF:FF:FF:FF:DB:FF:FF:FF:FF:FF:FF:EF:FF:FE:FF:FF:F7:FF:FF:FF:FF:FF:FF:EF:FF:FF:FF:FF:FF:FD:FF:FB:EF:FF:FE:FB:FF:FF:FF:FF:FF:FF:FF:EF:FF:F7:FF:FF:FF:FF:FF:FB:FF:FB:FF:FF
[17:39:33][D][uart_debug:114]: <<< FF:FF:FF:FF:FF:FE:FF:FF:FF:FF:FF:FF:FF:FF:FF:FD:FF:FF:FF:FF:FF:F7:F7:7D:FF:FF:FF:FF:FF:FF:FD:FF:FF:E6:FB:FF:FF:EF:FF:FF:FF:FF:FF:FF:FF:FF:BF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:EF:FF
```

I am sure the connectors Rx/Tx are correctly wired. I have used the stick to determine which to use from the "USB":

[image]

Changing them doesn't make a difference.

humpataa, 2023-03-08:
I see, maybe the cheap USB cable I used is not good. Will check if I can another with proper ground connector.

[image]

syssi, 2023-03-08:
This cannot work. GND is missing.

humpataa, 2023-03-08:
It's working now, thank you. Set it to battery limit mode. And it seemed to accept the power value, but somehow didn't change its output.

[image]

I changed back to the limiter version and it's working good. Except that the output is 20W lower than set - something to care about? Need to go now, more testing tomorrow, great work!!

syssi, 2023-03-08:
Probably setting the `output power limit` to `200W` isn't a good idea because this is the internal upper limit of the output power. If you ask the inverter to produce 900W now it would output 200W only.

The wrong `output power` (2364 W) is a bit odd. Especially because there wasn't a limiter attached.

Let's combine both YAMLs tomorrow and try to talk to two interfaces (WIFI-TTL & RS485) simultaneously.

syssi, 2023-03-08:
You can find the combined version incl. web interface here: https://raw.githubusercontent.com/syssi/esphome-soyosource-gtn-virtual-meter/make-powermeter-optional/esp8266-display-limiter-example.yaml

Please attach the RS485 converter to `GPIO14/GPIO12` (`D5/D6`) and the WIFI-TTL port to `GPIO4/GPIO5`:

```
substitutions:
  name: soyosource-gtn-display
  hostname: soyo
  device_description: "Monitor and control the Soyosource GTN via the display port and control the power output on demand via RS485"
  external_components_source: github://syssi/esphome-soyosource-gtn-virtual-meter@main
  tx_pin_ttl_wifi: GPIO4
  rx_pin_ttl_wifi: GPIO5
  tx_pin_rs485: GPIO14
  rx_pin_rs485: GPIO12
```

humpataa, 2023-03-09:
Works great, thanks a lot Sebastian!
Question: I understand the ESP needs to constantly send power limit otherwise output stops. Do you know the maximum time until the inverter falls back into zero output?
I see that everything is in your code to study ... but ... did you think about documenting what is known / you have found out about Soyo's protocols, both for limiter and display?

syssi, 2023-03-09:
I'm trying to improve the documentation but it consumes a lot of time. The project is pretty young and the user base small because of the complexity / barrier to entry of ESPHome. Feel free to ask more questions and/or create new issues. I'm happy to provide as much feedback as possible.

Out the timings: I've not much knowledge here. I'm aware of some experiments to get an idea how fast the inverter ramps up & down + some analysis about efficiency depending on the output power + the maximum interval of power demand updates (`2-3s`) but I haven't invested much time here yet.

humpataa, 2023-03-09:
I understand and highly appreciate your work on this subject. I am actually quite new to everything around ESP and happy that I have gotten things working in short time and without big trouble. Flashing ESP, using ESPHome, fumbling with GPIO connectors, command line stuff ... is not very easy and scares people. 

One more question: I see the web controls use POST requests to change settings, e.g.
`http://192.168.178.31/number/soyosource-gtn-display_output_power_limit/set?value=40`

Can this be easily changed to GET? This would make it VERY easy to control power output from remote.

syssi, 2023-03-09:
A `POST` is required. This cannot be changed easily but I could provide some console commands (`curl` commands) doing the job if you like.

syssi, 2023-03-10:
```
curl -X http://192.168.178.31/number/soyosource-gtn-display_output_power_limit/set?value=40
```

syssi, 2023-03-10:
Let's close this issue. Your setup is up and running in general.

closed 2023-03-10 by syssi