syssi/esphome-soyosource-gtn-virtual-meter #112: Change "power_demand_divider" and "buffer" value without reflashing

syssi, 2023-03-08:
### Discussed in https://github.com/syssi/esphome-soyosource-gtn-virtual-meter/discussions/111

<div type='discussions-op-text'>

<sup>Originally posted by **pfeupfeu** March  7, 2023</sup>
Since yersterday my setup is working! Now I would like to implement small improvements.

In winter, with little solar yield, only one of my 3 Soyos should run to get a better efficency. 
In summer, when there is "too much" solar yield, I would like to overfeed into the grid.

Therefore, I would like to be able to change the "power_demand_divider" and the "buffer" value without reflashing. 
Does anyone have a tip on how I can achieve this? </div>

syssi, 2023-03-08:
@pfeupfeu I've created an issue. This helps me to track the progress of the new feature.

syssi, 2023-03-08:
Please update your `number` component section to:

```
number:
  - platform: soyosource_virtual_meter
    soyosource_virtual_meter_id: virtualmeter0
    buffer:
      name: "${name} buffer"
      initial_value: 10
      restore_value: true
    manual_power_demand:
      name: "${name} manual power demand"
      max_value: 900
    max_power_demand:
      name: "${name} max power demand"
      initial_value: 600
      max_value: 900
      restore_value: true
    power_demand_divider:
      name: "${name} power demand divider"
      initial_value: 1
      restore_value: true
```

There are two new number entities available now if you are using the `main` branch of the external component.

closed 2023-03-08 by syssi

pfeupfeu, 2023-03-11:
Hello Sebastian,
thank you very much for this "high speed service"!
I have already installed the new version an the changes work perfectly!

syssi, 2023-03-11:
You are welcome! :-)