syssi/esphome-soyosource-gtn-virtual-meter #134: [W][soyosource_display:691]:   Operation mode: Unknown (9)

kev300, 2023-04-24:
Hi, my Soyosource seems to have another unknown operation mode.
When it's operational in bat limiter mode it reports:

```
[08:56:13][I][soyosource_display:232]: Status frame (Soyo, 15 bytes) received
[08:56:13][W][soyosource_display:691]:   Operation mode: Unknown (9)
[08:56:14][I][soyosource_display:383]: Settings (Soyo, 15 bytes):
[08:56:14][I][soyosource_display:396]:   Operation mode setting: 18 (0x12)
[08:56:14][W][soyosource_display.select:015]: Invalid value 18
[08:56:14][I][soyosource_display:409]:   Device model: 1200 W (212)
[08:56:14][I][soyosource_display:412]:   Device type: PV 55-90V / BAT 48V (48)
[08:56:14][I][soyosource_display:416]:   Start voltage: 45 V
[08:56:14][D][number:012]: 'soyo start voltage': Sending state 45.000000
[08:56:14][I][soyosource_display:422]:   Shutdown voltage: 43 V
[08:56:14][D][number:012]: 'soyo shutdown voltage': Sending state 43.000000
[08:56:14][I][soyosource_display:436]:   Output power limit: 900 W
[08:56:14][D][number:012]: 'soyo output power limit': Sending state 900.000000
[08:56:14][I][soyosource_display:442]:   Start delay: 3 s
```

It's a "new" one with purple board and a 2022 firmware (not sure wich one at the moment, since my display is not connected.
As you can see the operation mode selector is not working because of this.

Did you see that already?

syssi, 2023-04-24:
Interesting. I will have a look!

syssi, 2023-04-24:
Could you provide your YAML? I would like to make sure you are using the correct protocol version:

```
soyosource_display:
  - id: display0
    uart_id: uart_0
    protocol_version: SOYOSOURCE_DISPLAY_VERSION
    update_interval: 5s
```

syssi, 2023-04-24:
Are you able to sniff some display traffic? Do you have a new/other soyosource as in the past? I would like to know the operation mode values of the other operation modes and I would like to see a settings update frame of your display. 

It looks like the behaviour of your inverter is somehow between the display version firmware and the wifi version firmware.

kev300, 2023-04-24:
Well, I can't reproduce it anymore after I switched the settings via this esphome software.
I guess I need to reconnect the display and set it there.

But it's working fine with the software set battery limit mode atm. 

It's the same inverter but I fried it's cpu with a static discharge and got a replacement. 😱 
The replacement cpu was a never version. V5 instead of V3 and a new firmware 2022-(some end of year date).

kev300, 2023-04-25:
OK, it came back after the battery was empty.
So DC voltage low and than back to operation leads to the new, unknown, operation mode.

By sniffing you mean connecting the display and record the data via esp32?

syssi, 2023-04-25:
> By sniffing you mean connecting the display and record the data via esp32?

Yes. I would like to know the meaning of the new operation mode. It's probably decoded as "starting up..." or something else display at the display.

Perfunctory85, 2023-06-13:
Hey. Ive the same issue. After restart ive operation mode 9. @kev300 any news, how to fix it? That would be great

kev300, 2023-06-13:
The connectors are already in place. I Just need to connect, start sniffing and play around....
Will try to get it done this week. :)

kev300, 2023-06-14:
Here you go:
https://gadeco.de/s/2mGWM5sbTE54MHZ

I did not try to analyze and fix it myself yet, because you're probably faster anyway. :D
But if you don't have time for that, I can try.

Sorry for the uart debug in string format... I just copy&pasted it from esphome docs and saw it too late. ;)

Let me know if you want more information.

syssi, 2023-06-14:
Could you describe the difference between the `soyo_try_some_settings_restart2.txt` and `soyo_try_some_settings_restart_uart2.txt` capture?

kev300, 2023-06-14:
The uart files contains output from uart_debug. The others fro soyosource_display component. 
So probably redundant in most cases, but just to be sure.

syssi, 2023-06-15:
I've prepared a small change. Could you give it a try? @kev300 @Perfunctory85 

```
substitutions:
  external_components_source: github://syssi/esphome-soyosource-gtn-virtual-meter@add-operation-mode
```

Thanks in advance!

Perfunctory85, 2023-06-15:
Sounds great, i will try it in the afternoon.Sent from Android device

Thanks in advance!

kev300, 2023-06-15:
The error message is gone and it did set the right settings when I switched between PV, CP, and batt limit via HA, but the Soyo refused to startup.

I had to switch to CP and back to batt limit via the display to get it working again. 🤔

syssi, 2023-06-15:
Interesting. You could sniff the outgoing traffic of the display again and compare the command playload to the one of this project. I guess this line doesn't work for your Soyosource firmware version:

https://github.com/syssi/esphome-soyosource-gtn-virtual-meter/pull/139/files#diff-e269ecb66e30cd760f045eb52c6e0dd05681312b433a64957142daf8aeffec89L711

Perfunctory85, 2023-06-15:
It works for me. However, the two modes are reversed for me. When I set PV via HA, the Soyo switches to Battery Constant Power. If I switch to Battery Constant Power, the Soyo goes to PV mode.

syssi, 2023-06-15:
@Perfunctory85 Which Soyosource model (WiFi vs. Display version) do you own? Did you set the corresponding `protocol_version`?

https://github.com/syssi/esphome-soyosource-gtn-virtual-meter/blob/main/esp8266-display-display-version-example.yaml#L45

This is super important because it changes the protocol flavour a bit.

Perfunctory85, 2023-06-15:
`soyosource_display:
  - id: display0
    uart_id: uart_0
    protocol_version: SOYOSOURCE_DISPLAY_VERSION
    update_interval: 5s`

I own the Display Version. protocol_version is correct.  

How can i read the firmware? Only if i connect des display again?

syssi, 2023-06-15:
> How can i read the firmware? Only if i connect des display again?

Good question. I'm not aware of any payload containing the firmware version. You have to re-connect the display and navigate to the about page.

Perfunctory85, 2023-06-15:
oh thats a problem, because the display isnt working. That is one of the reasons, why i installed virtual display.

Perfunctory85, 2023-06-15:
when i set it in HA to PV the sensor of operation mode switches to battery constant power.

kev300, 2023-06-29:
Looks like the config examle is mixed up.
It works for me now, if I set my config like this:

```
select:
  - platform: soyosource_display
    operation_mode:
      name: "${name_soyo} operation mode"
      optionsmap:
        1: "PV"
        2: "Battery Constant Power"
        16: "Battery Limit"

I don't know why the inverter failed to startup initially, but now it all works. I can switch through all modes.

syssi, 2023-06-29:
Good catch! I've pushed a fix.

PR #139 (2023-06-29): Improve operation mode decoding
```
substitutions:
  external_components_source: github://syssi/esphome-soyosource-gtn-virtual-meter@add-operation-mode
```

Fixes #134
closed 2023-06-29 by syssi