syssi/esphome-soyosource-gtn-virtual-meter #135: Multiple soyosource inverters

ingoooooo, 2023-05-02:
Hello,

maybe this is a dumb question though I do not know how your esp32-multiple-uarts-example.yaml needs to be modified if three soyosource inverters should be controlled. Specifically the tx_pin, rx_pin part has to look then.

Any hints?

Thank you,
Ingo

ingoooooo, 2023-05-02:
As that could be one option I would like to try the RS485 bus approach first.

syssi, 2023-05-02:
Please try something like this:

```
substitutions:
  name: soyosource-gtn-virtual-meter
  device_description: "Monitor multiple Soyosource GTNs and control the power output on demand both via RS485"
  device0: firstfloor_soyosource_gtn_virtual_meter
  device1: groundfloor_soyosource_gtn_virtual_meter
  device2: secondfloor_soyosource_gtn_virtual_meter
  external_components_source: github://syssi/esphome-soyosource-gtn-virtual-meter@main
  tx_pin_uart_0: GPIO1
  rx_pin_uart_0: GPIO3
  tx_pin_uart_1: GPIO14
  rx_pin_uart_1: GPIO4
  tx_pin_uart_2: GPIO32
  rx_pin_uart_2: GPIO33

esphome:
  name: ${name}
  comment: ${device_description}
  project:
    name: "syssi.esphome-soyosource-gtn-virtual-meter"
    version: 2.1.0

esp32:
  board: esp-wrover-kit

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:

logger:
  baud_rate: 0
  level: INFO

# If you use Home Assistant please remove this `mqtt` section and uncomment the `api` component!
# The native API has many advantages over MQTT: https://esphome.io/components/api.html#advantages-over-mqtt
mqtt:
  broker: !secret mqtt_host
  username: !secret mqtt_username
  password: !secret mqtt_password
  id: mqtt_client

# api:

uart:
  - id: uart_0
    baud_rate: 4800
    tx_pin: ${tx_pin_uart_0}
    rx_pin: ${rx_pin_uart_0}
  - id: uart_1
    baud_rate: 4800
    tx_pin: ${tx_pin_uart_1}
    rx_pin: ${rx_pin_uart_1}
  - id: uart_2
    baud_rate: 4800
    tx_pin: ${tx_pin_uart_2}
    rx_pin: ${rx_pin_uart_2}

soyosource_modbus:
  - id: modbus0
    uart_id: uart_0
    # flow_control_pin: GPIO12
    # The name is used as prefix for some log messages and should
[... 266 lines cut ...]
    soyosource_inverter_id: inverter0
    operation_status:
      name: "${device0} operation status"
  - platform: soyosource_inverter
    soyosource_inverter_id: inverter1
    operation_status:
      name: "${device1} operation status"
  - platform: soyosource_inverter
    soyosource_inverter_id: inverter2
    operation_status:
      name: "${device2} operation status"

  - platform: soyosource_virtual_meter
    soyosource_virtual_meter_id: virtualmeter0
    operation_mode:
      name: "${device0} limiter operation mode"
  - platform: soyosource_virtual_meter
    soyosource_virtual_meter_id: virtualmeter1
    operation_mode:
      name: "${device1} limiter operation mode"
  - platform: soyosource_virtual_meter
    soyosource_virtual_meter_id: virtualmeter2
    operation_mode:
      name: "${device2} limiter operation mode"
```

ingoooooo, 2023-05-02:
Thank you very much.
But I am still curious where these numbers come from:
  tx_pin_uart_0: GPIO1
  rx_pin_uart_0: GPIO3
  tx_pin_uart_1: GPIO14
  rx_pin_uart_1: GPIO4
  tx_pin_uart_2: GPIO32
  rx_pin_uart_2: GPIO33

Can you give a little explanation if you do not mind?

syssi, 2023-05-02:
A ESP32 supports three hardware UARTs. Because of the GPIO Matrix / Pin mux the hardware UARTs can be assigned to any GPIO. In other words: You can choose any GPIO.

ingoooooo, 2023-05-02:
Does that mean that other pins need to be soldered compared to those shown on your photo?
https://github.com/syssi/esphome-soyosource-gtn-virtual-meter/blob/main/images/circuit.jpg

I am planning to upgrade from one inverter to three.

syssi, 2023-05-02:
There are two options:

1. Use the schematics above and attach all inverters to the same RS485. Set the divider at the configuration to the number of the attached inverters. As soon as there is an power demand all inverter will provide the power partially.
2. Attach three RS485 converter modules to three different sets of GPIOs and drive the inverters individually.

What's you goal?

ingoooooo, 2023-05-02:
Hello Sebastian,

okay, now I got it. I was confused as I thought there has to be a combination of divider AND GPIOs. Your last message made it clear. Thanks a lot.

I think the easiest way is option one. Obviously I won't have so much control over each inverter are there other disadvantages when using the divider?

ingoooooo, 2023-05-02:
Thank you very much for your help and your great project!

closed 2023-05-03 by syssi