syssi/esphome-soyosource-gtn-virtual-meter #136: Soyosource not responding

janrothkegel, 2023-05-28:
Hi,

I am currently trying to get my Soyosource integrated. It is the WiFi version and I linked it directly using a modified USB connector. Would be great if you could help :-)

My YAML (adapted from the examples section):

`substitutions:
  name: soyosource-gtn-display
  device_description: "Monitor and control the Soyosource GTN (WiFi version) using the display port"
  external_components_source: github://syssi/esphome-soyosource-gtn-virtual-meter@main
  # The display port should be connected to these GPIOs using a USB connector (VCC, RX, TX, GND)
  # A optocoupler / optical isolator (f.e. ADUM1201) between the display port and the ESP is recommended if the ESP isn't powered by the display port!
  tx_pin: GPIO4
  rx_pin: GPIO5

esphome:
  name: external-inverter
  friendly_name: external-inverter

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable Home Assistant API
api:
  encryption:
    key: "XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX"

ota:
  password: "XXXXXXXXXXXXXXXXXXXXXXXXXXXXXX"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "External-Inverter"
    password: "XXXXXXXXXX"

captive_portal:

logger:
  level: DEBUG
  logs:
    api.service: WARN
    ota: WARN
    sensor: DEBUG

external_components:
  - source: ${external_components_source}
    refresh: 0s

uart:
  - id: uart_0
    baud_rate: 9600
    tx_pin: ${tx_pin}
    rx_pin: ${rx_pin}
    debug:
      direction: BOTH

soyosource_display:
  - id: display0
    uart_id: uart_0
    protocol_version: SOYOSOURCE_WIFI_VERSION
    update_interval: 5s

binary_sensor:
  - platform: soyosource_display
[... 155 lines cut ...]
[07:21:08][C][soyosource_display.number:009]:   Unit of Measurement: 'V'
[07:21:08][C][soyosource_display.number:009]: SoyosourceDisplay Number 'soyosource-gtn-display output power limit'
[07:21:08][C][soyosource_display.number:009]:   Unit of Measurement: 'W'
[07:21:08][C][soyosource_display.number:009]: SoyosourceDisplay Number 'soyosource-gtn-display start delay'
[07:21:08][C][soyosource_display.number:009]:   Unit of Measurement: 's'
[07:21:08][C][soyosource_display.select:025]: soyosource_display.selectSoyosourceDisplay Select 'soyosource-gtn-display operation mode'
[07:21:08][C][soyosource_display.select:025]: soyosource_display.select  Icon: 'mdi:heart-pulse'
[07:21:08][C][soyosource_display.select:026]:   Select has register 10
[07:21:08][C][soyosource_display.select:027]:   Options are:
[07:21:08][C][soyosource_display.select:030]:     1: PV
[07:21:08][C][soyosource_display.select:030]:     2: Battery Constant Power
[07:21:08][C][soyosource_display.select:030]:     17: PV Limit
[07:21:08][C][soyosource_display.select:030]:     18: Battery Limit
[07:21:08][C][captive_portal:088]: Captive Portal:
[07:21:08][C][mdns:108]: mDNS:
[07:21:08][C][mdns:109]:   Hostname: external-inverter
[07:21:08][C][api:138]: API Server:
[07:21:08][C][api:139]:   Address: external-inverter.local:6053
[07:21:08][C][api:141]:   Using noise encryption: YES
[07:21:09][D][soyosource_display:495]: Send command: 55.01.00.00.00.00.00.00.00.00.00.FE (12)
[07:21:09][D][uart_debug:114]: >>> 55:01:00:00:00:00:00:00:00:00:00:FE
[07:21:14][D][soyosource_display:495]: Send command: 55.01.00.00.00.00.00.00.00.00.00.FE (12)
[07:21:14][D][uart_debug:114]: >>> 55:01:00:00:00:00:00:00:00:00:00:FE
[07:21:19][D][soyosource_display:495]: Send command: 55.01.00.00.00.00.00.00.00.00.00.FE (12)
[07:21:19][D][uart_debug:114]: >>> 55:01:00:00:00:00:00:00:00:00:00:FE

syssi, 2023-05-28:
How did you attach the ESP to the inverter in detail? Could you provide some photos?

janrothkegel, 2023-05-28:
Thanks for your fast response. I added two images showing the setup in gernal as well as one, which shows the pins used.
I also opened the Soyosource this morning and was able to find pairs of ESP32 pin and display plug inside the Soyosource with zero resistance --> I guess physical connection should be fine.

[image]
[image]

janrothkegel, 2023-05-28:
Is it possible to control the power using this setup? Or should I change to RS485 anyway?

syssi, 2023-05-28:
The USB / display port should be used to monitor and control (change settings) of the inverter just to replace the OEM WiFi dongle. As soon this is working you will extend your setup by an RS485 converter to control/drive the power output using the RS485 port.

syssi, 2023-05-28:
We should check the pinout of the USB connector carefully. If I remember correctly the USB cable colors aren't in line with the Soyosource USB port pinout. I will provide some details asap.

janrothkegel, 2023-05-28:
Ah ok, I see. Any idea regarding the not responding Soyosource so far?

syssi, 2023-05-28:
It looks like I was on the wrong track: https://github.com/syssi/esphome-soyosource-gtn-virtual-meter/tree/main/docs/wifi-version

Your wiring looks good. Could you swap the RX/TX line just for testing once?

janrothkegel, 2023-05-28:
You made my day! Thank you so much!

[image]

closed 2023-05-28 by janrothkegel

syssi, 2023-05-28:
You are welcome!