syssi/esphome-soyosource-gtn-virtual-meter #141: Dynamic power control over display/wifi interface

mb-software, 2023-06-22:
Hello,

I did my first test with a Soyo. Nice work! I already flashed my wifi dongle (wifi version) to esphome successfully. Now I would like to use dynamic power control for zero power feed.
Can this also be done via the TTL wifi dongle port? I mean, it works by setting the mode to constant battery power and adjusting the power output dynamically. But I guess this would kill the eeprom very quickly since the power output setting is stored in remanent memory, right?

Is there any way to control the power dynamically via the TTL port or do I really have to set up a second communication (RS485) to the soyo for this purpose?

syssi, 2023-06-22:
> But I guess this would kill the eeprom very quickly since the power output setting is stored in remanent memory, right?

Correct. This is the risk! Please attach a RS485 converter to the WiFi dongle and use a second UART to feed the official limiter interface.

mb-software, 2023-06-22:
OK thanks. Since I will be running out of gpio Pins in the original dongle then, does the wifi dongle TTL interface also work with only an esp8266/esp32? I mean without the additional chip that the original dongle has?

syssi, 2023-06-23:
There should be some free GPIOs left on the OEM dongle. If you search a bit at the issues & discussions you should find some photos of a extended OEM dongle. You should replace the OEM dongle at some point with a generic ESP8266 or ESP32 board. The firmware of the additional chip (microcontroller) doesn't do a good job.

If you take a look at the `docs` you will see a logic level converter (5V->3.3V) or and opto-coupler (if you don't power the ESP using the VCC pin of the display port) is recommend between the Soyosource and the ESP.

mb-software, 2023-06-23:
Alright, thank you. 
I have another small issue in EspHome, I think it's not worth opening its own topic here:
Is it normal with this component that the select and number entities cannot be adjusted via the EspHome webinterface? This is how it looks for me:
[image]

This is an example of how number entities look normally on my EspHome devices. (You can set the value by slider)
 
[image]

Is there something wrong in my EspHome config? It's only copy paste from the example. I have not seen this issue before on any of my esphome projects.

syssi, 2023-06-23:
Which ESPHome version do you use? I haven't seen such an issue before. There should be controls.

syssi, 2023-06-23:
Could you provide your YAML? I'm also confused because the "Operation Mode" text sensor is displayed twice.

mb-software, 2023-06-23:
```
esphome:
  name: soyosource-dongle
  friendly_name: soyosource-dongle

substitutions:
  external_components_source: github://syssi/esphome-soyosource-gtn-virtual-meter@main

esp8266:
  board: esp12e

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  

captive_portal:

web_server:

api:

ota:

logger:
  baud_rate: 0
  level: INFO

uart:
  - id: uart_0
    baud_rate: 9600
    tx_pin: GPIO1
    rx_pin: GPIO3

soyosource_display:
  - uart_id: uart_0
    protocol_version: SOYOSOURCE_WIFI_VERSION
    update_interval: 5s

binary_sensor:
  - platform: soyosource_display
    fan_running:
      name: "Fan running"
    limiter_connected:
      name: "Limiter connected"

button:
  - platform: soyosource_display
    restart:
      name: "Restart"

number:
  - platform: soyosource_display
    start_voltage:
      name: "Start Voltage"
    shutdown_voltage:
      name: "Shutdown Voltage"
    # Maximum output power in limiter mode / Output power in constant power mode
    output_power_limit:
      name: "Output Power Limit"
    start_delay:
      name: "Start Delay"

select:
  - platform: soyosource_display
    operation_mode:
      name: "Operation Mode"
      optionsmap:
        1: "PV"
        17: "PV Limit"
        2: "Battery Constant Power"
        18: "Battery Limit"

sensor:
  - platform: soyosource_display
    error_bitmask:
      name: "Error Bitmask"
    operation_mode_id:
      name: "Operation Mode ID"
    operation_status_id:
      name: "Operation Status ID"
    battery_voltage:
      name: "Battery Voltage"
    battery_current:
      name: "Battery Current"
    battery_power:
      name: "Battery Power"
    ac_voltage:
      name: "AC Voltage"
    ac_frequency:
      name: "AC Frequency"
    temperature:
      name: "Temperature"
    total_energy:
      name: "Total Energy"
    output_power:
      name: "Output Power"

text_sensor:
  - platform: soyosource_display
    errors:
      name: "Errors"
    operation_mode:
      name: "Operation Mode"
    operation_status:
      name: "Operation Status"
```

mb-software, 2023-06-23:
EspHome v2023.4.4.
I guess the second Operation Mode is the select entity.

syssi, 2023-06-24:
Please update your ESPHome install to a recent release. Try a clean build and flash your ESP another time. Clear your browser cache.

mb-software, 2023-06-24:
2023.4.4 is already the most recent release. I clean built, flashed 3 times and tried different browsers. Still the same. Very strange.

syssi, 2023-06-24:
2023.6.2 is the most recent release. I will try to reproduce your issue tomorrow using ESPHome 2023.4.4 and your YAML configuration above.

mb-software, 2023-06-24:
Yea you are right, sorry. I missed that on the EspHome changelog site. I will try to find some time to update my EspHome installation.

syssi, 2023-06-25:
I'm able to reproduce the issue also with the latest ESPHome release. Please add the `local: true` option to the `web_server` integration as workaround:

```
web_server:
  local: true
```

It looks like the external javascript isn't compatible at the moment or leads to strange issues of the UI.

closed 2023-06-25 by syssi