syssi/esphome-soyosource-gtn-virtual-meter #142: Tasmota electricity consumption meter?

thommy6102, 2023-07-06:
I`ve got an Soyo Inverter 48V without Display. Therfore I choose "esp8266-display-wifi-version-limiter-example.yaml", to flash on my D1 mini.
Must I  include some code to the "esp8266-display-wifi-version-limiter-example.yaml" file? And if so, how is it done?
With other words: How do I have to proceed to insert the code for querying my Tasmota electricity consumption meter into the yaml file?

Thanks for any help.

thommy6102, 2023-07-07:
Thanks a lot for your very quick help.
I`ve flashed your merged/combined file.. (Opened it with notepad++ and copied it to my ESPhome)
I`ve changed the  tasmota_status_url to that one what I have.

But I do not see a difference. What I`v e expected was to see the actual power consumption from my powermeter inside the web interface.
I don`t wanna use MQTT. So, Ididn`t configure it. 
The similar project (0 power consumtion by Bavarian Super Guy is running here since 6 month.
Were I`m wrong?

vy73 de DD1DZ from Bad Berleburg

thommy6102, 2023-07-07:
BTW: Is there any possibillity to accelarate the compiling/flashing process with ESPhome, inside Home Assistant?

syssi, 2023-07-07:
Could you open the `tasmota_status_url` in a browser and provide the JSON object here? I would like to verify the hierarchy and content.

73 de DO2KSM

syssi, 2023-07-07:
> BTW: Is there any possibillity to accelarate the compiling/flashing process with ESPhome, inside Home Assistant?

The short answer is "no". If you compile a ESPHome node a second time without cleaning the build folder it should be pretty fast because most of the parts can be reused.

thommy6102, 2023-07-07:
I used this url:
http://192.168.2.56/cm?cmnd=status+10

and it shows me:

StatusSMS:
    Time: "2023-07-07T09:21:48
SML

server_id: "0a01484c59020018a581"
export_total_kwh | 406.2979
total_kwh | 1608.8449
curr_w | -9
volt_p1 | 240.2
volt_p2 | 239.6
volt_p3 | 240.2
amp_p1 | 0.7
amp_p2 | 1.4
amp_p3 | 0.4
phase_angle_l2_l1 | 240
phase_angle_l3_l1 | 120
phase_angle_p1 | 221
phase_angle_p2 | 79
phase_angle_p3 | 50
freq | 50

thommy6102, 2023-07-07:
[image]

syssi, 2023-07-07:
This is the expected response (with german keys): https://github.com/syssi/esphome-soyosource-gtn-virtual-meter/blob/main/powermeter-sensor-examples/tasmota-http-status-sns.json

This part of the parser must be adapted:

```
# from
              - lambda: |-
                  std::string response_data = id(http_request0).get_string();
                  if(!response_data.empty()) {
                    json::parse_json(response_data, [](JsonObject root) {
                      id(powermeter0).publish_state(root["StatusSNS"]["SML"]["Leistung"]);
                    });
                  }

# to
              - lambda: |-
                  std::string response_data = id(http_request0).get_string();
                  if(!response_data.empty()) {
                    json::parse_json(response_data, [](JsonObject root) {
                      id(powermeter0).publish_state(root["StatusSNS"]["SML"]["curr_w"]);
                    });
                  }

syssi, 2023-07-07:
Could you provide the SML status response as raw data? I would like to add another powermeter example to the project.

thommy6102, 2023-07-07:
{
  "StatusSNS": {
    "Time": "2023-07-07T09:32:07",
    "SML": {
      "server_id": "0a01484c59020018a581",
      "export_total_kwh": 406.3138,
      "total_kwh": 1608.8451,
      "curr_w": -141,
      "volt_p1": 239.7,
      "volt_p2": 239.8,
      "volt_p3": 239,
      "amp_p1": 0.7,
      "amp_p2": 0.9,
      "amp_p3": 0.3,
      "phase_angle_l2_l1": 240,
      "phase_angle_l3_l1": 120,
      "phase_angle_p1": 233,
      "phase_angle_p2": 112,
      "phase_angle_p3": 306,
      "freq": 50
    }
  }
}

thommy6102, 2023-07-07:
[image]

thommy6102, 2023-07-07:
1.
https://github.com/syssi/esphome-soyosource-gtn-virtual-meter/pull/144
Sri, I don`t understand.

2.
The exchange from "Leistung" to "cur_w"  inside the string was it. Now it`s running.
[image]

Many thanks for your effort to me and your Hamspirit.

Dem Funkamateur ist nichts zu schwör!

Last question from my side.
I`ve often have tp press the F5 button im my Firefox Browser to keep the page up to date. Is it normal?

syssi, 2023-07-07:
1. It's just a notification about two new files: https://github.com/syssi/esphome-soyosource-gtn-virtual-meter/pull/144/files
2. Looks good to me!
3. The small website of the `web_server` component uses a tiny javascript implementation to subscribe to a event stream using websockets to be able to update the sensor measurements instantaneously. A refresh of the page isn't required in general.

thommy6102, 2023-07-07:
OK, thanks again for your great job.
Problem is fixed now. I`m going to close this issue.

BTW: I`m using a "Hitchi IR Reader" connected to Node MCU 8266 with Tasmota. Maybe it`s helpfull for other users.

73 de Thomas

closed 2023-07-07 by thommy6102

syssi, 2023-07-07:
May be you would like to transform the device to a ESPHome node some day. This is the required component: https://esphome.io/components/sml.html