syssi/esphome-soyosource-gtn-virtual-meter #149: The inverter didn't respond to the last 15 status

WZBFL, 2023-08-05:
Hi,
i used this software to control the soyosource 1200W inverter.
controlling function is working, but i get no values from the inverter itself. I only get this error:

[image]

the fw of inverter is: ST08-22-1025 -> maybe this is the problem? Or is it already tested? The Inverter is the model: with display, without wifi.

Thank you in advance

Chickenbreast0, 2023-08-05:
Hi,

maybe your Inverter is a silent one. 
Mine has the FW 2021-301, blue PCB, non Wifi but LCD, but I can't see the fan running.

WZBFL, 2023-08-05:
Yeah, i think also. But why it is silent?

syssi, 2023-08-06:
The root cause for silent devices in unknown. Some ideas:

1. It's a hardware flaw and the internal RS485 comes broken (unable to transmit) from factory.
2. The firmware was changed and the status the frame was removed.
3. The request has changed and the current status request frame isn't accepted anymore.

WZBFL, 2023-08-06:
Maybe yes. If you have some ideas I can check it on my inverter. I can try it with Arduino and can give you feedback on the results. Limiter is working perfectly.

syssi, 2023-08-09:
There isn't much we can do here. IMO you should stop here.

I will try to talk to the STM32 directly (UART -> RX/TX) some day without using the internal RS485 converter to rule out the converter as root cause.

Please remove the `soyosource_inverter` section and all sensors using this platform to get rid of the warning. The `soyosource_inverter` component is used to decode the status frame with isn't available in your case.

kev300, 2023-08-17:
> I will try to talk to the STM32 directly (UART -> RX/TX) some day without using the internal RS485 converter to rule out the converter as root cause.

You mean via the display connection? Do you think it's possible to transmit demand commands via that way or do you intend to update the constant power setting?

syssi, 2023-08-17:
@kev300 No. ;-) Long long ago it was sufficient to talk to the limiter port using RS485. You could transmit the power demand here and ask for a inverter status periodically. Newer devices doesn't respond to status requests anymore. The root cause is unknown. To rule out the internal RS485 converter as root cause we could try to talk to the STM32 directly.

To more explicit. The experiment would look like this:

1. Open the inverter
2. Remove the interal RS485 daughter board
3. Connect a ADUM to the RX/TX pins instead
4. Make sure to pull-up/pull-down the third wire (purpose unknown)
5. Attach a ESP to the ADUM and send the `status request` periodically. Does the STM32 respond?

kev300, 2023-08-17:
> @kev300 No. ;-) Long long ago it was sufficient to talk to the limiter port using RS485. You could transmit the power demand here and ask for a inverter status periodically. Newer devices doesn't [...]

Thanks, got it!

closed 2024-02-03 by syssi