syssi/esphome-soyosource-gtn-virtual-meter #151: Invalid header: 0x40 0x20 0xC0 0xC

WZBFL, 2023-08-16:
A lot of error Messages here:

[image]

syssi, 2023-08-16:
Could you describe your setup and provide another log (with `uart` debugging enabled: https://github.com/syssi/esphome-soyosource-gtn-virtual-meter#debugging) in plaintext?

WZBFL, 2023-08-16:
no Problem:

[image]

syssi, 2023-08-16:
Could you describe the connection/wires between the ESP and the limiter port of your soyosource? How long is the cable? Which RS485 converter module do you use? Is there something else connected to the bus? A photo would be nice. For now it looks like there is some noise/garbage on the bus.

closed 2024-08-12 by syssi