syssi/esphome-soyosource-gtn-virtual-meter #153: no power value for virtualmeter

wasn-eu, 2023-09-05:
i have installed the esp8266-display-display-version-limiter-example.yaml on an esp8266 and connected it to the rs485 and display port.
the display port functions are ok, but why do it not see a value for sensor.soyodisplay_soyosource_gtn_display_power_demand?
i have set the right sensor in the yaml file and i am using the api.

this is what i see in the esphome logs:

`[22:28:58][I][soyosource_display:235]: Status frame (Soyo, 15 bytes) received
[22:28:58][I][soyosource_display:386]: Settings (Soyo, 15 bytes):
[22:28:58][I][soyosource_display:399]:   Operation mode setting: 16 (0x10)
[22:28:58][I][soyosource_display:412]:   Device model: 1000 W (210)
[22:28:58][I][soyosource_display:415]:   Device type: PV 26-56V / BAT 24V (24)
[22:28:58][I][soyosource_display:419]:   Start voltage: 25 V
[22:28:58][I][soyosource_display:425]:   Shutdown voltage: 23 V
[22:28:58][I][soyosource_display:439]:   Output power limit: 50 W
[22:28:58][I][soyosource_display:445]:   Start delay: 6 s`

wasn-eu, 2023-09-06:
i have changed the sensor for reporting the output demand and now it is working

closed 2023-09-06 by wasn-eu

syssi, 2023-09-06:
Good job!