syssi/esphome-soyosource-gtn-virtual-meter #154: Add restore mode for the operation mode (display version)?

Kobbe1, 2023-09-16:
Hi there,
great project!

I think the above option misses the restore function, so when reboting it seems to default to battery for me. Might be the inverter loosing its setting too of course, maybe someone else knows.

I can easily fix it and set it through HA automation but its not exactly elegant.

So consider this fairly low priority ...

Thanks!

-T

syssi, 2023-09-17:
This is strange because the inverter is responsible for storing the operation mode. It get's retrieved from the inverter if the ESP reboots. A ESP reboot shouldn't trigger a settings update on the inverter.

Chickenbreast0, 2023-09-17:
@Kobbe1 

For me your issue sounds like a damaged CPU Board. It is pretty common, that your Soyo 'forgets' it's settings, if the cpu board is damaged.

Kobbe1, 2023-09-18:
Thanks. You are right, this is something the ESP should read from the inverter. It is already a replacement board as the inverter does funny things above 700W (pretty much constantly cycles between startup and standby), so maybe that caused it. As it works otherwise and only happens very seldomly (the settings I mean) I won't do anything about it.

Chickenbreast0, 2023-09-18:
Can you post your *.yaml code here and maybe some logs?
Otherwise we are diggin' in a deep, dark hole ;) Cheers

Kobbe1, 2023-09-18:
Sure. This is a silent soyo, running with a modbus limiter. I added a temperature sensor and a PWM fan (it is the light in the yaml)
All works so far apart from the 700W limit (this is unrelated to limiter - it also happens when the inverter runs on its own), and sometimes the status values are not reported for some hours. Might be heat related, I think I stupidly used hotglue to hold some of the display wires ...
And yes, wifi is weak, but it works suprisingly well.
```
substitutions:
  name: soyosource-gtn-display
  device_description: "Monitor and control the Soyosource GTN (display version) via the display port and control the power output on demand via RS485"
  external_components_source: github://syssi/esphome-soyosource-gtn-virtual-meter@main
  tx_pin_ttl_wifi: GPIO15
  rx_pin_ttl_wifi: GPIO13
  tx_pin_rs485: GPIO1
  rx_pin_rs485: GPIO3

output:
  - platform: esp8266_pwm
    pin: D4
    frequency: 1000 Hz
    id: pwm_output
    zero_means_zero: true 
    min_power: 0.1

light:
  - platform: monochromatic
    output: pwm_output
    name: "Extra fan"
    id: extra_fan
    restore_mode: RESTORE_DEFAULT_ON 

esphome:
  name: ${name}
  comment: ${device_description}
  project:
    name: "syssi.esphome-soyosource-gtn-virtual-meter"
    version: 2.1.2

esp8266:
  board: nodemcuv2
  restore_from_flash: true
external_components:
  - source: ${external_components_source}
 #   refresh: 0s

wifi:

  networks:
  - ssid: !secret wifi_ssid
    password: !secret wifi_password
  use_address: 192.168.178.xxx
  
  ap:
    ssid: "xxx"
    password: "xxx"

ota:
  safe_mode: true

logger:
  level: DEBUG
  baud_rate: 0

api:

captive_portal:
dallas:
  - pin: GPIO14
    update_interval: 60s
uart:
  - id: uart_0
    baud_rate: 9600
    tx_pin: ${tx_pin_ttl_wifi}
[... 456 lines cut ...]
[22:53:56][D][sensor:109]: 'soyosource-gtn-display error bitmask': Sending state 2.00000  with 0 decimals of accuracy
[22:53:56][D][text_sensor:064]: 'soyosource-gtn-display errors': Sending state 'DC voltage too low'
[22:53:56][D][soyosource_display:514]: Send command: 55.03.00.00.00.FC (6)
[22:53:56][I][soyosource_display:386]: Settings (Soyo, 15 bytes):
[22:53:56][D][soyosource_display:387]:   A6.02.8A.63.42.D2.18.1E.16.00.DE.64.5A.03.11 (15)
[22:53:56][I][soyosource_display:399]:   Operation mode setting: 1 (0x01)
[22:53:56][D][select:015]: 'soyosource-gtn-display operation mode': Sending state PV (index 0)
[22:53:56][I][soyosource_display:412]:   Device model: 1000 W (210)
[22:53:56][I][soyosource_display:415]:   Device type: PV 26-56V / BAT 24V (24)
[22:53:56][I][soyosource_display:419]:   Start voltage: 30 V
[22:53:56][D][number:012]: 'soyosource-gtn-display start voltage': Sending state 30.000000
[22:53:56][I][soyosource_display:425]:   Shutdown voltage: 22 V
[22:53:56][D][number:012]: 'soyosource-gtn-display shutdown voltage': Sending state 22.000000
[22:53:56][I][soyosource_display:439]:   Output power limit: 900 W
[22:53:56][D][number:012]: 'soyosource-gtn-display output power limit': Sending state 900.000000
[22:53:56][I][soyosource_display:445]:   Start delay: 3 s
[22:53:56][D][number:012]: 'soyosource-gtn-display start delay': Sending state 3.000000
[22:53:58][D][soyosource_virtual_meter:072]: 'firstfloor_inverter': Setting the limiter to 650 watts per inverter (650 in total)
[22:53:58][D][sensor:109]: 'soyosource-gtn-display power demand': Sending state 650.00000 W with 0 decimals of accuracy
[22:53:58][D][text_sensor:064]: 'soyosource-gtn-display limiter operation mode': Sending state 'Manual'
[22:53:58][D][sensor:109]: 'soyosource-gtn-display smartmeter instantaneous power': Sending state nan  with 1 decimals of accuracy
[22:53:59][D][sensor:109]: 'Uptime Sensor': Sending state 37731.44531 s with 0 decimals of accuracy
[22:53:59][D][dallas.sensor:143]: 'DS18B20 temp': Got Temperature=19.2°C
[22:53:59][D][sensor:109]: 'DS18B20 temp': Sending state 19.25000 °C with 1 decimals of accuracy
```

kev300, 2023-09-19:
You can set the settings yourself via the esp directly, if you want (better then HA automation)
Here's my examle in the interval section (starting from line 62):
https://github.com/kev300/HA_with_EspHome/blob/main/esp-home_config/garage.yaml

syssi, 2023-11-07:
Is this issue resolved? In my opinion there is no room for improvements.

Kobbe1, 2023-11-08:
Yes, it did not happen anymore with above workaround.

closed 2023-11-08 by Kobbe1