syssi/esphome-soyosource-gtn-virtual-meter #159: Using without RS485 board

poky, 2023-11-30:
Hi, I have a display version 24V 1000W, FW STC8-23-0130 with not RS485 board.
However, the RS485 header is there (5pin with tx,rx, 5V, GND), is it possible to get it to communicate directly without the RS485?
Could anyone direct me how to do it? Thanks!

syssi, 2023-11-30:
Is my understanding correctly: These components (RS485 converter and connector board) are missing?

[image]

poky, 2023-11-30:
Yes, Correct!
[image]

syssi, 2023-11-30:
Please don't attach a ESP directly. You should use a digital isolator module (ADUM1201). The setup will look like this:

https://github.com/syssi/esphome-soyosource-gtn-virtual-meter/blob/main/docs/display-version/isolated-data-cable.jpg

Without the JST connector. Don't expect a response if you send instructions (a power demand).

poky, 2023-11-30:
> Please don't attach a ESP directly. You should use a digital isolator module (ADUM1201). The setup will look like this: [...]

I currently don't have ADUM lying around, can I get away with it if power from the 5V to VIN, and 1k resistor on both RX, TX?

poky, 2023-12-01:
> It's normal if your firmware / model doesn't respond to status requests on the RS485 bus. In limiter mode the Soyosource should respect the power demands transmitted using RS485 nevertheless.

Yes, I can see the blue RS485 icon shows up when sending the power demands, so I think it works fine directly without the messy rs485 conversion.

[image]

closed 2023-12-01 by poky