syssi/esphome-soyosource-gtn-virtual-meter #161: Sorry but a little bit confused...

ericgrand, 2024-01-03:
Thanks for your great job! I will order a GTN 1000 shortly which most probably will arrive with newer firmware/PCB, so not able to communicate status through RS 485 port.
In your files, there are some description to connes a WEMOS to the display port or the WIFI TTL port.
Does it means the status and power control could be achieved through these connections instead of the RS 485 port?

Thanks in advance!

Eric

syssi, 2024-01-03:
You have to connect/use both. The display port / WiFi dongle port is used to retrieve the current status and change the inverter settings. The RS485 port must be used to request a power demand. The retrieval of the status frame via the RS485 port isn't possible anymore. The RS485 will accept/receive instructions but won't respond/transmit.

ericgrand, 2024-01-03:
Thanks for your prompt feedback.
Ok, so I will have to connect 2 devices, one for each function?
Which yaml files should be used in such case? 
Could the same ESP 32 be used for both connection?
And last, which version is preferred: wifi or display?

syssi, 2024-01-03:
You can handle both jobs with the same ESP using two different UARTs / sets of GPIOs. A ESP32 supports three hardware UARTs. A ESP8266 has a single hardware UART. The second one will be software emulated using SoftwareSerial automatically. If you have the choice pick a ESP32 for improved stability.

syssi, 2024-01-03:
I've no preference (with or without display). The display dies after a while pretty often. May be the WiFi version is the better choice at the moment.

ericgrand, 2024-01-03:
Thanks, I think now I have to figure how to code this under esphome! Still discovering this!

syssi, 2024-01-03:
There is an example YAML configuration per setup at the root directory of this project. Feel free to ask more questions if they aren't self explaining.

ericgrand, 2024-01-03:
Yes, I was just checking the examples and I believe the one I need is this one: esp8266-display-display-version-limiter-example.yaml for LCD version of the inverter.
I suppose it is easy to replace the ESP8266 by an ESP 32, just changing the GPIO and board type? 
I suspect there is no examples for ESP32 with dual interfaces?

syssi, 2024-01-03:
> I suppose it is easy to replace the ESP8266 by an ESP 32, just changing the GPIO and board type?

Correct. For example:

```
# remove
esp8266:
  board: d1_mini

# add
esp32:
  board: esp-wrover-kit
```

And adapt the GPIOs slightly: 

```
  tx_pin_ttl_wifi: GPIO4
  rx_pin_ttl_wifi: GPIO5
  tx_pin_rs485: GPIO14
  rx_pin_rs485: GPIO12
```

ericgrand, 2024-01-04:
Perfect! Many thanks!

closed 2024-01-04 by syssi