syssi/esphome-soyosource-gtn-virtual-meter #165: Inverter for 36v battery

marine1988, 2024-01-26:
Hello o have a bunch of 36v lithio batteries and wnat to make usefulk on my solar system. I saw this inverter grid tie with limiter that are perfect except for the fact that is wired limited an my inverter is away from the total of my grid. I have home assistant and shelly 3pm values can I use it to limit my inverter? Sorry I'm noob thank you

card917, 2024-01-26:
yes you can do it if your consumption sensor exposed to ha. you can put the sensor name in powermeter0 section in the code. make sure that the unit is in W. mine is in kW then i have to put a filter to multiply by 1000. then it works just fine.

syssi, 2024-01-27:
Please keep in mind the Soyosource can be used as photovoltaics inverter **or** battery inverter only. You cannot use the inverter to charge your batteries!

marine1988, 2024-02-08:
Yes i know i bought 1 mppt charge to charge batteries i bought soyosource with wifi version seller say it only can have wifi or screen not both so I prefed wifi now i installed the code in esp home logs show is working but i have no information about the inverter whet do I need to do now?

marine1988, 2024-02-29:
@syssi can I send a pm to you please I need some help with my configuration 
[image]

syssi, 2024-03-01:
Could you try to upload your image another time?

marine1988, 2024-03-02:
yes i can
[image]

syssi, 2024-08-12:
Do you still have questions? :-) Are you happy with your setup and can we close this issue?