syssi/esphome-soyosource-gtn-virtual-meter #171: Defekt am Soyo?

pogo01, 2024-04-04:
Moin.
Bis vor ein paar Tagen lief alles super. Dann hatte ich leider Ausfälle bei dem Input. Nun steht unten limiter mode    inactivity timeout. Kann es sein, dass die cpu des soyo durch ist? Hatte jemand schon einen ähnlichen Vorfall?

[image]

syssi, 2024-04-04:
Es bedeutet lediglich, dass der ESP keinen aktuellen Verbrauch vom Stromzähler kommuniziert bekommt. Wenn du in den manuellen Modus wechselst, kannst du dann wieder Strom einspeisen?

pogo01, 2024-04-07:
Okay.
Im Moment helfe ich mir mit Automationen, weil die Kommunikation wohl nicht so funzt. Dann könnte es das RS485 Modul sein....?
Kann ich dann direkt auf Board gehen? Wie hier:
https://github.com/PepeTheFroggie/Soyosource-GridTie-inverter-24V-LCD-replacement
Oder baue ich etwas wie hier?:
https://www.youtube.com/watch?v=SCB1CiCBdd0

syssi, 2024-04-07:
Ich habe den Eindruck, dass du am falschen Ende suchst. Wenn du in Home Assistant den manuellen Modus aktivierst scheint der Wechselrichter sauber einzuspeisen. Der Komponenten Wechselrichter und Lesen/Schreiben durch den ESP scheinen in Ordnung. Auch im manuellen Modus sendet der ESP periodisch die gewünschte Leistung via RS485. In deiner YAML-Konfiguration ist definiert, wie häufig der ESP eine neue Messung vom Stromzähler benötigt bevor er ins "Inactivity timeout" fällt. Es handelt sich hier um einen Schutzmechanismus der verhindern soll, dass Leistung eingespeist wird, obwohl unbekannt ist, wieviel gerade benötigt wird. Angenommen dein Stromzähler meldet aktuellen Bedarf nur alle 10 Sekunden, dann solltest du das Timeout anheben.

Kannst du ein ESPHome Log zur Verfügung stellen? Hier wird man ablesen können, ob die Messungen des Stromzählers überhaupt eingehen und wann die Anwendung ins Timeout fällt.

pogo01, 2024-04-07:
Danke für die schnellen Antworten
Ich stolpere nur darüber, dass alles mit dieser Konfiguration funktioniert hat. Es ist an die Entität von einem Fronius Smartmeter gekoppelt.
[image]
Ist das richtig?

syssi, 2024-06-14:
Besteht das Problem noch? Dein Log-Level ist zu niedrig (`INFO`) eingestellt, so dass man keine relevanten Informationen im Log finden kann zur Fehlerdiagnose.

pogo01, 2024-06-14:
Hallo.Danke für die Nachfrage. Ich war wohl auch Opfer eines Wlandefektes an der Fritzbox. Zusätzlich fällt auch die Wechselrichterkommunikation aus.Am besten läuft der Soyo und dein Code.Vielen Dank dafür --Diese Nachricht wurde von meinem Android Mobiltelefon mit WEB.DE Mail gesendet.

closed 2024-06-14 by syssi