syssi/esphome-soyosource-gtn-virtual-meter #172: Fan on Temperature

Chickenbreast0, 2024-04-08:
Good morning,

This is less of an issue for me and more of a question regarding the starting temperature of the fan.
Your code shows an active fan from 45°C. In my case, however, the fan switches to active above 43°C and switches off again below 40.5°C. This means that the ICON of the fan is not active, even though it is already running, as I have only reached 45°C a few times so far.

Is it somehow possible to change this in the code so that the value is set as "Fan on" from 43°C?

[logs_wt32-eth01_logs.txt](https://github.com/syssi/esphome-soyosource-gtn-virtual-meter/files/14901777/logs_wt32-eth01_logs.txt)

syssi, 2024-04-08:
I will compare the threshold on my side another time and make the value configurable if it's different. At the moment it's hardcoded and cannot be changed easily.

Chickenbreast0, 2024-04-08:
Thanks for your reply.
Addtionally I found out, that your current code works corrcetly, if the inverter temp reaches 45°C. Then it shows, "FAN on" even it is running since 43°C.

syssi, 2024-04-09:
The fan of my Soyosource 1200W (Display version + chatty via RS485) turns on at 44.5°C (the display shows `ceil(44.5) = 45°C`) and off around 40°C (40.4°C).

syssi, 2024-04-09:
@Chickenbreast0 Which model do you own?

Chickenbreast0, 2024-04-09:
> @Chickenbreast0 Which model do you own?

Mine is a 1000W model with display and talkative RS485 Port. Those values above I got from the RS485 Port.

Offtopic reg. the talkative part: I replaced the CPU PCB with a "new" one from Aliexpress for testing purposes. Suddenly it stopped showing me all values. After changing it back, it "spoke" to me again. Therefore I guess the communication thing depends on the CPU Board.