syssi/esphome-soyosource-gtn-virtual-meter #176: ESP32-C3 - WiFi version with limiter - Component soyosource_display took a long time for an operation

Wolfgang-Gi, 2024-06-14:
Hello,
I tought to give this tiny board esp32-c3-devkitm-1 a try, it is smaller and more powerful than esp8266.
I installed code from your repository. 
Thanks for that.
It works in general but it generates a lot of this messages.
- Component soyosource_display took a long time for an operation (54 ms).
- Components should block for at most 30 ms.
Do you have any idea to get rid of it?

syssi, 2024-06-14:
No. These warnings are normal and can be ignored.

See https://github.com/esphome/issues/issues/4717

syssi, 2024-07-10:
This is just processing time and 54ms isn't much. Why do you want to speed up this? Did you try to remove some sensors and decrease the log level?

syssi, 2024-07-11:
This isn't a delay. The display port of the devices just pushes some data and the decoding of the data takes 54ms.

Oldie-driver, 2024-07-11:
Thanks. maybe i must here disable the logging coure other reasons. The ESP32-C super mini have a poor  antenna, to poor for mounting inside the alu case of the soyo inverter. I'm currently trying to create a circuit board for the ESP01. The router shows me almost 20dBm more. The resources are somewhat limited and I need 4 GPIO.

closed 2024-07-11 by syssi