syssi/esphome-soyosource-gtn-virtual-meter #177: Readout of Shelly 3 EM broken with update of ESPHome to 2024.6

fploen, 2024-06-20:
Unfortunately, coding changes to http_request does break the previous readout code for Shelly 3 EM.

As workaround for code file shelly-3em-http-status-json.yaml
https://github.com/syssi/esphome-soyosource-gtn-virtual-meter/blob/main/powermeter-sensor-examples/shelly-3em-http-status-json.yaml

I suggest (works for me):

substitutions:
  shelly_3em_url: "http://192.168.xxx.x/status/" // adjust to your Shelly 3 EM IP address

http_request:
  id: http_request0
  useragent: esphome/device
  timeout: 1000ms
  esp8266_disable_ssl_support: true (comment if no ESP8266 in use)

sensor:
  - platform: template
    id: powermeter0
    name: "Instantaneous Power"
    unit_of_measurement: W
    device_class: "power"
    accuracy_decimals: 2
    update_interval: never

interval:
  - interval: 1100ms
    then:
      - http_request.get:
          url: ${shelly_3em_url}
          headers:
            Content-Type: application/json
          capture_response: true
          max_response_buffer_size: 2000
		  on_response:
            then:
              - lambda: |-
                  json::parse_json(body, [](JsonObject root) -> bool {
                    id(powermeter0).publish_state(root["total_power"]);
                    return true;
                  });

syssi, 2024-06-24:
Do you like to make a contribution / prepare a pull request to update the broken(?) snippet.

fploen, 2024-06-24:
> Do you like to make a contribution / prepare a pull request to update the broken(?) snippet.

Actually I see a switch to the updated code version is not possible yet, due to an identified bug in the http_request memory allocation, leading to crash of ESP due to short memory. The bug is identified and should be fixed with the next update.

[See here](https://github.com/esphome/issues/issues/5955)

syssi, 2024-07-18:
Thanks for your contribution. Let's close this issue!

closed 2024-07-18 by syssi