syssi/esphome-soyosource-gtn-virtual-meter #182: Prevent overshooting (esp32 go down)?

Giro-S, 2024-07-28:
Hello to all :)
I have a problem with overshooting very frequent, I tried to change many settings in esphome but without results :/

I have 3 inverters with 1 rs485 bus for to sync them, esp32 mini pilot them
The three circuits are the same:  PV-->MPPT(epever)-->battery-->inverter

This is an yesterday example from 12AM to 5 PM; the main counter sends signails every 3 seconds:
[image]

These are the power of inverters:
[image]
[image]
[image]

Every ~2 minutes the esp32 go down and after go powershooting :/
How to solve this problem?

In the code i tried many values on virtual meter update_interval: 3 sec, 2 sec, 1 sec and 0.5sec
On soyosource_display update_interval tried 5, 6, 7, 8 sec
`soyosource_virtual_meter:
  - id: virtualmeter0
    soyosource_modbus_id: modbus0
    power_id: powermeter0
    power_sensor_inactivity_timeout: 20s
    power_demand_calculation: NEGATIVE_MEASUREMENTS_REQUIRED
    min_power_demand: 0
    max_power_demand: 2700
    zero_output_on_min_power_demand: true
    power_demand_divider: 3
    buffer: 0
    operation_status_id: operation_status_id0
    update_interval: 0.5s

soyosource_display:
  - id: display1
    uart_id: i2c_uart_0
    protocol_version: SOYOSOURCE_WIFI_VERSION
    update_interval: 8s
  - id: display2
    uart_id: i2c_uart_1
    protocol_version: SOYOSOURCE_WIFI_VERSION
    update_interval: 8s
  - id: display3
    uart_id: uart_1
    protocol_version: SOYOSOURCE_WIFI_VERSION
    update_interval: 8s`

syssi, 2024-07-29:
Could you provide a ESPHome log (with log level `DEBUG`) too? The implementation should report some important details here.

syssi, 2024-07-29:
The `update_interval` of the `soyosource_virtual_meter` must be less fast as your smart meter measurements. If you smartmeter publishs a new value every 3 seconds you should use an `update_interval` of around 6s.

Are you able to change the update interval of your smartmeter to `1s`? In this case a `3s` update interval of the `soyosource_virtual_meter` could be a good choice.

Giro-S, 2024-07-29:
Thank you Sebastian :)

> Could you provide a ESPHome log (with log level `DEBUG`) too? The implementation should report some important details here.

Ok, enabled it

> The `update_interval` of the `soyosource_virtual_meter` must be less fast as your smart meter measurements. If you smartmeter publishs a new value every 3 seconds you should use an `update_interval` of around 6s.

Ok, set it to 6sec
> Are you able to change the update interval of your smartmeter to `1s`? In this case a `3s` update interval of the `soyosource_virtual_meter` could be a good choice.

Sorry, no, minimum update is 3 seconds.
This is my meter, works with local tuya
[image]

Can you tell me one that report every second or less without wires? If I use PZEM004T with esphome and set ever 1 second ...should it acceptable?

and this value on 5 seconds?

```
soyosource_display:
  - id: display1
    uart_id: i2c_uart_0
    protocol_version: SOYOSOURCE_WIFI_VERSION
    update_interval: 8s
```

Giro-S, 2024-07-29:
With debug, virtual meter 6 sec and display 5 sec is a disaster :/
[image]

I notice that API connection go down for few seconds and then reup
I reverted back to 0.5sec because is more stable
[image]

syssi, 2024-07-29:
Please provide ESPHome logs next to your charts. The will contain all important details about the behaviour.

Giro-S, 2024-07-29:
> Please provide ESPHome logs next to your charts. The will contain all important details about the behaviour.

Added thank you!

Giro-S, 2024-07-29:
In this log I found what appen when all three inverters go down, API disconnect and reconnect
[logs_esphome-web-25e1b4_logs (1).txt](https://github.com/user-attachments/files/16411981/logs_esphome-web-25e1b4_logs.1.txt)

```WARNING soyosource-gtn-display @ SOYOSOURCE_IP: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for soyosource-gtn-display @ SOYOSOURCE_IP
WARNING Disconnected from API
INFO Successfully connected to soyosource-gtn-display @ SOYOSOURCE_IP in 0.029s
INFO Successful handshake with soyosource-gtn-display @ SOYOSOURCE_IP in 0.052s```

syssi, 2024-07-29:
Could you try to use the `api` or the `mqtt` and not both. Does this reduce the hiccups?

Giro-S, 2024-07-29:
> Potresti provare a usare the `api`o the `mqtt`e non entrambi. Questo riduce il singhiozzo?

Ok, I will comment (#) mqtt section

syssi, 2024-07-29:
> Sorry, no, minimum update is 3 seconds.
> This is my meter, works with local tuya

I'm a lucky guy because the smartmeter of my grid provider pushs an infrared telegram per second. I'm aware of some Shelly 3EM users (which is supported by ESPHome too: https://devices.esphome.io/devices/shelly-3em/). May be this is a good but expensive choice.

syssi, 2024-07-29:
If you decrease the log level again (`INFO`) you will free some additional resources to gain some more stability.

Do you have the chance to avoid WiFi and go the ethernet route? The `WT32-ETH01` is a good candidate here.

Giro-S, 2024-07-29:
> I'm a lucky guy because the smartmeter of my grid provider pushs an infrared telegram per second. I'm aware of some Shelly 3EM users (which is supported by ESPHome too: https://devices.esphome.io/devices/shelly-3em/). May be this is a good but expensive choice.

Oh ok ....another 100€ ahahahah

> If you decrease the log level again (`INFO`) you will free some additional resources to gain some more stability.

Yes, decreased!
 
> Do you have the chance to avoid WiFi and go the ethernet route? The `WT32-ETH01` is a good candidate here.

I've  bought 3 of them, they will arrive on around 6th august...
I will try a WT32-ETH01 with 2 of these:
[image]
Total 4 UART through I2C: 3 for 3 display, 1 for free
Plus esp32 uart0 for rs485 converter
And ETH connection for Home Assistant.

PS: I've read that you are an expert of EPEVER (I have 4 of them with esphome) ...can I ask you a little help about this configuration?
PV-->epever-->battery-->inverter

Always a big thank you!!

syssi, 2024-07-30:
Re PS: Please create an issue [here](https://github.com/syssi/esphome-config-examples) to talk about the Epever.

syssi, 2024-07-30:
Let's close this issue because the root cause is identified. As soon the ESP is always available/properly connected the hiccups shoudn't happen anymore.

closed 2024-07-30 by syssi