syssi/esphome-soyosource-gtn-virtual-meter #184: Power Output von Display Version nicht korrekt

Asathor85, 2024-08-01:
Bei  der Verwendung der "esp8266-display-display-version-example.yaml" ist mir aufgefallen, dass die angezeigten output power Werte nicht stimmen. Bei 90W Limit wird 103W angezeigt, bei 120W 138W bei 200W 260W usw. also um einen konstanten Faktor steigend zu viel.

Sowohl das parallel angeschlossene Display als auch das Sonoff zeigen dagegen die als Limit eingestellten Werte an, die auch tatsächlich gelieferte werden.

Es handelt sich um einen GTN-1000-24 Version 3, also gelbe Platine und CPU. Gemessen im Batterie CP Modus.

[image]

[image]

syssi, 2024-08-01:
Gut aufgepasst! :-) Ja, der Wert wird grundsaetzlich nicht direkt vom Inverter geliefert, sondern muss sich hergeleitet werden. Das WiFi-Dongle vom Hersteller sorgt fuer diese Berechnung und liefert diesen Wert. Ersetzt man das Dongle, so muss man den Wert selbst berechnen. Diese Muehe habe ich mir hier scheinbar nicht gemacht:

https://github.com/syssi/esphome-soyosource-gtn-virtual-meter/blob/main/components/soyosource_display/soyosource_display.cpp#L247-L250

Hast du die Moeglichkeit einen Faktor oder eine Messreihe zur Verfuegung zu stellen um diesen Sensor per Software zu kalibrieren?

Asathor85, 2024-08-01:
Das kann ich gerne machen.
Allerdings müsstest du den Software Sensor kalibrieren. 
Programmierung ist nun wirklich nicht mein Fachgebiet.

BTW: Gibt es dazu eine Einschätzung ob es Probleme gibt das Display und das ESP parallel an einem TTL-UART zu betreiben? Bisher funktioniert es (im Testaufbau)

syssi, 2024-08-01:
> Das kann ich gerne machen.
> Allerdings müsstest du den Software Sensor kalibrieren.
> Programmierung ist nun wirklich nicht mein Fachgebiet.

Ich brauche nur eine Reihe von Ist- vs. Soll-Werten. Dann habe ich alle Voraussetzungen fuer die Implementierung.

> BTW: Gibt es dazu eine Einschätzung ob es Probleme gibt das Display und das ESP parallel an einem TTL-UART zu betreiben? Bisher funktioniert es (im Testaufbau).

Angenommen du laesst den ESP nur lauschen (`RX`) und nicht auf die Leitung schreiben (`TX`), dann kann ich mir keine Probleme vorstellen. Sobald zwei Parteien schreiben (und pollen) ist undefiniert, wie die Display-Firmware sich verhaelt. Der ESPHome-Implementierung ist egal wer die Daten angefragt hat. Es wird jedes Frame decodiert, welches empfangen wird.

Asathor85, 2024-08-02:
Hier die versprochene Messreihe:
Die Werte schwanken +-2Watt daher hier immer der Mittelwert
Set Power / output power ESP / output power real + Display      -      Faktor

10/13/10		1,3
50/56/49		1,12
100/117/99		1,17
150/173/149		1,153
200/228/198		1,14
300/346/299		1,153		
400/460/398		1,15
500/577/498		1,154
600/692/598		1,153
650/750/649		1,154
700/802/697		1,143

Lässt sich also sagen, dass außer in den Bereichen in denen sich der Soyo nicht wohl fühlt, ein globaler Faktor von 1.15 die Berechnung schon ausreichend genau machen würde, falls du keine aufwändige Mehrpunkt Kalibrierung machen willst.

syssi, 2024-08-02:
Lass uns mit einem konstanten Faktor anfangen. Eine exakte Kalibrierung kann dann immer noch individuell per Filter erfolgen: https://www.danielmartingonzalez.com/en/calibrate-sensors-esphome/

syssi, 2024-08-02:
Ich habe eine kleine Aenderung in einen Freature-Branch gepusht. Du kannst diesen nutzen, indem du deine YAML-Konfiguration anpasst:

```
substitutions:
  # ...
  external_components_source: github://syssi/esphome-soyosource-gtn-virtual-meter@respect-efficiency
```

Die Zeile muss von `@main` auf `@respect-efficiency` geaendert werden. Danach bitte einmal neu compilieren & flashen. Danach muesste der Wert des `output power` sensors sehr nah an den Werten des Displays liegen.

syssi, 2024-08-06:
Super! Damit schließe ich den Issue. Erstellen gerne weitere solltest Fehler finden oder Raum für Verbesserungen sehen.

closed 2024-08-06 by syssi

syssi, 2024-08-06:
Ich habe die Änderung nach `main` gemerged. Du kannst diese Zeile deshalb wieder zurück ändern auf:

```
substitutions:
  # ...
  external_components_source: github://syssi/esphome-soyosource-gtn-virtual-meter@main
```

Den Feature-Branch (`respect-efficiency`) werde ich mittelfristig wieder loeschen.