syssi/esphome-soyosource-gtn-virtual-meter #185: PV mode not working

snkbrz, 2024-08-01:
Hello, my gridtie works in PV mode up to 1200W, but here I can't make it work in any way, if I choose any of the PV modes it resets the inverter's output power, and in battery mode it is limited to 900W. I made an adaptation and improved the cooling of the device, I would like it to work at its maximum power. Thanks in advance for your help.

syssi, 2024-08-01:
Which Soyosource model do you own? A display version or the Wi-Fi dongle version? How do you try to change the operation mode? Could you describe your setup in more detail?

syssi, 2024-08-07:
Any news? Can this issue be closed?

snkbrz, 2024-08-07:
> Any news? Can this issue be closed?

Hello. I'm sorry for the delay. Soyosource version 1200W - 110V, without display and with wifi dongle from the manufacturer installed on the esphome with its code and an RS485 module connected to the inverter. I have 2 problems.

1. If you set the operation mode to PV or PV limit, the power is reset, it does not work at all, not even setting manual mode. In this case, I would like this mode so it can work at 1200W, since, as I said, I improved the cooling and I would like to take advantage of that.

2. Power demand divider when I set it to 2, the power is limited to a maximum of 600W, I have 2 working together in the same phase, but if I put the power divider to work the loads better it limits the 2 to a maximum of 500W, only reaching it again the 900W if you set the power demand divider to 1 again, I'm getting around this with an automation in homeassistant for now, but in relation to maximum power I haven't been able to succeed in any way. I will send my code below for 1 of the devices, both are the same, just changing the name and IP address.

```

substitutions:
  name: soyo-wifi-dongle-l1
  friendly_name: soyo-wifi-dongle-l1
  esphome_rest_api_sensor_url: "http://192.168.177.120/sensor/fase_l1_power"
  shelly_3em_url: "http://192.168.177.7/emeter/0/3em_data"
  device_description: "Monitor and control the WiFi version of the Soyosource GTN using the OEM WiFi dongle with a custom firmware"
  external_components_source: github://syssi/esphome-soyosource-gtn-virtual-meter@main
  tx_pin_rs485: GPIO14
  rx_pin_rs485: GPIO5

esphome:
  name: ${name}
  comment: ${device_description}
  project:
    name: "syssi.esphome-soyosource-gtn-virtual-meter"
    version: 2.1.0

esp8266:
  board: esp12e

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

logger:
  baud_rate: 0
  level: ERROR

uart:
  - id: uart_0
    baud_rate: 9600
    tx_pin: GPIO1
    rx_pin: GPIO3
  - id: uart_1
    baud_rate: 4800
    tx_pin: ${tx_pin_rs485}
    rx_pin: ${rx_pin_rs485}

soyosource_modbus:
  - id: modbus0
    uart_id: uart_1
    name: l1_inverter

mqtt:
  broker: !secret mqtt_host
  username: !secret mqtt_username
  password: !secret mqtt_password
  id: mqtt_client

soyosource_virtual_meter:
  - id: virtualmeter0
    soyosource_modbus_id: modbus0
    power_id: powermeter0
    power_sensor_inactivity_timeout: 20s
    power_demand_calculation: NEGATIVE_MEASUREMENTS_REQUIRED
    min_power_demand: 0
    max_power_demand: 1200
    zero_output_on_min_power_demand: true
    power_demand_divider: 1
    buffer: 10
    operation_status_id: operation_status_id0
    update_interval: 1s

soyosource_display:
  - uart_id: uart_0
    protocol_version: SOYOSOURCE_WIFI_VERSION
    update_interval: 5s

binary_sensor:
  - platform: soyosource_display
    fan_running:
      name: "${name} fan running"
    limiter_connected:
      name: "${name} limiter connected"

button:
  - platform: soyosource_display
    restart:
      name: "${name} restart"

number:
  - platform: soyosource_virtual_meter
    soyosource_virtual_meter_id: virtualmeter0
    buffer:
      name: "${name} buffer"
      initial_value: 10
      restore_value: true
    manual_power_demand:
      name: "${name} manual power demand"
      max_value: 1200
    max_power_demand:
      name: "${name} max power demand"
      initial_value: 1200
      max_value: 1200
      restore_value: true
    power_demand_divider:
      name: "${name} power demand divider"
      initial_value: 2
      max_value: 3
      restore_value: true

  - platform: soyosource_display
    start_voltage:
      name: "${name} start voltage"
    shutdown_voltage:
      name: "${name} shutdown voltage"
    output_power_limit:
      name: "${name} output power limit"
      max_value: 1200
    start_delay:
      name: "${name} start delay"

select:
  - platform: soyosource_display
    operation_mode:
      name: "${name} operation mode"
      optionsmap:
        1: "PV"
        2: "Battery Constant Power"
        17: "PV Limit"
        18: "Battery Limit"

sensor:
  - platform: soyosource_display
    error_bitmask:
      name: "${name} error bitmask"
    operation_mode_id:
      name: "${name} operation mode id"
    operation_status_id:
      name: "${name} operation status id"
      id: operation_status_id0
    battery_voltage:
      name: "${name} battery voltage"
    battery_current:
      name: "${name} battery current"
    battery_power:
      name: "${name} battery power"
    ac_voltage:
      name: "${name} ac voltage"
    ac_frequency:
      name: "${name} ac frequency"
    temperature:
      name: "${name} temperature"
    total_energy:
      name: "${name} total energy"
    output_power:
      name: "${name} output power"

  - id: powermeter0
    platform: mqtt_subscribe
    name: "Instantaneous Power"
    topic: "openbk_powermeter_clamp/16/get"
    accuracy_decimals: 2
    unit_of_measurement: W
    device_class: power
    filters:
      - lambda: return x * 0.1;

text_sensor:
  - platform: soyosource_virtual_meter
    soyosource_virtual_meter_id: virtualmeter0
    operation_mode:
      name: "${name} limiter operation mode"

  - platform: soyosource_display
    errors:
      name: "${name} errors"
    operation_mode:
      name: "${name} operation mode"
    operation_status:
      name: "${name} operation status"

switch:
  - platform: soyosource_virtual_meter
    soyosource_virtual_meter_id: virtualmeter0
    manual_mode:
      name: "${name} manual mode"
      restore_mode: RESTORE_DEFAULT_OFF
    emergency_power_off:
      name: "${name} emergency power off"
      restore_mode: RESTORE_DEFAULT_OFF

http_request:
  id: http_request0
  useragent: esphome/device
  timeout: 1000ms

```

syssi, 2024-08-08:
If you attach two inverters to the same RS485 you have to increase the `max_power_demand` setting too:

```
soyosource_virtual_meter:
  - id: virtualmeter0
    soyosource_modbus_id: modbus0
    # ...
    max_power_demand: 2400
    power_demand_divider: 2
    # ...
```

syssi, 2024-08-08:
Could you try to describe paragraph 1 in more detail? What do you mean by "the power is reset"? You are able to use the OEM WiFi dongle and/or the `soyosource_display` implementation to change the operation mode to "PV limit"?

If the operation mode is "PV" or "PV limit" the inverter doesn't accept power demands via RS485?

snkbrz, 2024-08-08:
> Could you try to describe paragraph 1 in more detail? What do you mean by "the power is reset"? You are able to use the OEM WiFi dongle and/or the `soyosource_display` implementation to change the operation mode to "PV limit"?
> If the operation mode is "PV" or "PV limit" the inverter doesn't accept power demands via RS485? **YES**

Each inverter has its dongle converted to esphome. I don't have 2 on the same RS485
and 
How you said, If the operation mode is "PV" or "PV limit" the inverter doesn't accept power demands via RS485.

My inverter's motherboard is purple and yesterday, reading the solved problems, I read that there is a difference, you asked someone to test a different version at the time, but I don't know if it now works in the standard version or do I have to use the one you indicated? At the time I use it, I don't know if it's still valid.

github://syssi/esphome-soyosource-gtn-virtual-meter@add-operation-mode

I no longer have the original firmware on any dongle and my version doesn't have a display. Otherwise, I'll have to stop one day to disassemble everything and find the firmware backup to return to the standard version.

I apologize for the mistakes, for using a translator

Thank you in advance

syssi, 2024-08-08:
Okay. You are using this project/implementation to switch between operation modes of the inverter. There is the risk this implementation doesn't fit to your inverter/firmware version/setup. If you have the chance please try to restore the original firmware of the WiFi dongle and try to change the operation mode using the original firmware. Does the inverter behaves different this time?

kev300, 2024-11-04:
I habe the display version, but I have the same issue some times.
It changes the operation mode and output limit options on it's own. I don't know why.

My fix is to just set the values in an automation via esphome back to 900W and also the operation mode to battery limit (in my case)
This should also work with the wifi version I think?

`  - interval: 101s                                                                                                                                                                    startup_delay: 19s                                                                                                                                                                then:                                                                                                                                                                               - if:                                                                                                                                                                                 condition:                                                                                                                                                                          - lambda: "return { id(soyo_output_limit).state != 900.0 };"                                                                                                                    then:                                                                                                                                                                               - number.set:                                                                                                                                                                         id: soyo_output_limit                                                                                                                                                             value: 900                                                                                                                                                              - if:                                                                                                                                                                                 condition:                                                                                                                                                                          - lambda: "return { id(soyo_start_delay).state != 3.0 };"                                                                                                                       then:                                                                                                                                                                               - number.set:                                                                                                                                                                         id: soyo_start_delay                                                                                                                                                              value: 3                                                                                                                                                                - if:                                                                                                                                                                                 condition:                                                                                                                                                                          - lambda: "return { id(soyo_operation_mode).state == \"Battery Constant Power\" };"                                                                                             then:                                                                                                                                                                               - select.set:                                                                                                                                                                         id: soyo_operation_select                                                                                                                                                         option: "Battery Limit"`