syssi/esphome-soyosource-gtn-virtual-meter #188: Inverter correctly doing zero export, but no information returned from inverter

tomasro, 2024-08-12:
[image]
I am using a Shelly EM1 to get my Mains power draw, and that is being received by a D1 Mini Esp32 (probably a clone) with TX/RX on Uart2 pins (GPIO17/16) to an rs485 converter board (exactly as pictured in the readme).

```
substitutions:
  name: "gtn-l1"
  friendly_name: GTN-L1
  device_description: "Monitor a GTN-L1 and control the power output on demand both via RS485"
  external_components_source: github://syssi/esphome-soyosource-gtn-virtual-meter@main
  tx_pin: GPIO17
  rx_pin: GPIO16

esphome:
  name: ${name}
  comment: ${device_description}
  min_version: 2024.6.0
  project:
    name: "syssi.esphome-soyosource-gtn-virtual-meter"
    version: 2.2.0

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:
  level: DEBUG
  logs:
    api.service: WARN
    ota: WARN
    sensor: DEBUG

# Enable Home Assistant API
api:

# Allow Over-The-Air updates
ota:
- platform: esphome

# Allow provisioning Wi-Fi via serial
improv_serial:

wifi:
  # Set up a wifi access point
  ap: {}

# In combination with the `ap` this allows the user
# to provision wifi credentials to the device via WiFi AP.
captive_portal:

dashboard_import:
  package_import_url: github://esphome/example-configs/esphome-web/esp32.yaml@main
  import_full_config: true

# Sets up Bluetooth LE (Only on ESP32) to allow the user
# to provision wifi credentials to the device.
esp32_improv:
  authorizer: none

# To have a "next url" for improv serial
web_server:

external_components:
  - source: ${external_components_source}
    refresh: 0s

uart:
  - id: uart_0
    baud_rate: 4800
    tx_pin: ${tx_pin}
    rx_pin: ${rx_pin}
    debug:
      direction: BOTH

soyosource_modbus:
  - id: modbus0
    uart_id: uart_0
    # flow_control_pin: GPIO12

    # Optional settings
    #
    # The name is used as prefix for some log messages and should
    # help to distinguish between different instances/devices
    name: GTN-L1

soyosource_virtual_meter:
  - id: virtualmeter0
    soyosource_modbus_id: modbus0

    # The state of this sensor (instantaneous power in watt) is used as source
    power_id: powermeter0

    # Optional settings
    power_sensor_inactivity_timeout: 20s
    power_demand_calculation: NEGATIVE_MEASUREMENTS_REQUIRED
    min_power_demand: 0
    max_power_demand: 900
    zero_output_on_min_power_demand: true
    # Split/distribute the power demand if you have multiple inverters attached to the same RS485 bus
    power_demand_divider: 1
    # A positive buffer value (10) tries to avoid exporting power to the grid (demand - 10 watts)
    # A negative buffer value (-10) exports power to the grid (demand + 10 watts)
    buffer: 10
    # The operation_status_id sensor is expected here. Passing the operation_status won't work
    # The state is used to suspend the limiter if the operation status of the inverter isn't 0x0 (normal)
    # operation_status_id: operation_status_id0

    # The update interval is important and defaults to 3 seconds. If the demand is sent too frequently
    # or rarely the interverter stops. TODO: Identify and validate the lower and upper update limit
    update_interval: 3s

soyosource_inverter:
  - id: inverter0
    soyosource_modbus_id: modbus0

binary_sensor:
  - platform: soyosource_inverter
    soyosource_inverter_id: inverter0
    fan_running:
      name: "${name} fan running"

number:
  - platform: soyosource_virtual_meter
    soyosource_virtual_meter_id: virtualmeter0
    buffer:
      name: "${name} buffer"
      initial_value: 10
      restore_value: true
    manual_power_demand:
      name: "${name} manual power demand"
      max_value: 900
    max_power_demand:
      name: "${name} max power demand"
      initial_value: 600
      max_value: 900
      restore_value: true
    power_demand_divider:
      name: "${name} power demand divider"
      initial_value: 1
      restore_value: true

sensor:
  - platform: soyosource_virtual_meter
    soyosource_virtual_meter_id: virtualmeter0
    power_demand:
      name: "${name} power demand"

  - platform: soyosource_inverter
    soyosource_inverter_id: inverter0
    operation_status_id:
      name: "${name} operation status id"
      id: operation_status_id0
    battery_voltage:
      name: "${name} battery voltage"
    battery_current:
      name: "${name} battery current"
    battery_power:
      name: "${name} battery power"
    ac_voltage:
      name: "${name} ac voltage"
    ac_frequency:
      name: "${name} ac frequency"
    temperature:
      name: "${name} temperature"

   # import smartmeter reading from homeassistant
   # requires the "api" component see above
  - platform: homeassistant
    id: powermeter0
    name: "${name} smartmeter instantaneous power"
    entity_id: sensor.shellyem_channel_1_power
    filters:
      - throttle_average: 15s

switch:
  - platform: soyosource_virtual_meter
    soyosource_virtual_meter_id: virtualmeter0
    manual_mode:
      name: "${name} manual mode"
      restore_mode: RESTORE_DEFAULT_ON
    emergency_power_off:
      name: "${name} emergency power off"
      restore_mode: RESTORE_DEFAULT_OFF

text_sensor:
  - platform: soyosource_inverter
    soyosource_inverter_id: inverter0
    operation_status:
      name: "${name} operation status"

  - platform: soyosource_virtual_meter
    soyosource_virtual_meter_id: virtualmeter0
    operation_mode:
      name: "${name} limiter operation mode"
```

Help me Obi-Wan! You're my only hope!

Thanks!

syssi, 2024-08-12:
It looks like you have a silent inverter. It's a display or wifi dongle model?

tomasro, 2024-08-12:
it has an LCD: https://www.amazon.com/dp/B0CKQVQLQZ?psc=1&ref=ppx_yo2ov_dt_b_product_details

syssi, 2024-08-13:
Long story short: Your inverter doesn't respond for unkown reason. See https://github.com/syssi/esphome-soyosource-gtn-virtual-meter?tab=readme-ov-file#user-content-fn-1-4d9fb981e2326a06ba26816e33db5256

You can remove the `soyosource_inverter` component from your YAML configuration including all related entities. If you are still interested in some status informations from your inverter you could tap the interal display port. This YAML is able to decode the display port traffic of your inverter using a dedicated uart:

https://github.com/syssi/esphome-soyosource-gtn-virtual-meter/blob/main/esp8266-display-display-version-example.yaml

And this could be your final setup: https://github.com/syssi/esphome-soyosource-gtn-virtual-meter/blob/main/esp8266-display-display-version-limiter-example.yaml

This configuration talks to the display port using `uart_0` and publishs a power demand on `uart_1`.

tomasro, 2024-08-14:
too bad. I don't feel like doing surgery on my new inverters, so I'll just leave it be. Hopefully something will be discovered for the newer boards in the future. Thank you for making the integration!

closed 2024-08-14 by tomasro