syssi/esphome-soyosource-gtn-virtual-meter #196: Inactivity Timeout

pogo01, 2024-11-29:
Moin und Sorry für die Störung.
Ich habe ein Problem, dass das ESP nach ein paar Sekunden in den Inactivity Timeout fällt. Ich habe die Fritzbox zurückgesetzt, den Radberry Pi, ESP Home, und den ESP 8266. Alles X-Fach und ohne grosses Ergebnis.
Hat jemand eine Ahnung, was ich machen kann?
Den Wemos D1 Mini mit einem anderen Tool resetten?
Bitte um Hülfe

syssi, 2024-11-29:
Das "Inactivity timeout" tritt ein, wenn die Stromverbrauchsmeldungen vom Stromzähler zu selten empfangen werden. Entweder musst du das Timeout anheben oder deinen Stromzähler überreden häufiger Meldung zu machen. Wie häufig gibt es denn einem Wert vom Stromzähler und kommt er auch sicher an?

pogo01, 2024-11-29:
Danke für die superschnelle Antwort. Ich habe die Entität gewechselt und die Zeit angepasst. Im Moment läuft alles wieder ganz toll.
Danke

closed 2024-11-30 by syssi