syssi/esphome-soyosource-gtn-virtual-meter #204: Fehlermeldung beim kompilieren

thommy6102, 2025-05-21:
Hallo Sebastian,
ich kriege diese Fehlermeldung wenn ich versuche daas weiter unten angehängte yaml File zu kompilieren:
Ich benutze ESPhome unter Windows 11, und eigentlich nur zum kompilieren, bzw. die bin Files zu erzeugen.

Microsoft Windows [Version 10.0.22631.5335]
(c) Microsoft Corporation. Alle Rechte vorbehalten.

c:\esphome-soyosource-gtn-virtual-meter-main>esphome run soyo-mit-regelung-ok-erweiterung-test.yaml
INFO ESPHome 2025.4.2
INFO Reading configuration soyo-mit-regelung-ok-erweiterung-test.yaml...
INFO Updating https://github.com/syssi/esphome-soyosource-gtn-virtual-meter.git@main
Failed config

soyosource_virtual_meter: [source soyo-mit-regelung-ok-erweiterung-test.yaml:73]
  - id: virtualmeter0
    soyosource_modbus_id: modbus0

    Couldn't find ID 'powermeter0'. Please check you have defined an ID with that name in your configuration.
    power_id: powermeter0
    power_sensor_inactivity_timeout: 20s
    power_demand_calculation: NEGATIVE_MEASUREMENTS_REQUIRED
    min_power_demand: 0
    max_power_demand: 900
    zero_output_on_min_power_demand: True
    power_demand_divider: 1
    buffer: 10
    operation_status_id: operation_status_id0
    update_interval: 3s
interval: [source soyo-mit-regelung-ok-erweiterung-test.yaml:243]
  - interval: 1100ms
    then:
      - http_request.get:
          url: http://192.168.2.56/cm?cmnd=status+10
          headers:
            Content-Type: application/json
          on_response:
            - then:
                -
                  Couldn't find ID 'powermeter0'. Please check you have defined an ID with that name in your configuration.
                  lambda: !lambda |-
                    std::string response_data = id(http_request0).get_string();
                    if(!response_data.empty()) {
                      json::parse_json(response_data, [](JsonObject root) {

c:\esphome-soyosource-gtn-virtual-meter-main> 

Das doofe ist, dass ich das alles vor 2 Jahren schonmal gemacht hatte. Und da hat es geklappt.

Hier mein Problem yaml:

substitutions:
  name: test-fuer-tasmota-ir-lesekopf
  device_description: "Monitor and control the Soyosource GTN (WiFi version) via the display port and control the power output on demand via RS485"
  external_components_source: github://syssi/esphome-soyosource-gtn-virtual-meter@main
  # The display port should be connected to these GPIOs using a USB connector (VCC, RX, TX, GND)
  # A optocoupler / optical isolator (f.e. ADUM1201) between the display port and the ESP is recommended if the ESP isn't powered by the display port!
  tx_pin_ttl_wifi: GPIO4
  rx_pin_ttl_wifi: GPIO5
  tx_pin_rs485: GPIO14
  rx_pin_rs485: GPIO12
  tasmota_status_url: "http://xxx.xxx.2.56/cm?cmnd=status+10"

esphome:
  name: ${name} 
  comment: ${device_description}
  min_version: 2024.6.0
  project:
    name: "syssi.esphome-soyosource-gtn-virtual-meter"
    version: 2.2.0

[... 217 lines cut ...]
      name: "${name} operation status"

  
  
interval:
  - interval: 1100ms
    then:
      - http_request.get:
          url: ${tasmota_status_url}
          headers:
            Content-Type: application/json
          #verify_ssl: false
          on_response:
            then:
              - lambda: |-
                  std::string response_data = id(http_request0).get_string();
                  if(!response_data.empty()) {
                    json::parse_json(response_data, [](JsonObject root) {
                      id(powermeter0).publish_state(root["StatusSNS"]["SML"]["curr_w"]);
                    });
                  }  

Kannst Du mir nochmal helfen?

73 de Thomas

thommy6102, 2025-05-21:
Nein die Verbrauchserhebung ist unverändert.

Habs gerade mal probiert. Geht leider nicht.

Kann das hier alles drin bleiben:

 - platform: soyosource_virtual_meter
    soyosource_virtual_meter_id: virtualmeter0
    power_demand:
      name: "${name} power demand"

  - platform: soyosource_display
    error_bitmask:
      name: "${name} error bitmask"
    operation_mode_id:
      name: "${name} operation mode id"
    operation_status_id:
      name: "${name} operation status id"
      id: operation_status_id0
    battery_voltage:
      name: "${name} battery voltage"
    battery_current:
      name: "${name} battery current"
    battery_power:
      name: "${name} battery power"
    ac_voltage:
      name: "${name} ac voltage"
    ac_frequency:
      name: "${name} ac frequency"
    temperature:
      name: "${name} temperature"
    output_power:
      name: "${name} output power"

Falls ja, kommt das oben von Dir genannte davor oder dahinter?

thommy6102, 2025-05-21:
Ich bekomme diese Fehlermeldung beim kompilieren:

Microsoft Windows [Version 10.0.22631.5335]
(c) Microsoft Corporation. Alle Rechte vorbehalten.

c:\esphome-soyosource-gtn-virtual-meter-main>esphome run soyo-mit-regelung-ok-erweiterung-test.yaml
INFO ESPHome 2025.4.2
INFO Reading configuration soyo-mit-regelung-ok-erweiterung-test.yaml...
ERROR Error while reading config: Invalid YAML syntax:

while parsing a block collection
  in "soyo-mit-regelung-ok-erweiterung-test.yaml", line 168, column 3
expected <block end>, but found '<block sequence start>'
  in "soyo-mit-regelung-ok-erweiterung-test.yaml", line 196, column 4

c:\esphome-soyosource-gtn-virtual-meter-main>

Und so sieht jetzt meine Sequenz Sensor aus:

sensor:
  - platform: soyosource_virtual_meter
    soyosource_virtual_meter_id: virtualmeter0
    power_demand:
      name: "${name} power demand"

  - platform: soyosource_display
    error_bitmask:
      name: "${name} error bitmask"
    operation_mode_id:
      name: "${name} operation mode id"
    operation_status_id:
      name: "${name} operation status id"
      id: operation_status_id0
    battery_voltage:
      name: "${name} battery voltage"
    battery_current:
      name: "${name} battery current"
    battery_power:
      name: "${name} battery power"
    ac_voltage:
      name: "${name} ac voltage"
    ac_frequency:
      name: "${name} ac frequency"
    temperature:
      name: "${name} temperature"
    output_power:
      name: "${name} output power"
      
   - platform: template
     id: powermeter0
     name: "Instantaneous Power"
     unit_of_measurement: W
     device_class: "power"
     accuracy_decimals: 2
     update_interval: never

syssi, 2025-05-21:
Ich gesehe gerade, dass die Lambda-Funktion im `interval`-Abschnitt nicht mehr funktioniert, weil sich die Art geändert hat, wie man HTTP-Requests verarbeitet. Ich denke über eine Lösung nach.

thommy6102, 2025-05-22:
Zunächst einmal danke das Du mir hilfst.
Die korrigierte Version läuft bei mir auch durch, bringt aber dann eine Fehlermeldung in der lambda Funktion.

Wäre toll wenn Du eine Lösung finden könntest.

73

syssi, 2025-05-22:
Versuch es mal bitte mit dieser Variante:

```
interval:
  - interval: 1100ms
    then:
      - http_request.get:
          url: ${tasmota_status_url}
          headers:
            Content-Type: application/json
          #verify_ssl: false
          capture_response: true
          on_response:
            then:
              - lambda: |-
                  if (response->status_code == 200 && body.length() > 0) {
                    json::parse_json(body, [](JsonObject root) -> bool {
                      id(powermeter0).publish_state(root["StatusSNS"]["SML"]["curr_w"]);
                      return true;
                    });
                  }
```

Sie ist ungetestet. Ich habe aber die Hoffnung, dass sie auf Anhieb ihren Zweck erfüllt.

thommy6102, 2025-05-22:
Mit dieser Variante funktioniert das kompilieren.

Ich hatte eben vergessen die IP Adresse von meinem Tasmota Lesekopf im yaml file einzutragen.
Hatte somit natürlich keine Verbrauchswerte.

Nachdem ich danach dann die IP Adresse vom Tasmota Lesekopf korrekt eingetragen, und erneut kompiliert hatte, gabs dann diese Fehlermeldung im Halbsekundentakt:

[14:37:07][E][http_request.arduino:138]: Stream pointer vanished!

syssi, 2025-05-22:
Spannend. Ich gehe noch einmal in mich. Ich erinnere mich, dass der `http_request.arduino` sehr zimperlich ist, wenn die Gegenseite die Daten in Chunks und nicht am Stück anbietet.

thommy6102, 2025-05-22:
Danke!

syssi, 2025-05-22:
Just for reference: https://github.com/esphome/issues/issues/5949

thommy6102, 2025-05-26:
Hallo Sebastian,

ich habe mal auf meinem Raspi3 mit Homeassistant, einige ältere Versionen von ESPHome (2023.10, 2024.5 und 2024.6) installiert und ausprobiert.
Damit gibt es aber jedesmal unterschiedliche Fehlermeldungen beim kompilieren. Ich meine innerhalb der von Dir genannten Referenz,  verstanden zu haben, dass es wohl mit den älteren Versionen funktionieren sollte.
Tuts aber nicht.
Wäre toll wenn Du eine Lösung hättest.

thommy6102, 2025-06-04:
Hallo Sebastian,

gibt`s vielleicht Neuigkeiten?

ASAS443, 2025-06-14:
Gerne hilfe, was ich tun mus
Danke Andre

INFO Unable to import component soyosource_modbus: No module named 'esphome.components.soyosource_modbus'
Failed config

soyosource_modbus: [source /config/esphome/soyosource-gtn-virtual-meter.yaml:61]
  
  Component not found: soyosource_modbus.
  - id: modbus0
    uart_id: uart_0
    name: firstfloor_inverter

thommy6102, 2025-06-30:
Hallo Sebastian,

Die Variante die Du mir am 22.05.25 lässt sich kompilieren, bringt dann aber im laufenden Betrieb im Web Interface die Fehlermeldung:

[14:37:07][E][http_request.arduino:138]: Stream pointer vanished!

Aber das hatte ich ja bereits mitgeteilt.

Blöderweise kann ich bei mir das "alte" yaml File, bei dem Du mir vor 2 Jahren bereits einmal geholfen hattest, weder unter Win 11, noch unter Rasbian mit einem Raspi 3 nicht mehr kompilieren.
Wenn ich bloss wüsste warum das ältere File auch nicht zu kompilieren ist. Wollte mir die Steuerung mittels MQTT im IObroker realisieren.

Dürfte ich Dich wohl noch mal um Hilfe bitten?

Gruß
Thomas

syssi, 2025-06-30:
Kannst du diese (zusätzliche) externe Komponente einbinden:

```
external_components:
  - source: ${external_components_source}
    refresh: 0s
  - source: github://pr#7884
    components: [ http_request ]
```

Dies sorgt dafuer, dass eine erweiterte `http_request`-Komponent aus dem Pull Request 7884 gezogen wird. Mit etwas Glück verhält sich der HTTP-Client danach besser. Probiers mal aus!

thommy6102, 2025-07-01:
Danke, dass Du versuchst mir zu helfen.

So habe ich das o.g. eingebunden:
Auszug aus meinem yaml File:

esp8266:
  board: d1_mini

external_components:
  - source: ${external_components_source}
    refresh: 0s
  - source: github://pr#7884
    components: [ http_request ]
    
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

Beim kompilieren gibt es leider diese Fehlermeldung:

Microsoft Windows [Version 10.0.22631.5472]
(c) Microsoft Corporation. Alle Rechte vorbehalten.

c:\esphome-soyosource-gtn-virtual-meter-main>esphome run neutest.yaml
INFO ESPHome 2025.4.2
INFO Reading configuration neutest.yaml...
INFO Updating https://github.com/syssi/esphome-soyosource-gtn-virtual-meter.git@main
INFO Cloning https://github.com/esphome/esphome.git@pull/7884/head
INFO Fetching pull/7884/head
INFO Unable to import component http_request: No module named 'esphome.components.const'
Failed config

http_request: [source neutest.yaml:160]

  Component not found: http_request.
  id: http_request0
  useragent: esphome/device
  timeout: 1000ms
  verify_ssl: False

c:\esphome-soyosource-gtn-virtual-meter-main>

syssi, 2025-07-02:
Kannst du ESPHome auf die aktuellste Version aktualisieren? Aktuell nutzt du 2025.4.2.

thommy6102, 2025-07-02:
Für Windows scheint es keine neuere als 2025.4.2. zu geben.

Auf meinem Raspi habe ich soeben die ESPHome Version 2025.7.0-dev20250702 installiert.
Leider immer noch mit Fehlermeldung beim kompilieren.

INFO ESPHome 2025.7.0-dev
INFO Reading configuration /config/esphome/neutest.yaml...
INFO Updating https://github.com/syssi/esphome-soyosource-gtn-virtual-meter.git@main
INFO Generating C++ source...
INFO Compiling app...
Processing soya-1 (board: d1_mini; framework: arduino; platform: platformio/espressif8266@4.2.1)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
Dependency Graph
|-- ESPAsyncTCP @ 2.0.0
|-- ESP8266WiFi @ 1.0
|-- ESPAsyncWebServer @ 3.7.8
|-- ESP8266mDNS @ 1.2
|-- AsyncMqttClient-esphome @ 2.0.0
|-- ArduinoJson @ 6.18.5
|-- ESP8266HTTPClient @ 1.2
Compiling .pioenvs/soya-1/src/esphome/components/binary_sensor/automation.cpp.o
xtensa-lx106-elf-g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
*** [.pioenvs/soya-1/src/esphome/components/binary_sensor/automation.cpp.o] Error 1
========================= [FAILED] Took 19.56 seconds =========================

So sieht nun mein yaml File aus:

substitutions:
  name: soya-1
  device_description: "Monitor and control the Soyosource GTN (WiFi version) via the display port and control the power output on demand via RS485"
  external_components_source: github://syssi/esphome-soyosource-gtn-virtual-meter@main
  # The display port should be connected to these GPIOs using a USB connector (VCC, RX, TX, GND)
  # A optocoupler / optical isolator (f.e. ADUM1201) between the display port and the ESP is recommended if the ESP isn't powered by the display port!
  tx_pin_ttl_wifi: GPIO4
  rx_pin_ttl_wifi: GPIO5
  tx_pin_rs485: GPIO14
  rx_pin_rs485: GPIO12
  tasmota_status_url: "http://192.168.2.56/cm?cmnd=status+10"

esphome:
  name: ${name} 
  comment: ${device_description}
  min_version: 2024.6.0
  project:
    name: "syssi.esphome-soyosource-gtn-virtual-meter"
    version: 2.2.0

esp8266:
  board: d1_mini

external_components:
  - source: ${external_components_source}
    refresh: 0s
  - source: github://pr#7884
    components: [ http_request ]
    
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:
  platform: esphome
  
web_server:

logger:
  level: INFO

[... 201 lines cut ...]
    errors:
      name: "${name} errors"
    operation_mode:
      name: "${name} operation mode"
    operation_status:
      name: "${name} operation status"

interval:
  - interval: 1100ms
    then:
      - http_request.get:
          url: ${tasmota_status_url}
          headers:
            Content-Type: application/json
          #verify_ssl: false
          capture_response: true
          on_response:
            then:
              - lambda: |-
                  if (response->status_code == 200 && body.length() > 0) {
                    json::parse_json(body, [](JsonObject root) -> bool {
                      id(powermeter0).publish_state(root["StatusSNS"]["SML"]["curr_w"]);
                      return true;
                    });
                  }

thommy6102, 2025-07-02:
Ähh, wie kriege ich denn hier die Formatierung hin? Das sieht ja furchtbar aus!
Hatte mein yaml aus notepad++ herrauskopiert.

syssi, 2025-07-02:
Please try a virtual machine (f.e. virtualbox). The Raspberry Pi hasn't enough RAM to compile the YAML.

thommy6102, 2025-07-26:
Erfolg!!!!
:)

Melde mich später.
Erst Mal Dankeschön bis hierhin.

thommy6102, 2025-07-29:
Hallo Sebastian,

nachdem ich es mit Virtual Box von Oracle versucht hatte, bin ich bei Hyper-V gelandet.
Damit konnte ich mir unter Win 11 Ubuntu in der VM installieren. War alles Neuland für mich und ziemlich tricky.

Mit Ubuntu konnte ich nun ein Homeassistant Image für x86 Rechner auf eine Festplatte spielen.
Mit Balena Etcher konnte ich zunächst nicht das Image auf die Festplatte übertragen, jedoch mit Disks was bei meinem Ubuntu bedreits integriert war.

Damit gebootet hatte ich nun ein Homeassistant für meinen "normalen" Computer.
Dort hatte ich dann das ESPHome Addon installiert, und konnte damit endlich mein yaml File ohne Fehlermeldungen kompilieren, und ein lauffähiges bin File mit MQTT, und alles was ich brauche, erzeugen.

Im Webinterface wir allerdings, in gelber Schrift, diese Meldung angezeigt:

18:46:06 | [W] | [http_request.arduino:066] | Using HTTP on Arduino version >= 3.1 is **very** slow. Consider setting framework version to 3.0.2 in your YAML, or use HTTPS

Ist das von Bedeutung?
Wie kann ich das ändern/abschalten?

Nochmal dankeschön und 73.

Thomas